openwrt/openwrt #17200: Busybox 1.37 fails to build for octeon

flygarn12, 2024-12-07:
### Describe the bug

When I tried to build from main after the Busybox 1.37 update the build work with mvbu wrt3200acm but fails with busybox error when building image for octeon Edgerouter4. Both builds have the same config package setup.

### OpenWrt version

main

### OpenWrt release

9c55866

### OpenWrt target/subtarget

octeon

### Device

edgerouter4

### Image kind

Self-built image

### Steps to reproduce

```
git clone https://github.com/openwrt/openwrt.git ~/openwrt/op/er4_6.6
cd ~/openwrt/op/er4_6.6

git branch -a
git checkout main

./scripts/feeds update -a

./scripts/feeds install -a

cp K6.6_diffconfig_er4 .config #The diffconfig below

make defconfig

make menuconfig

make clean

make V=sc
```

### Actual behaviour

The build fails in the end with this message:
```
Trying libraries: 
Failed: 
Output of:
mips64-openwrt-linux-musl-gcc -Wall -Wshadow -Wwrite-strings -Wundef -Wstrict-prototypes -Wunused -Wunused-parameter -Wunused-function -Wunused-value -Wmissing-prototypes -Wmissing-declarations -Wno-format-security -Wdeclaration-after-statement -Wold-style-definition -finline-limit=0 -fno-builtin-strlen -fomit-frame-pointer -ffunction-sections -fdata-sections -fno-guess-branch-probability -funsigned-char -falign-functions=1 -falign-jumps=1 -falign-labels=1 -falign-loops=1 -fno-unwind-tables -fno-asynchronous-unwind-tables -fno-builtin-printf -Oz -L/home/daniel/openwrt/op/er4_6.6/staging_dir/toolchain-mips64_octeonplus_64_gcc-13.3.0_musl/usr/lib -L/home/daniel/openwrt/op/er4_6.6/staging_dir/toolchain-mips64_octeonplus_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -flto=auto -fuse-linker-plugin -znow -zrelro -o busybox_unstripped -Wl,--sort-common -Wl,--sort-section,alignment -Wl,--gc-sections -Wl,--start-group applets/built-in.o archival/lib.a archival/libarchive/lib.a console-tools/lib.a coreutils/lib.a coreutils/libcoreutils/lib.a debianutils/lib.a klibc-utils/lib.a e2fsprogs/lib.a editors/lib.a findutils/lib.a init/lib.a libbb/lib.a libpwdgrp/lib.a loginutils/lib.a mailutils/lib.a miscutils/lib.a modutils/lib.a networking/lib.a networking/libiproute/lib.a networking/udhcp/lib.a printutils/lib.a procps/lib.a runit/lib.a selinux/lib.a shell/lib.a sysklogd/lib.a util-linux/lib.a util-linux/volume_id/lib.a archival/built-in.o archival/libarchive/built-in.o console-tools/built-in.o coreutils/built-in.o coreutils/libcoreutils/built-in.o debianutils/built-in.o klibc-utils/built-in.o e2fsprogs/built-in.o editors/built-in.o findutils/built-in.o init/built-in.o libbb/built-in.o libpwdgrp/built-in.o loginutils/built-in.o mailutils/built-in.o miscutils/built-in.o modutils/built-in.o networking/built-in.o networking/libiproute/built-in.o networking/udhcp/built-in.o printutils/built-in.o procps/built-in.o runit/built-in.o selinux/built-in.o shell/built-in.o sysklogd/built-in.o util-linux/built-in.o util-linux/volume_id/built-in.o -Wl,--end-group
==========
during RTL pass: expand
networking/udhcp/dhcpc.c: In function 'd4_run_script':
networking/udhcp/dhcpc.c:559:13: internal compiler error: in expand_fn_using_insn, at internal-fn.cc:155
  559 | static void d4_run_script(struct dhcp_packet *packet, const char *name)
      |             ^
0x7a84aae2a1c9 __libc_start_call_main
	../sysdeps/nptl/libc_start_call_main.h:58
0x7a84aae2a28a __libc_start_main_impl
	../csu/libc-start.c:360
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <http://bugs.openwrt.org/> for instructions.
make[5]: *** [/home/daniel/openwrt/op/er4_6.6/tmp/ccYwUaCb.mk:8: /home/daniel/openwrt/op/er4_6.6/tmp/ccH10n9W.ltrans2.ltrans.o] Error 1
make[5]: *** Waiting for unfinished jobs....
during RTL pass: expand
networking/libiproute/utils.c: In function 'get_prefix':
networking/libiproute/utils.c:211:16: internal compiler error: in expand_fn_using_insn, at internal-fn.cc:155
  211 | void FAST_FUNC get_prefix(inet_prefix *dst, char *arg, int family)
      |                ^
0x74bce7a2a1c9 __libc_start_call_main
	../sysdeps/nptl/libc_start_call_main.h:58
0x74bce7a2a28a __libc_start_main_impl
	../csu/libc-start.c:360
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <http://bugs.openwrt.org/> for instructions.
make[5]: *** [/home/daniel/openwrt/op/er4_6.6/tmp/ccYwUaCb.mk:5: /home/daniel/openwrt/op/er4_6.6/tmp/ccH10n9W.ltrans1.ltrans.o] Error 1
Assembler messages:
Warning: a different -march was already specified, is now mips64
Assembler messages:
Warning: a different -march was already specified, is now mips64
Assembler messages:
Warning: a different -march was already specified, is now mips64
Assembler messages:
Warning: a different -march was already specified, is now mips64
Assembler messages:
Warning: a different -march was already specified, is now mips64
Assembler messages:
Warning: a different -march was already specified, is now mips64
lto-wrapper: fatal error: make returned 2 exit status
compilation terminated.
/home/daniel/openwrt/op/er4_6.6/staging_dir/toolchain-mips64_octeonplus_64_gcc-13.3.0_musl/lib/gcc/mips64-openwrt-linux-musl/13.3.0/../../../../mips64-openwrt-linux-musl/bin/ld.bfd: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
Note: if build needs additional libraries, put them in CONFIG_EXTRA_LDLIBS.
Example: CONFIG_EXTRA_LDLIBS="pthread dl tirpc audit pam"
make[4]: *** [Makefile:719: busybox_unstripped] Error 1
make[4]: Leaving directory '/home/daniel/openwrt/op/er4_6.6/build_dir/target-mips64_octeonplus_64_musl/busybox-default/busybox-1.37.0'
make[3]: *** [Makefile:168: /home/daniel/openwrt/op/er4_6.6/build_dir/target-mips64_octeonplus_64_musl/busybox-default/busybox-1.37.0/.built] Error 2
make[3]: Leaving directory '/home/daniel/openwrt/op/er4_6.6/package/utils/busybox'
time: package/utils/busybox/default/compile#24.03#4.07#21.68
    ERROR: package/utils/busybox failed to build (build variant: default).
make[2]: *** [package/Makefile:185: package/utils/busybox/compile] Error 1
make[2]: Leaving directory '/home/daniel/openwrt/op/er4_6.6'
make[1]: *** [package/Makefile:179: /home/daniel/openwrt/op/er4_6.6/staging_dir/target-mips64_octeonplus_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/daniel/openwrt/op/er4_6.6'
make: *** [/home/daniel/openwrt/op/er4_6.6/include/toplevel.mk:233: world] Error 2
daniel@daniel-ProBook:~/openwrt/op/er4_6.6$ ```

### Expected behaviour

A finished working build, the exact same source code with the same package setup diffconfig works with wrt3200acm and both these devices was built without problem before busybox 1.37.

### Diffconfig

```text
CONFIG_TARGET_octeon=y
CONFIG_TARGET_octeon_generic=y
CONFIG_TARGET_octeon_generic_DEVICE_ubnt_edgerouter-4=y
#
# General build settings:
CONFIG_TESTING_KERNEL=n

# Busybox settings:
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SYSLOG_INFO=y
CONFIG_BUSYBOX_CONFIG_FLOAT_DURATION=y
#
# General tools:
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_nano-full=y
#
# General packages:
CONFIG_PACKAGE_adblock=y
CONFIG_PACKAGE_ethtool=n
CONFIG_PACKAGE_ethtool-full=y
CONFIG_PACKAGE_openvpn-openssl=y
CONFIG_PACKAGE_sqm-scripts=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_apk-mbedtls=n
CONFIG_PACKAGE_apk-openssl=y
#CONFIG_PACKAGE_banip=y
#
# USB flash support:
CONFIG_PACKAGE_kmod-usb-storage-uas=y
CONFIG_PACKAGE_kmod-usb3=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_f2fs-tools=y
CONFIG_PACKAGE_kmod-fs-f2fs=y
#
# Remove pppoe:
CONFIG_PACKAGE_ppp=n
CONFIG_PACKAGE_ppp-mod-pppoe=n
#
# IPv6 packages:
CONFIG_PACKAGE_odhcpd-ipv6only=n
#
#DHCP
CONFIG_PACKAGE_dnsmasq=n
CONFIG_PACKAGE_dnsmasq-full=y
CONFIG_PACKAGE_dnsmasq_full_auth=y
CONFIG_PACKAGE_dnsmasq_full_conntrack=y
CONFIG_PACKAGE_dnsmasq_full_dhcp=y
CONFIG_PACKAGE_dnsmasq_full_dnssec=n
CONFIG_PACKAGE_dnsmasq_full_nftset=y
CONFIG_PACKAGE_dnsmasq_full_noid=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_dnsmasq_full_tftp=n
#
# Add LuCI custom w/o pppoe:
CONFIG_PACKAGE_luci=n
CONFIG_PACKAGE_libiwinfo-lua=n
CONFIG_PACKAGE_luci-app-adblock=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-app-sqm=y
CONFIG_PACKAGE_luci-app-uhttpd=y
#CONFIG_PACKAGE_luci-app-banip=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-theme-bootstrap=n
CONFIG_PACKAGE_luci-theme-openwrt-2020=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-wireguard=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
#
# Luci-ssl custom:
CONFIG_PACKAGE_libustream-openssl=y
CONFIG_PACKAGE_openssl-util=y
# Luci-ssl inactivate WolfSSL:
CONFIG_PACKAGE_libustream-wolfssl=n
CONFIG_PACKAGE_px5g-wolfssl=n
# Luci-ssl inactivate mbedTLS:
CONFIG_PACKAGE_libustream-mbedtls=n
CONFIG_PACKAGE_px5g-mbedtls=n
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-12-08:
Well, interesting thing is that using OpenWrt toolchain on current busybox git with OpenWrt config actually compiles.

robimarko, 2024-12-08:
I bisected this down to:
https://git.busybox.net/busybox/commit/?id=2075553a1b0eb843d4ee73a039b3e30bed9274e0

But you need this as well:
https://git.busybox.net/busybox/commit/?id=41b47398f52c6eb8ef2a061127bc36c54e3cc372

As previous commit broke compilation.

But this only somehow happens in OpenWrt.

robimarko, 2024-12-08:
It seems that something is overriding the passed `-march=octeon+` to `mips64` (Not `mips64r2` that we need) and this breaks compilation.

If I switch the octeon target `CPU_TYPE` to `mips64r2` then it magically compiles and doesn't have the assembler warning that its overrriden

robimarko, 2024-12-08:
Well, I have kind of hit a dead end, looks like a possible bug in the compiler

flygarn12, 2024-12-08:
> Well, I have kind of hit a dead end, looks like a possible bug in the compiler

Bug in the compiler, where did that come from? And how do we unbug it?

robimarko, 2024-12-08:
I guess the bug has been always here, but it got exposed by:
https://git.busybox.net/busybox/commit/?id=2075553a1b0eb843d4ee73a039b3e30bed9274e0

Sadly, I dont know how to fix it other than passing `-march=mips64r2`, I dont like that assembler warning but I dont see why its overriding octeon+ which is mips64r2 to mips64

flygarn12, 2024-12-09:
Is it the march in kernel menuconfig you change?

Can't we change it to mips64r2 permanently if it doesn't work and is supposed to be that anyway?

flygarn12, 2024-12-09:
[image]
Can it be a problem when the config file have Y for both mips alternatives, that busybox only take the first one?

robimarko, 2024-12-09:
I dont see how kernel config can affect this.

@hauke Any ideas

commit e6ce868c (2024-12-28): busybox: fix LTO compiler flags

When doing LTO builds, the target related CFLAGS need to be passed to the
linker, so that they are considered for target code generation.
Pass TARGET_CFLAGS in EXTRA_LDFLAGS to ensure that this is handled properly.

Fixes: #17200
commit 9a0a05d5 (2024-12-28): busybox: fix LTO compiler flags

When doing LTO builds, the target related CFLAGS need to be passed to the
linker, so that they are considered for target code generation.
Pass TARGET_CFLAGS in EXTRA_LDFLAGS to ensure that this is handled properly.

Fixes: #17200

(cherry picked from commit e6ce868c3a6f843f2c14498d0174011c8713a8bd)
closed 2024-12-28 by openwrt-bot