openwrt/openwrt #17204: latest `ucode`  uses `CONFIG_WIFI_SCRIPTS_UCODE` make wireless cannot be started.

zxlhhyccc, 2024-12-08:
### Describe the bug

````
Sat Dec  7 23:01:18 2024 daemon.notice netifd: radio0 (4422): In device_capabilities(), file /usr/share/ucode/wifi/hostapd.uc, line 444, byte 19:
Sat Dec  7 23:01:18 2024 daemon.notice netifd: radio0 (4422):   called from function generate (/usr/share/ucode/wifi/hostapd.uc:470:32)
Sat Dec  7 23:01:18 2024 daemon.notice netifd: radio0 (4422):   called from function setup (/usr/share/ucode/wifi/hostapd.uc:544:22)
Sat Dec  7 23:01:18 2024 daemon.notice netifd: radio0 (4422):   called from function setup (./mac80211.sh:307:21)
Sat Dec  7 23:01:18 2024 daemon.notice netifd: radio0 (4422):   called from anonymous function (./mac80211.sh:342:14)
Sat Dec  7 23:01:18 2024 daemon.notice netifd: radio0 (4422):
Sat Dec  7 23:01:18 2024 daemon.notice netifd: radio0 (4422):  `        if (!phy || phy.wiphy != idx)`
Sat Dec  7 23:01:18 2024 daemon.notice netifd: radio0 (4422):   Near here --------------^
`````

### OpenWrt version

r28266-9c55866fad

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Redmi AX6

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-12-08:
Can you please provide your wireless config as well as iw dev and iw phy outputs

zxlhhyccc, 2024-12-08:
> Can you please provide your wireless config as well as iw dev and iw phy outputs

[image]
````
root@OpenWrt:~# iw phy
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 16
        max scan IEs length: 394 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Capabilities: 0x19e7
                        RX LDPC
                        HT20/HT40
                        Dynamic SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                VHT Capabilities (0x33993992):
                        Max MPDU length: 11454
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        TX STBC
                        SU Beamformer
                        SU Beamformee
                        MU Beamformer
                        MU Beamformee
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                VHT extended NSS: not supported
                HE Iftypes: managed
[... 517 lines cut ...]
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Maximum associated stations in AP mode: 128
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ ENABLE_FTM_RESPONDER ]: enable FTM (Fine Time Measurement) r                                                                                                                     esponder
                * [ STA_TX_PWR ]: TX power control per station
                * [ VLAN_OFFLOAD ]: VLAN offload support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ BSS_COLOR ]: BSS coloring support
                * [ POWERED_ADDR_CHANGE ]: can change MAC address while up
`````

closed 2024-12-08 by zxlhhyccc