openwrt/openwrt #17207: [24.10.0-rc2] performance problem on 10gb port

MichelHanzen, 2024-12-08:
### Describe the bug

Tests performed on fiber 10gb connection:
2 x10Gtek for Ubiquiti UniFi SFP+ Multimode Transceiver UF-MM-10G, 10G SFP+ SR Fibre Module, Dual LC Connector, 850nm
Fiber Patch Cable - LC to LC OM3 10Gb/Gigabit]
With openwrt 24.10-RC2
[image]

With the openwrt image available on the bpi-r4 site (BPI-R4-BE1350-WIFI_MP4_0-SDK-20240620)
[image]

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

. /etc/openwrt_release && echo $DISTRIB_RELEASE

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

Run iperf3 on the two images

### Actual behaviour

speed below 5Gbites/sec

### Expected behaviour

speed should be near 10 Gbits/sec

### Additional info

Link to the bpi-r4 image:
https://drive.google.com/file/d/1U0tnvoaxDujO1q5QNxnw0i2d93X2zi9r/view?usp=sharing

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MichelHanzen, 2024-12-08:
Report updated. Sorry.

MichelHanzen, 2024-12-09:
Thé tests I performed were made between my PC and the router on the LAN. Is bridger helpful in this case? If yes, how do I have to implement it? I guess "opkg install bridger" is not enough.

RomanovDanila, 2024-12-09:
RSS and LRO aren't implemented in the mainline linux. Please, re-read the message above.

MichelHanzen, 2024-12-09:
You don't answer may question and I don't understand your answer. I will close the issue.

closed 2024-12-09 by MichelHanzen