openwrt/openwrt #17215: [23.05-SNAPSHOT] mac80211 issue stops wifi on TP-Link Archer AX23 v1 [ramips/mt7621]

48mWsPK, 2024-12-09:
### Describe the bug

5 GHz WiFi stopped working after 9 days of uptime. 

The device is a TP-Link Archer AX23 v1 that was updated almost weekly with a handpicked build of 23.05-SNAPSHOT.

This was found in the kernel log:

`[524751.744956] Ignoring NSS change in VHT Operating Mode Notification from e8:48:b8:5e:79:b0 with invalid nss 2
[524751.824920] Ignoring NSS change in VHT Operating Mode Notification from e8:48:b8:5e:79:b0 with invalid nss 2
[623194.322261] mt7915e 0000:02:00.0: Message 000025ed (seq 9) timeout
[623214.802035] mt7915e 0000:02:00.0: Message 0000aded (seq 10) timeout
[623235.281830] mt7915e 0000:02:00.0: Message 00005aed (seq 11) timeout
[623255.761582] mt7915e 0000:02:00.0: Message 00005aed (seq 12) timeout
[623276.241343] mt7915e 0000:02:00.0: Message 00005aed (seq 13) timeout
[623296.721151] mt7915e 0000:02:00.0: Message 00005aed (seq 14) timeout
[623317.200944] mt7915e 0000:02:00.0: Message 00005aed (seq 15) timeout
[623337.680696] mt7915e 0000:02:00.0: Message 00005aed (seq 1) timeout
[623358.160480] mt7915e 0000:02:00.0: Message 000025ed (seq 2) timeout
[623358.166806] ------------[ cut here ]------------
[623358.171557] WARNING: CPU: 2 PID: 23471 at backports-6.1.110-1/net/mac80211/agg-tx.c:407 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623358.183106] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat sha512_generic seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[623358.242365] CPU: 2 PID: 23471 Comm: kworker/u8:3 Not tainted 5.15.173 #0
[623358.249153] Workqueue: phy1 0x83314160 [mac80211@d1ef510f+0x8e1e0]
[623358.255424] Stack : 8081ce7b 800802fc 84886268 80083e00 00000000 00000000 00000000 00000000
[623358.263864]         00000000 00000000 00000000 00000000 00000000 00000001 82c0fc98 80c78620
[623358.272300]         82c0fd30 00000000 00000000 82c0fae0 00000038 803a2024 ffffffea 00000000
[623358.280738]         82c0faec 0000015b 80822ab0 ffffffff 80740dd4 82c0fc78 00000000 83315af4
[623358.289176]         00000009 84886050 83200e78 00000000 00000000 804140ec 00000008 809e0008
[623358.297613]         ...
[623358.300136] Call Trace:
[623358.300145] [<800802fc>] 0x800802fc
[623358.306216] [<80083e00>] 0x80083e00
[623358.309791] [<803a2024>] 0x803a2024
[623358.313355] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623358.319345] [<804140ec>] 0x804140ec
[623358.322926] [<80007898>] 0x80007898
[623358.326509] [<800078a0>] 0x800078a0
[623358.330069] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623358.336060] [<803860e4>] 0x803860e4
[623358.339620] [<8002deec>] 0x8002deec
[623358.343184] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623358.349172] [<8002dfd0>] 0x8002dfd0
[623358.352740] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623358.358736] [<8331452c>] 0x8331452c [mac80211@d1ef510f+0x8e1e0]
[623358.364723] [<80723534>] 0x80723534
[623358.368291] [<80048ec8>] 0x80048ec8
[623358.371854] [<80049264>] 0x80049264
[623358.375413] [<8072274c>] 0x8072274c
[623358.378974] [<80049108>] 0x80049108
[623358.382532] [<80051d34>] 0x80051d34
[623358.386091] [<80051bf4>] 0x80051bf4
[623358.389651] [<80051bf4>] 0x80051bf4
[623358.393211] [<80051bf4>] 0x80051bf4
[623358.396768] [<80002f98>] 0x80002f98
[623358.400341] 
[623358.402004] ---[ end trace 3a15684792347daf ]---
[623378.640281] mt7915e 0000:02:00.0: Message 00005aed (seq 3) timeout
[623399.120118] mt7915e 0000:02:00.0: Message 000025ed (seq 4) timeout
[623399.126441] ------------[ cut here ]------------
[623399.131198] WARNING: CPU: 2 PID: 23471 at backports-6.1.110-1/net/mac80211/agg-tx.c:407 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623399.142744] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat sha512_generic seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[623399.202022] CPU: 2 PID: 23471 Comm: kworker/u8:3 Tainted: G        W         5.15.173 #0
[623399.210195] Workqueue: phy1 0x83314160 [mac80211@d1ef510f+0x8e1e0]
[623399.216469] Stack : 8081ce7b 800802fc 819c0268 80083e00 00000000 00000000 00000000 00000000
[623399.224909]         00000000 00000000 00000000 00000000 00000000 00000001 82c0fc98 80c78620
[623399.233347]         82c0fd30 00000000 00000000 82c0fae0 00000038 803a2024 ffffffea 00000000
[623399.241784]         82c0faec 00000183 80822ab0 ffffffff 80740dd4 82c0fc78 00000000 83315af4
[623399.250223]         00000009 819c0050 83200e78 00000000 00000000 804140ec 00000008 809e0008
[623399.258661]         ...
[623399.261184] Call Trace:
[623399.261193] [<800802fc>] 0x800802fc
[623399.267273] [<80083e00>] 0x80083e00
[623399.270843] [<803a2024>] 0x803a2024
[623399.274411] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623399.280424] [<804140ec>] 0x804140ec
[623399.284018] [<80007898>] 0x80007898
[623399.287577] [<800078a0>] 0x800078a0
[623399.291140] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623399.297141] [<803860e4>] 0x803860e4
[623399.300704] [<8002deec>] 0x8002deec
[623399.304271] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623399.310261] [<8002dfd0>] 0x8002dfd0
[623399.313837] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[623399.319840] [<8331452c>] 0x8331452c [mac80211@d1ef510f+0x8e1e0]
[623399.325837] [<8331d1ec>] 0x8331d1ec [mac80211@d1ef510f+0x8e1e0]
[623399.331864] [<80048ec8>] 0x80048ec8
[623399.335437] [<80049264>] 0x80049264
[623399.338996] [<8072274c>] 0x8072274c
[623399.342556] [<80049108>] 0x80049108
[623399.346115] [<80051d34>] 0x80051d34
[623399.349674] [<80051bf4>] 0x80051bf4
[623399.353241] [<80051bf4>] 0x80051bf4
[623399.356810] [<80051bf4>] 0x80051bf4
[623399.360367] [<80002f98>] 0x80002f98
[623399.363939] 
[623399.365822] ---[ end trace 3a15684792347db0 ]---
[623419.599838] mt7915e 0000:02:00.0: Message 00005aed (seq 5) timeout
[623440.079677] mt7915e 0000:02:00.0: Message 00005aed (seq 6) timeout`

A couple of days prior, there were messages about high memory usage from the installed zabbix agent.

### OpenWrt version

r24143-a281c02288

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer AX23 v1

### Image kind

Official downloaded image

### Steps to reproduce

Install 23.05-SNAPSHOT r24143-a281c02288 on a TP-Link Archer AX23 v1, and use it normally. Same config used to work for months on the stable 23.05.4 and prior releases.

### Actual behaviour

A week or so after use the wifi card becomes unavailable, so router cannot be AP any longer.

### Expected behaviour

Work stable for long periods of time.

### Additional info

Device has zabbix-agentd - 6.4.7-1 installed and configured to report to a LAN accessible zabbix server.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

48mWsPK, 2024-12-09:
Attaching the kernel log - seems the CODE section does not respect new lines properly..

[OpenWRT 23.05-SNAPSHOT-r24143-a281c02288 WiFi crash report.txt](https://github.com/user-attachments/files/18058733/OpenWRT.23.05-SNAPSHOT-r24143-a281c02288.WiFi.crash.report.txt)

48mWsPK, 2024-12-09:
```
[628694.614538] ------------[ cut here ]------------
[628694.619262] WARNING: CPU: 3 PID: 1391 at backports-6.1.110-1/net/mac80211/agg-tx.c:407 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[628694.630711] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat sha512_generic seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[628694.690150] CPU: 3 PID: 1391 Comm: hostapd Tainted: G        W         5.15.173 #0
[628694.697800] Stack : 83390e78 80c26064 84027ee8 80083e00 00000000 00000000 00000000 00000000
[628694.706244]         00000000 00000000 00000000 00000000 00000000 00000001 84027970 80c7b720
[628694.714681]         84027a08 00000000 00000000 84027818 00000038 803a2024 ffffffea 00000000
[628694.723119]         84027824 000002ab 80822ab0 ffffffff 80740dd4 84027950 00000000 83315af4
[628694.731558]         00000009 860e6050 83390e78 80c26064 00000000 804140ec 0000000c 809e000c
[628694.739996]         ...
[628694.742519] Call Trace:
[628694.742549] [<80083e00>] 0x80083e00
[628694.748631] [<803a2024>] 0x803a2024
[628694.752198] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[628694.758194] [<804140ec>] 0x804140ec
[628694.761759] [<80007898>] 0x80007898
[628694.765318] [<800078a0>] 0x800078a0
[628694.768877] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[628694.774900] [<803860e4>] 0x803860e4
[628694.778524] [<8002deec>] 0x8002deec
[628694.782096] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[628694.788094] [<8002dfd0>] 0x8002dfd0
[628694.791663] [<83315af4>] 0x83315af4 [mac80211@d1ef510f+0x8e1e0]
[628694.797650] [<805a2144>] 0x805a2144
[628694.801213] [<83314090>] 0x83314090 [mac80211@d1ef510f+0x8e1e0]
[628694.807217] [<83307254>] 0x83307254 [mac80211@d1ef510f+0x8e1e0]
[628694.813220] [<8330bb54>] 0x8330bb54 [mac80211@d1ef510f+0x8e1e0]
[628694.819216] [<83294450>] 0x83294450 [cfg80211@80790743+0x45850]
[628694.825211] [<805a6dc4>] 0x805a6dc4
[628694.828770] [<8050013c>] 0x8050013c
[628694.832330] [<805a4f44>] 0x805a4f44
[628694.835891] [<805a2144>] 0x805a2144
[628694.839450] [<8329437c>] 0x8329437c [cfg80211@80790743+0x45850]
[628694.845456] [<805a6c70>] 0x805a6c70
[628694.849014] [<805a5c4c>] 0x805a5c4c
[628694.852580] [<805a65fc>] 0x805a65fc
[628694.856140] [<805a51fc>] 0x805a51fc
[628694.859697] [<805a5618>] 0x805a5618
[628694.863255] [<80502dd4>] 0x80502dd4
[628694.866813] [<801ba87c>] 0x801ba87c
[628694.870380] [<805a56a8>] 0x805a56a8
[628694.873941] [<80077178>] 0x80077178
[628694.877499] [<8050544c>] 0x8050544c
[628694.881066] [<805a5338>] 0x805a5338
[628694.884655] [<804f5d20>] 0x804f5d20
[628694.888292] [<8032b828>] 0x8032b828
[628694.891870] [<804f7c6c>] 0x804f7c6c
[628694.895437] [<8032b828>] 0x8032b828
[628694.899003] [<8021e2b0>] 0x8021e2b0
[628694.902577] [<804f7de8>] 0x804f7de8
[628694.906135] [<804f47e4>] 0x804f47e4
[628694.909703] [<8001456c>] 0x8001456c
[628694.913275] 
[628694.914908] ---[ end trace 3a15684792347db1 ]---
```

lorand-horvath, 2024-12-10:
I'd recommend using the latest snapshot, I think there have been Mediatek-related updates...

48mWsPK, 2024-12-15:
> I'd recommend using the latest snapshot, there have been many Mediatek-related updates... https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ramips%2Fmt7621&id=tplink_archer-ax23-v1 Click on customize, then request build.

All I can say is - [Current stable series: OpenWrt 23.05](https://openwrt.org/#current_stable_seriesopenwrt_2305)

The changes I am using on top of the last stable release (23.0.5.5) are NOT [that many](https://git.openwrt.org/?p=openwrt/openwrt.git;a=shortlog;h=refs/heads/openwrt-23.05); 39 commits in total since 23.05.5, and up until kernel 5.15.171 everything worked fine.

Problems above for me started started at [r24143-a281c02288](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=a281c022883937e6eec27b7811bcde55d5917d63).

Last tested working for me with zero problems was [r24138-cd9998ef1b](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=cd9998ef1b7b842bf052fc5e7034f23a22e67bf7), that was kernel 5.15.171. Something changed after that one that caused things to brake.

So for me the problem is somewhere in the last 9 commits on the 23.05 branch, I gues one of the 2 next kernel updates.

I would recommend using anything after properly testing - this is why I test on less important machines first, and when happy move to new releases. And I am testing 24.10. But I am expecting 23.05 to stay stable, as it is still the named "stable" release.

This here is a report of the "stable" release breaking enough so that a router comletly loses its WiFi capability. I am happy OpenWRT stayed alive long enough for me even broken to allow me to roll back to a known stable release - thanks, devs!

Cheers!

lorand-horvath, 2024-12-15:
@48mWsPK I understand your point, but the issue with the 23.05 branch is it didn't support AX WiFi very well and it's also an old kernel 5.15. The newer branch 24.10 is kernel 6.6, which doesn't need so many backports as 5.15 and is more performant, plus has better AX support. As for being stable, honestly for me the latest SNAPSHOT 28330 is more stable than any build I had on 23.05.

If you are wondering about the difference between 24.10 and SNAPSHOT: SNAPSHOT has a newer package manager `apk`, instead of the older `opkg` in 24.10. So there might be some incompatibilities regarding it with SNAPSHOT builds.

I'm not a dev, just wanted to help with your AX23 v1 device, which I'm in fact considering to purchase.
Cheers!

48mWsPK, 2024-12-15:
> If you are wondering about the difference between 24.10 and SNAPSHOT: SNAPSHOT has a newer package manager `apk`, instead of the older `opkg` in 24.10. So there might be some incompatibilities regarding it with SNAPSHOT builds.
> 
> I'm not a dev, just wanted to help with your AX23 v1 unit, which I'm in fact considering to purchase. Cheers!

Thanks for chiming in, but if you don't have anything on the subject of the issue reported - I don't see how that helps.

And I guess you missed it:
> During the last OpenWrt developer meeting 2024-11-26 it was decided,
that 24.10 will be the last release using opkg package manager and thus
apk won't be supported, so make it explicit and mark apk package manager
as broken.

Just before the rc1 release was tagged. Source: [git.openwrt.org](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=17bdccb4a52049ee06142c0b5fbc2001bc519a3a)

lorand-horvath, 2024-12-16:
> 
> And I guess you missed it:
> 
> > During the last OpenWrt developer meeting 2024-11-26 it was decided,
> > that 24.10 will be the last release using opkg package manager and thus
> > apk won't be supported, so make it explicit and mark apk package manager
> > as broken.
> 

No, I didn't miss it, read again what I wrote before. And `apk` is supported, but I guess you didn't get the memo.

48mWsPK, 2024-12-16:
> ... I'm in fact considering to purchase.

So the experience you talk from is from what device if you are "considering" to purchase an AX23 v1?

Also, from the quote above: 
>for 24.10 apk won't be supported

lorand-horvath, 2024-12-16:
Alright, I apologize for trying to help. You obviously don't need it.
Regarding my experience... well, you have no idea and let's leave it at that ;-)
Cheers!

kamazee, 2025-02-02:
I experienced a similar issue on `r27341-0db6f9a856` (which is, I assume a snapshot of 24.10 from September). Wi-Fi stopped working at some point. I didn't collect logs, though.
Is there anything I can collect to help next time it happens? Just post the output of `dmesg`?