openwrt/openwrt #17216: login page not displayed after boot or reboot

systemdlete, 2024-12-09:
### Describe the bug

After an initial boot or a reboot of the router, the browser does not display the login page.   It hangs for a long time with a message indicating that the browser is waiting, then the message changes to indicate that the browser is _still_ waiting, and then after some time the page times out altogether. 

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

x86/64

### Device

virtualbox VM

### Image kind

Official downloaded image

### Steps to reproduce

Boot or reboot the router.
In a browser, open the URL to the router (or leave the previous tab/page open on the URL).
Wait for the browser tab to refresh and display the login page.

### Actual behaviour

Upon any boot or reboot of the router, the browser pointed at the URL for the router does not (eventually) display the openwrt login page.
The browser will wait for the boot to complete, displaying a message saying it is waiting.
After a minute or so, the message changes to a slightly updated message (highlighted in color) indicating that the page is still waiting for the router to finish its boot.
Finally after some period of time, the entire page times out, leaving the browser with its failure message indicating it could not reach the URL (with a control allowing for a page reload).
If browser loading is stopped manually during the load wait, or if the page is re-loaded using the control displayed if it has timed out, the login page will be displayed and normal login and operations may continue.

### Expected behaviour

Before openwrt version 23.05.5, a boot or reboot would always display the login page for openwrt, usually as soon as the router had fully completed its bootup.

### Additional info

Using firefox-esr 128.5.0 64bit running on a devuan daedalus system.
The router is virtualized under virtualbox 7.1.22.   There are other VMs running simultaneously on the same system as the openwrt router.  This scenario is consistent across several physical machines running the same configuration of Virtualbox and VMs.
Other than the missing login page, the router is extremely stable, fast, and reliable, and works without incident.
Behavior changed with openwrt version 23.05.5, and continues in the 24.10.0-rc versions.

This is a nuisance more than anything else, but seems noticeable enough to report it; it is disappointing that this was working until recent releases.

### Diffconfig

```text
(Image is downloaded from openwrt.org's download page; squashfs images are used to perform all upgrades.)
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rodsmar, 2024-12-09:
I think you should open this ticket in openwrt/luci

systemdlete, 2024-12-11:
Repo maintainer or other relevant role:  If you agree with rodsmar, please move this issue to openwrt/luci.

Thank you.

systemdlete, 2024-12-13:
@brada4, thank you but see comments above.  Rodsmar already noted this and I am waiting/hoping an admin can move this issue to the appropriate repo.

brada4, 2024-12-13:
This repo is not used for sorting issues.
Try to nor micro-manage organisation admins on your mistakes.