openwrt/openwrt #17222: [MT7621]  directory /sys/class/gpio/power_usb3 does not exists

maximst, 2024-12-10:
### Describe the bug

24.10.0-rc1
24.10.0-rc2
current snapshot
GPIO directory for managing usb power does not exists

### OpenWrt version

24.10.0-rc2

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

ramips/mt7621

### Device

D-Team Newifi D2

### Image kind

Official downloaded image

### Actual behaviour

```bash
# ls -la /sys/class/gpio
drwxr-xr-x    2 root     root             0 Jan  1  1970 .
drwxr-xr-x   38 root     root             0 Jan  1  1970 ..
--w-------    1 root     root          4096 Jan  1  1970 export
lrwxrwxrwx    1 root     root             0 Jan  1  1970 gpiochip512 -> ../../devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip512
lrwxrwxrwx    1 root     root             0 Jan  1  1970 gpiochip544 -> ../../devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip544
lrwxrwxrwx    1 root     root             0 Jan  1  1970 gpiochip576 -> ../../devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip576
--w-------    1 root     root          4096 Jan  1  1970 unexport
```

### Expected behaviour

Directory /sys/class/gpio/ should be contain power_usb3. This is must be a link to GPIO what used for enable/disable USB power. Like in 23.05

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2024-12-11:
That GPIO pin was converted to a fixed regulator, so it's not showed as a GPIO pin anymore.

ref: https://github.com/openwrt/openwrt/commit/e612900ae0fd6c4624fa661e004536216409107f

maximst, 2024-12-11:
Thanks. Please tell me, if not difficult, how i can disable/enable power now?

closed 2024-12-11 by maximst

DragonBluep, 2024-12-14:
try: `echo 1-1 > /sys/bus/usb/drivers/usb/unbind` and `echo 1-1 > /sys/bus/usb/drivers/usb/bind`.

maximst, 2024-12-27:
> try: `echo 1-1 > /sys/bus/usb/drivers/usb/unbind` and `echo 1-1 > /sys/bus/usb/drivers/usb/bind`.

I tried `1-0, 2-0, 1-1, 2-1` and get error "Unknown device"
When I try `usb1 or usb2` error does not appear but do nothing.
In `/sys/class/regulator` all files is read-only.
It turns out that now user cannot control USB port.
I had to change dts file and rebuild the firmware to get this functionality back.
I don't understand why removed some openwrt functionality? Now it's impossible to use an external drive normally. And what are the advantages of having a regulator device?

neheb, 2025-06-08:
Disable the USB port with uhubctl

maximst, 2025-09-26:
> ...
> cat /sys/kernel/debug/regulator/regulator_summary
> 
> is probably better.

```
# find /sys/kernel/debug/regulator | grep constraint_flags
/sys/kernel/debug/regulator/power_usb3/1e1c0000.xhci-vbus/constraint_flags
/sys/kernel/debug/regulator/reg-dummy-regulator-dummy/1e1c0000.xhci-vusb33/constraint_flags
# cat /sys/kernel/debug/regulator/power_usb3/1e1c0000.xhci-vbus/constraint_flags
always_on: 0
boot_on: 1
apply_uV: 0
ramp_disable: 0
soft_start: 0
pull_down: 0
over_current_protection: 0
# cat /sys/kernel/debug/regulator/reg-dummy-regulator-dummy/1e1c0000.xhci-vusb33/constraint_flags
always_on: 1
boot_on: 0
apply_uV: 0
ramp_disable: 0
soft_start: 0
pull_down: 0
over_current_protection: 0
# cat /sys/kernel/debug/regulator/regulator_summary
 regulator                      use open bypass  opmode voltage current     min     max
---------------------------------------------------------------------------------------
 regulator-dummy                  2    1      0 unknown     0mV     0mA     0mV     0mV 
    1e1c0000.xhci-vusb33          1                                 0mA     0mV     0mV
 mmc_io                           1    0      0 unknown  1800mV     0mA  1800mV  1800mV 
 mmc_power                        1    0      0 unknown  3300mV     0mA  3300mV  3300mV 
 power_usb3                       1    1      0 unknown  5000mV     0mA  5000mV  5000mV 
    1e1c0000.xhci-vbus            1                                 0mA     0mV     0mV
```

If I understand kernel documentation correctly, regulators can't be written to from userspace. Only kernel modules can control them. This makes sense for desktop, where power management is handled by the devices drivers. But I can't understand why this is necessary in routers?

neheb, 2025-09-26:
my understanding is turning off the USB device also turns off the regulator, or GPIO.

Maybe this will work. No idea.

```
--- a/target/linux/ramips/dts/mt7621_d-team_newifi-d2.dts
+++ b/target/linux/ramips/dts/mt7621_d-team_newifi-d2.dts
@@ -172,7 +172,7 @@
        nvmem-cell-names = "mac-address";
 };

-&reg_vbus {
+&reg_vbus33 {
        /delete-property/ regulator-always-on;
        gpios = <&gpio 11 GPIO_ACTIVE_HIGH>;
        enable-active-high;
```

FredApps, 2026-09-03:
Data point from the same target, in case it helps narrow this down: `gpio-export` does still create the named links on mt7621. This is an ASUS 4G-AX56 on 25.12-SNAPSHOT r0-b6a820e, kernel 6.12.103:

```
# ls -l /sys/class/gpio/
--w-------    1 root     root  4096 export
lrwxrwxrwx    1 root     root     0 gpiochip512 -> ../../devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip512
lrwxrwxrwx    1 root     root     0 gpiochip544 -> ../../devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip544
lrwxrwxrwx    1 root     root     0 gpiochip576 -> ../../devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip576
lrwxrwxrwx    1 root     root     0 reboot:modem -> ../../devices/platform/1e000000.palmbus/1e000600.gpio/gpiochip0/gpio/reboot:modem
lrwxrwxrwx    1 root     root     0 reset:modem -> ../../devices/platform/1e000000.palmbus/1e000600.gpio/gpiochip0/gpio/reset:modem
--w-------    1 root     root  4096 unexport
```

Same `gpiochip512/544/576` numbering as your "actual behaviour" output, so the base numbering change is not what is breaking it — the mechanism itself is intact on current mt7621. The two named entries come from a `gpio-export` node in the board DTS.

So this looks board-side rather than target- or kernel-side, which is consistent with the DTS patch suggested above.

neheb, 2026-09-03:
I wonder if GPIO_ACTIVE_HIGH should be GPIO_ACTIVE_LOW instead.

FredApps, 2026-09-04:
It would have no effect — for `regulator-fixed` the phandle flags are deliberately discarded. From `drivers/gpio/gpiolib-of.c`:

```c
	/*
	 * The regulator GPIO handles are specified such that the
	 * presence or absence of "enable-active-high" solely controls
	 * the polarity of the GPIO line. Any phandle flags must
	 * be actively ignored.
	 */
#if IS_ENABLED(CONFIG_REGULATOR_FIXED_VOLTAGE)
	{ "regulator-fixed",   "gpios",        "enable-active-high" },
```

`CONFIG_REGULATOR_FIXED_VOLTAGE=y` on ramips, so that quirk is live and `GPIO_ACTIVE_LOW` in the cell would be ignored. To invert the line, drop `enable-active-high` instead — the binding states that its absence means active low.

Two things worth noting for anyone retesting this:

- The node is already in main and 25.12 as `&reg_vbus`, with `enable-active-high` present. It is not in 24.10, so a 24.10 build will not show the behaviour at all.
- `reg_vbus33` does not exist; the 3.3V label in `mt7621.dtsi` is `reg_vusb33`, and it is the dummy always-on rail feeding `xhci-vusb33`. The GPIO-controlled 5V one is `reg_vbus` (`usb_power`), which matches the `power_usb3` entry in the `regulator_summary` above.

Source-derived rather than measured — my board has no GPIO-controlled vbus regulator to test this on.