openwrt/openwrt #17224: TP-Link Archer AX23v1 (mt7621) has broken RGMII{1,2} muxing

ckiee, 2024-12-10:
### Describe the bug

Since https://github.com/openwrt/openwrt/pull/10238, RGMII muxing has been enabled for MT7621. 

AX23v1 is newer than this PR, but has also gotten it [theoretically enabled](https://github.com/openwrt/openwrt/blob/ff5f77ecdd218a9451bd6c531be86c6335fadd7a/target/linux/ramips/dts/mt7621_tplink_archer-ax23-v1.dts#L184). In reality, I still get only ~500/500Mbits on iperf3 bidirectional with another host in the network.

The pin assignments do not conflict with RGMII according to [the datasheet](https://www.t-firefly.com/download/FireWRT/hardware/MT7621.pdf) so I think it should be working. I assume the DTS is wrong but haven't been able to find the bug.

cc @arinc9 

### OpenWrt version

r28266-9c55866fad (Also reprods on 23.05.4)

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer AX23v1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

arinc9, 2024-12-10:
There is not much I can do. The devicetree looks correct. Just make sure the test is done between the wan port and another port. My current suspicion is you're doing the test between interfaces that won't have both of the MACs used.

brada4, 2024-12-10:
Also you need tcp udp conntrack and hardware flow offload enabled to push 2gbps between soc cpu mii ports.or bridger if in same bridged subnet.

arinc9, 2024-12-13:
This is something that, it would either work or it just wouldn't. As you have network connectivity using the wan interface (I do not read you clarifying that you use the wan interface, therefore, I assume so), it is working.

ckiee, 2024-12-13:
Yup, I was connecting with the `wan` interface & PPPoE. So I think that part is good.

lorand-horvath, 2024-12-16:
@ckiee Did you enable HFO or at least SFO? As @brada4 mentioned above, flow offloading is required to reach gigabit speeds with PPPoE.

ckiee, 2025-01-01:
I reverted to the vendor firmware due to needing to use the connection. Might try again sometime but no promises