openwrt/openwrt #17232: Broken 2.5G WAN port on MT7981 - Cudy TR3000

samed, 2024-12-11:
### Describe the bug

WAN link does not get detected on SNAPSHOTS and RC

### OpenWrt version

r28217-af611bce44

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy TR3000 v1

### Image kind

Official downloaded image

### Steps to reproduce

Flash latest SNAPSHOT firmware from downloads.openwrt.org, do not preserve any settings
Watch syslog with logread -f
Connect something to WAN port.

### Actual behaviour

wan link is down

### Expected behaviour

wan link is up/autonegotiation

### Additional info

This was previously fixed in https://github.com/openwrt/openwrt/issues/16448 but for some reason broken with the symptoms.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

v4lkyr, 2024-12-12:
Same problem on Ruijie RG-X60 Pro too. It's the same 2.5G PHY chip - RTL8221B-VB-CG

Reported here: https://github.com/openwrt/openwrt/issues/17183

vincejv, 2024-12-14:
hello @v4lkyr @samed could this be related, can try to apply the workaround? Although it's a different chipset...
https://github.com/openwrt/openwrt/issues/17157#issuecomment-2530417714
https://github.com/openwrt/openwrt/issues/16604

samed, 2024-12-14:
@vincejv unfortunately that didn't help. 

But currently, there's a workaround to get the `wan` port working.

Steps:
* Boot tr3000 without the cable plugged in the wan port.
* Once booted, run `ethtool -s eth0 autoneg on speed 1000 duplex full` 
* Plug the cable on the wan port and surprisingly it works 🤷

samed, 2024-12-20:
Another workaround is to swap lan port with wan

LuisMitaHL, 2024-12-25:
The bug continues on 24.10.0-rc4

```
[   15.556699] RTL8221B-VB-CG 2.5Gbps PHY (C45) mdio-bus:01: rtl822xb_config_init failed: -110
[   15.675872] mtk_soc_eth 15100000.ethernet eth0: mtk_open: could not attach PHY: -110
```

ab77, 2025-01-08:
Also tried 23.05, WAN link is **up**, but no IP/DHCP until `/etc/init.d/network restart` is issued. The interface remains is some weird `pending` state according to `ifstat` until then.

I've tried patching `netifd` from [here](https://github.com/openwrt/netifd/commits/openwrt-23.05/) to latest with no effect:

```
$ find package/network/config/netifd/patches -name '*.patch' | sort
package/network/config/netifd/patches/100-cmake_bump-the-minimum-required-cmake-version-to-3.5.patch
package/network/config/netifd/patches/110-proto-shell_add-proto-property-for-skipping-device-config.patch
package/network/config/netifd/patches/120-wireless_use-blobmsg_parse_attr.patch
package/network/config/netifd/patches/130-wireless_add-support-for-defining-wifi-interfaces-via-procd.patch
package/network/config/netifd/patches/140-system-linux_re-apply-ethtool-on-phy-attachment.patch
package/network/config/netifd/patches/150-interface_add-support-for-disabling-renew-on-topology-change.patch
package/network/config/netifd/patches/160-device_add-more-debugging-code.patch
package/network/config/netifd/patches/170-device_do-not-pull-device-present-state-from-hotplug-events.patch
package/network/config/netifd/patches/180-wireless_handle-link-updates-even-if-devices-are-present-already.patch
package/network/config/netifd/patches/190-interface_improve-hotplug-handling-reliability.patch
package/network/config/netifd/patches/200-device_simplify-external-device-settings-handling.patch
package/network/config/netifd/patches/210-Revert-system-linux-re-apply-ethtool-on-phy-attachment.patch
package/network/config/netifd/patches/220-interface-fix-regression-on-adding-hotplug-devices-to.patch
package/network/config/netifd/patches/230-wireless-reload-wireless-device-if-any-vif-network-bridge.patch
package/network/config/netifd/patches/240-fix-signed-unsigned-comparison-warnings.patch
package/network/config/netifd/patches/250-system-dummy-add-system_if_apply_settings_after_up.patch
package/network/config/netifd/patches/260-ubus-add-notifications-on-wireless-device-state-changes.patch
package/network/config/netifd/patches/270-device-interface-add-tags-attribute-from-config-to-status.patch
package/network/config/netifd/patches/280-system-linux-do-not-pull-ifindex-flags-from-netlink-messages.patch
package/network/config/netifd/patches/290-proto-shell-fix-spurious-interface-teardowns-with-host-deps.patch
package/network/config/netifd/patches/300-iprule-add-ipproto-property.patch
package/network/config/netifd/patches/310-system-linux-restore-reading-flags-from-netlink-messages.patch
package/network/config/netifd/patches/320-interface-fix-memleak-and-reload-issue-for-the-zone.patch
package/network/config/netifd/patches/330-interface-remove-unnecessary-NULL-checks-before-free.patch
```

I added a helper script to run on boot to "fix" this, but it's obviously only a workaround:

```
#!/bin/ash

# wait here until the WAN interface appears
until ifstatus wan; do sleep $(( (RANDOM % 1) + 1)); done
sleep $(( (RANDOM % 5) + 1))

# .. then keep trying to get WAN IP via DHCP
until ifconfig wan | grep "inet addr"; do
    # https://udhcp.busybox.net/README.udhcpc
    killall -SIGUSR2 udhcpc  # first force inactive state
    killall -SIGUSR1 udhcpc  # .. then force renew
    sleep $(( (RANDOM % 5) + 1))

    # .. still no IP?
    if ! ifconfig wan | grep "inet addr"; then
        /sbin/uci commit network && /etc/init.d/network reload
        /etc/init.d/network restart  # restart the whole network stack
        sleep $(( (RANDOM % 5) + 1))
    fi
done
ifstatus wan
```

juampe, 2025-01-21:
FYI WR3000H https://github.com/openwrt/openwrt/pull/17458 and started to be supported in December SNAPSHOT and all tests for 2.5G wan port negotiation and link detection after testing with C45 and many DTS combinations inspired in WRT cudy devices (personally I didn't know the workaround ethtool) I managed to get it working with C22. Now I see that it was working before thanks to PR https://github.com/openwrt/openwrt/pull/17600 . Are we in the middle of a regression?

LuisMitaHL, 2025-01-26:
Testing 730db6b893748517949b1d3a0c0cb8b6c1288040 with Cudy TR3000 and a Gigabit switch on WAN:
Powering on the router without anything connected at boot: everything's normal. If I connect the switch the router can obtain DHCP normally.

<details><summary>Full log</summary>
<pre>
root@OpenWrt:~# logread
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Sun Jan 26 00:17:21 2025 kern.notice kernel: [    0.000000] Linux version 6.6.74 (looper@amelia) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r25978+1189-ea609fe486) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Sun Jan 26 00:17:11 2025
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Machine model: Cudy TR3000 v1
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000047c80000..0x0000000047d7ffff (1024 KiB) nomap non-reusable wmcpu-reserved@47c80000
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000047d80000..0x0000000047dbffff (256 KiB) nomap non-reusable wo-emi@47d80000
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000047dc0000..0x0000000047ffffff (2304 KiB) nomap non-reusable wo-data@47dc0000
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Zone ranges:
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000]   DMA32    empty
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000]   Normal   empty
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Early memory node ranges
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043030000-0x0000000047c7ffff]
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000047c80000-0x0000000047ffffff]
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000048000000-0x000000005fffffff]
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.2
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] percpu: Embedded 18 pages/cpu s35048 r8192 d30488 u73728
Sun Jan 26 00:17:21 2025 kern.debug kernel: [    0.000000] pcpu-alloc: s35048 r8192 d30488 u73728 alloc=18*4096
Sun Jan 26 00:17:21 2025 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] CPU features: detected: GIC system register CPU interface
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] alternatives: applying boot alternatives
Sun Jan 26 00:17:21 2025 kern.notice kernel: [    0.000000] Kernel command line:
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 0MB
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] software IO TLB: area num 2.
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] software IO TLB: mapped [mem 0x000000005fd68000-0x000000005fde8000] (0MB)
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] Memory: 496336K/524288K available (8832K kernel code, 994K rwdata, 2568K rodata, 448K init, 289K bss, 27952K reserved, 0K cma-reserved)
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun Jan 26 00:17:21 2025 kern.info kernel: [    0.000000] rcu:  RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[... cut ...]
        Supported FEC modes: Not reported
        Advertised link modes:  Not reported
        Advertised pause frame use: No
        Advertised auto-negotiation: No
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Half
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: off
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: no
</pre>
</details> 

@dangowrt

dangowrt, 2025-01-30:
@ab77 @LuisMitaHL please try #17784

closed 2025-01-30 by openwrt-bot

ab77, 2025-02-04:
@dangowrt if I understood correctly, https://github.com/openwrt/openwrt/pull/17784 only applies to OpwnWrt based on kernel-6.6 (i.e. 24.10)? If so, this issue also appears on 5.15 (openwrt-23.05). I'll try porting your patch(s)..

PR #17784 (2025-01-30): generic: net: phy: realtek: work-around hang on SerDes setup
On some but not all devices using the RTL8221B 2.5GBit/s PHY the SerDes setup sequence may hang under some circumstances (eg. <2500M link partner present during boot).

```
RTL8221B-VB-CG 2.5Gbps PHY (C45) mdio-bus:01: rtl822xb_config_init failed: -110
```

Work-around the issue by performing a hardware reset and subsequent retry of the SerDes setup, which seems to always succeed.

Doing this requires moving ALDPS setup to config_init (which is anyway the better place for that) as it otherwise doesn't survive the reset.

Also disable [... cut]