openwrt/openwrt #17234: [24.10-rc2][ath79][ubnt_lap-120]

jmspswny, 2024-12-11:
### Describe the bug

Just installed 24.10.0-rc2 on a Ubiquiti LAP-120. It seems that we have a regression around the SPI-NOR write protect defaults that these Ubiquiti WA devices need to not be stuck in unwritable state.

Here's the original PR that fixed this issue:
[https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=c55aaa7c9a98e7c0a5e1ea8293a534dc5b395cf3](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=c55aaa7c9a98e7c0a5e1ea8293a534dc5b395cf3)

config-default still has the changes made in that commit, as well as an additional SPI_NOR config - CONFIG_MTD_SPI_NOR_USE_VARIABLE_ERASE=y. Not sure if that is the culprit, but it is where I started digging.

### OpenWrt version

24.10-rc2

### OpenWrt release

24.10-rc2

### OpenWrt target/subtarget

ath79/generic

### Device

ubnt_lap-120

### Image kind

Official downloaded image

### Steps to reproduce

Install 24.10-rc2. After that, you're not able to save config or install a different image.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jmspswny, 2024-12-12:
@ynezz - is this really `invalid`? if not, can that `invalid` be removed and this issue be tagged as `release/24.10` and whatever else is appropriate?  Thanks.

jmspswny, 2024-12-12:
https://github.com/openwrt/openwrt/issues/14941
https://github.com/openwrt/openwrt/issues/12882
https://github.com/openwrt/openwrt/issues/13991
are existing issues, in various states, that seem to be related to this

aanon4, 2024-12-15:
I'll add that you can't write the flash of the Rocket AC Lite, the Bullet M XW nor the Rocket M XW. It does work on the Rocket M XM.

This problem came up in the 23.05 series also, but the fixes were merged into OpenWRT and are present in this build. They dont appear to help now.

I see the problem with two macronix chips - the mx25l12805d and the mx25l6405d

aanon4, 2024-12-16:
You'll find a fix here - https://github.com/openwrt/openwrt/issues/17285

hauke, 2025-09-02:
This fix was integrated in Linux kernel [6.12.44](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?h=v6.12.44&id=264e99d), [6.6.103](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?h=v6.6.103&id=b9e9afc) merged into main in b92bab633fe8cf3c3a72345c08d1ab6a8dd0abd2 and 1c92e468d544db98018c55d5d18e41fc1a3af20b and openwrt-24.10 in c676281e7e26c532b53908f84c59786fde189a76.