openwrt/openwrt #17236: WIFI MAC address H3C TX1801

huytran123, 2024-12-12:
### Describe the bug

Hi, I have read guide and installed OPENWRT on H3C TX1801 success from: https://github.com/openwrt/openwrt/commit/13308161788c98ae6cd48c22b13339fdb8c77130
But Mac address of wireless not normal, can you help me. I have installed OPENWRT for 4 H3C TX1801. But have two units the same Mac address. I do not why?!
[image]

### OpenWrt version

OpenWrt  23.05.5

### OpenWrt release

r24106-10cc5fcd00

### OpenWrt target/subtarget

ramips/mt7621

### Device

H3C TX1801Plus

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

huytran123, 2024-12-12:
> Invalid Release reported. `r24106-10cc5fcd00` Is this from a clean repository?

I have downloaded from https://firmware-selector.openwrt.org/?version=23.05.5&target=ramips%2Fmt7621&id=h3c_tx1801-plus

huytran123, 2024-12-12:
> I have downloaded from https://firmware-selector.openwrt.org/?version=23.05.5&target=ramips%2Fmt7621&id=h3c_tx1801-plus

I have downloaded from https://firmware-selector.openwrt.org/?version=23.05.5&target=ramips%2Fmt7621&id=h3c_tx1801-plus

huytran123, 2024-12-12:
Result when run your command :

00000000  f3 c0 f9 c5 62 61 75 64  72 61 74 65 3d 31 31 35  |....baudrate=115|
00000010  32 30 30 00 62 6f 6f 74  61 72 67 73 3d 63 6f 6e  |200.bootargs=con|
00000020  73 6f 6c 65 3d 74 74 79  53 30 2c 31 31 35 32 30  |sole=ttyS0,11520|
00000030  30 20 72 6f 6f 74 66 73  74 79 70 65 3d 73 71 75  |0 rootfstype=squ|
00000040  61 73 68 66 73 20 6c 6f  67 6c 65 76 65 6c 3d 33  |ashfs loglevel=3|
00000050  20 72 6f 6f 74 3d 2f 64  65 76 2f 6d 74 64 62 6c  | root=/dev/mtdbl|
00000060  6f 63 6b 31 30 00 62 6f  6f 74 63 6d 64 3d 6d 74  |ock10.bootcmd=mt|
00000070  6b 61 75 74 6f 62 6f 6f  74 00 62 6f 6f 74 64 65  |kautoboot.bootde|
00000080  6c 61 79 3d 30 00 62 6f  6f 74 66 69 6c 65 2e 62  |lay=0.bootfile.b|
00000090  6f 6f 74 6c 6f 61 64 65  72 3d 6d 74 64 31 5f 42  |ootloader=mtd1_B|
000000a0  6f 6f 74 6c 6f 61 64 65  72 2e 62 69 6e 00 62 6f  |ootloader.bin.bo|
000000b0  6f 74 66 69 6c 65 2e 66  69 72 6d 77 61 72 65 3d  |otfile.firmware=|
000000c0  6d 74 64 35 5f 66 69 72  6d 77 61 72 65 31 2e 62  |mtd5_firmware1.b|
000000d0  69 6e 00 62 6f 6f 74 66  6c 61 67 3d 30 00 62 6f  |in.bootflag=0.bo|
000000e0  6f 74 6d 65 6e 75 5f 30  3d 53 74 61 72 74 75 70  |otmenu_0=Startup|
000000f0  20 73 79 73 74 65 6d 20  28 44 65 66 61 75 6c 74  | system (Default|
00000100  29 3d 6d 74 6b 62 6f 61  72 64 62 6f 6f 74 00 62  |)=mtkboardboot.b|
00000110  6f 6f 74 6d 65 6e 75 5f  31 3d 55 70 67 72 61 64  |ootmenu_1=Upgrad|
00000120  65 20 66 69 72 6d 77 61  72 65 3d 6d 74 6b 75 70  |e firmware=mtkup|
00000130  67 72 61 64 65 20 66 77  00 62 6f 6f 74 6d 65 6e  |grade fw.bootmen|
00000140  75 5f 32 3d 55 70 67 72  61 64 65 20 46 61 63 74  |u_2=Upgrade Fact|
00000150  6f 72 79 3d 6d 74 6b 75  70 67 72 61 64 65 20 61  |ory=mtkupgrade a|
00000160  72 74 00 62 6f 6f 74 6d  65 6e 75 5f 33 3d 55 70  |rt.bootmenu_3=Up|
00000170  67 72 61 64 65 20 62 6f  6f 74 6c 6f 61 64 65 72  |grade bootloader|
00000180  3d 6d 74 6b 75 70 67 72  61 64 65 20 62 6c 00 62  |=mtkupgrade bl.b|
00000190  6f 6f 74 6d 65 6e 75 5f  34 3d 55 70 67 72 61 64  |ootmenu_4=Upgrad|
000001a0  65 20 62 6f 6f 74 6c 6f  61 64 65 72 20 28 61 64  |e bootloader (ad|
000001b0  76 61 6e 63 65 64 20 6d  6f 64 65 29 3d 6d 74 6b  |vanced mode)=mtk|
000001c0  75 70 67 72 61 64 65 20  62 6c 61 64 76 00 62 6f  |upgrade bladv.bo|
000001d0  6f 74 6d 65 6e 75 5f 35  3d 4c 6f 61 64 20 69 6d  |otmenu_5=Load im|
000001e0  61 67 65 3d 6d 74 6b 6c  6f 61 64 00 62 6f 6f 74  |age=mtkload.boot|
000001f0  76 65 72 73 69 6f 6e 3d  31 30 32 00 63 74 65 69  |version=102.ctei|
00000200  3d 31 38 34 38 39 39 37  39 31 32 36 33 35 31 33  |=184899791263513|
00000210  00 65 6c 69 6e 6b 73 6e  3d 39 39 43 46 30 30 41  |.elinksn=99CF00A|
00000220  30 30 31 37 36 45 46 33  37 39 43 35 30 30 30 00  |00176EF379C5000.|
00000230  65 74 68 61 63 74 3d 65  74 68 40 31 65 31 30 30  |ethact=eth@1e100|
00000240  30 30 30 00 65 74 68 61  64 64 72 3d 31 34 3a 35  |000.ethaddr=14:5|
00000250  31 3a 37 45 3a 44 37 3a  33 31 3a 30 33 00 66 63  |1:7E:D7:31:03.fc|
00000260  74 6c 6f 63 6b 66 6c 61  67 3d 31 00 66 64 74 63  |tlockflag=1.fdtc|
00000270  6f 6e 74 72 6f 6c 61 64  64 72 3d 38 66 66 66 31  |ontroladdr=8fff1|
00000280  32 64 30 00 66 69 6c 65  61 64 64 72 3d 38 30 30  |2d0.fileaddr=800|
00000290  31 30 30 30 30 00 66 69  6c 65 73 69 7a 65 3d 32  |10000.filesize=2|
000002a0  30 30 30 30 30 30 00 69  70 61 64 64 72 3d 31 39  |000000.ipaddr=19|
000002b0  32 2e 31 36 38 2e 31 32  34 2e 35 35 00 6d 61 6e  |2.168.124.55.man|
000002c0  75 63 6f 64 65 3d 32 31  39 38 30 31 41 32 48 31  |ucode=219801A2H1|
000002d0  50 32 31 34 30 38 30 33  4b 33 00 6d 74 64 64 65  |P2140803K3.mtdde|
000002e0  76 6e 61 6d 65 3d 75 2d  62 6f 6f 74 00 6d 74 64  |vname=u-boot.mtd|
000002f0  64 65 76 6e 75 6d 3d 30  00 6d 74 64 69 64 73 3d  |devnum=0.mtdids=|
00000300  6e 6d 62 6d 30 3d 6e 6d  62 6d 30 00 6d 74 64 70  |nmbm0=nmbm0.mtdp|
00000310  61 72 74 73 3d 6d 74 64  70 61 72 74 73 3d 6e 6d  |arts=mtdparts=nm|
00000320  62 6d 30 3a 35 31 32 6b  28 75 2d 62 6f 6f 74 29  |bm0:512k(u-boot)|
00000330  2c 35 31 32 6b 28 75 2d  62 6f 6f 74 2d 65 6e 76  |,512k(u-boot-env|
00000340  29 2c 35 31 32 4b 28 65  78 70 29 2c 35 31 32 6b  |),512K(exp),512k|
00000350  28 66 61 63 74 6f 72 79  29 2c 2d 28 66 69 72 6d  |(factory),-(firm|
00000360  77 61 72 65 29 00 6e 65  74 6d 61 73 6b 3d 32 35  |ware).netmask=25|
00000370  35 2e 32 35 35 2e 32 35  35 2e 30 00 70 61 72 74  |5.255.255.0.part|
00000380  69 74 69 6f 6e 3d 6e 6d  62 6d 30 2c 30 00 70 72  |ition=nmbm0,0.pr|
00000390  6f 64 75 63 74 6e 61 6d  65 3d 54 58 31 38 30 31  |oductname=TX1801|
000003a0  5f 50 6c 75 73 5f 43 00  72 65 73 74 6f 72 65 5f  |_Plus_C.restore_|
000003b0  64 65 66 61 75 6c 74 3d  30 00 73 65 72 76 65 72  |default=0.server|
000003c0  69 70 3d 31 39 32 2e 31  36 38 2e 31 32 34 2e 39  |ip=192.168.124.9|
000003d0  39 00 73 74 64 65 72 72  3d 75 61 72 74 6c 69 74  |9.stderr=uartlit|
000003e0  65 30 40 31 65 30 30 30  63 30 30 00 73 74 64 69  |e0@1e000c00.stdi|
000003f0  6e 3d 75 61 72 74 6c 69  74 65 30 40 31 65 30 30  |n=uartlite0@1e00|
00000400  30 63 30 30 00 73 74 64  6f 75 74 3d 75 61 72 74  |0c00.stdout=uart|
00000410  6c 69 74 65 30 40 31 65  30 30 30 63 30 30 00 00  |lite0@1e000c00..|
00000420  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
[... 42 lines cut ...]
000602a0  31 32 6b 28 66 61 63 74  6f 72 79 29 2c 2d 28 66  |12k(factory),-(f|
[... cut ...]
000603a0  39 39 43 46 30 30 41 30  30 31 37 36 45 46 33 37  |99CF00A00176EF37|
000603b0  39 43 35 30 30 30 00 63  74 65 69 3d 31 38 34 38  |9C5000.ctei=1848|
000603c0  39 39 37 39 31 32 36 33  35 31 33 00 66 63 74 6c  |99791263513.fctl|
000603d0  6f 63 6b 66 6c 61 67 3d  31 00 62 6f 6f 74 66 6c  |ockflag=1.bootfl|
000603e0  61 67 3d 31 00 72 65 73  74 6f 72 65 5f 64 65 66  |ag=1.restore_def|
000603f0  61 75 6c 74 3d 31 00 00  00 00 00 00 00 00 00 00  |ault=1..........|
00060400  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00080000

DragonBluep, 2024-12-12:
Have you tried installing sysupgrade.bin? It seems that the mac address in your u-boot-env partition is correct.

huytran123, 2024-12-12:
> Have you tried installing sysupgrade.bin? It seems that the mac address in your u-boot-env partition is correct.

Yes. I have installed  sysupgrade.bin from: https://firmware-selector.openwrt.org/?version=23.05.5&target=ramips%2Fmt7621&id=h3c_tx1801-plus. please tell me what information you need? to resolve my issues, i can export them for you

huytran123, 2024-12-12:
Partitions of stock firmware. I have exported from mtd1 to mtd10. do you need them?
cat /proc/mtd
dev:    size   erasesize  name
mtd0: 08000000 00020000 "mtk-nand"
mtd1: 00080000 00020000 "Bootloader"
mtd2: 00080000 00020000 "Boot-env"
mtd3: 00080000 00020000 "exp"
mtd4: 00080000 00020000 "Factory"
mtd5: 02000000 00020000 "firmware1"
mtd6: 00340000 00020000 "kernel1"
mtd7: 01cc0000 00020000 "fs1"
mtd8: 02000000 00020000 "firmware2"
mtd9: 00340000 00020000 "kernel2"
mtd10: 01cc0000 00020000 "fs2"
mtd11: 02c00000 00020000 "plugin"
mtd12: 00a00000 00020000 "yaffs2"

huytran123, 2024-12-13:
@DragonBluep are you researching this?. Please help me resolve it, my friend

DragonBluep, 2024-12-13:
Please apply this patch on your build:
```
--- a/package/base-files/files/lib/functions/system.sh
+++ b/package/base-files/files/lib/functions/system.sh
@@ -66,7 +66,7 @@ get_mac_ascii() {
 	local key="$2"
 	local mac_dirty
 
-	mac_dirty=$(strings "$part" | sed -n 's/^'"$key"'=//p')
+	mac_dirty=$(strings "$part" | sed -n 's/^'"$key"'=//p' | head -1)
 
 	# "canonicalize" mac
 	[ -n "$mac_dirty" ] && macaddr_canonicalize "$mac_dirty"

```

If you  don't want to compile the source codes, you can also apply the change to the file ` /lib/functions/system.sh` on your devices and then reboot.

Make sure you have installed the `sysupgrade.bin`. This page shows you are running `initramfs.bin`.
[image]

huytran123, 2024-12-13:
> Please apply this patch on your build:
>
> ```
> --- a/package/base-files/files/lib/functions/system.sh
> +++ b/package/base-files/files/lib/functions/system.sh
> @@ -66,7 +66,7 @@ get_mac_ascii() {
>  	local [...]

I will try it this tonight. I make sure that i have installed from “ sysupgrade.bin”. 
if your patch is good and fix my problems. Can you push it to official source openwrt?. I will give you my result after try your patch

huytran123, 2024-12-16:
Hi @DragonBluep
Why you do not support u-boot with NMBM for H3C TX1801? is it very difficult, or you do not have many times to do it?
Current I can write H3C TX1801 bootloader without open it and no need TTL serial. If you can do it, please enable NMBM for u-boot (https://github.com/DragonBluep/uboot-mt7621) and OPENWRT firmware for H3C TX1801. Thank you so much my friend.
If you need how to write bootloader without TTL serial. I can share it for you, please tell me what you need from me

DragonBluep, 2024-12-16:
> Hi @DragonBluep Why you do not support u-boot with NMBM for H3C TX1801? is it very difficult, or you do not have many times to do it? 

it was added a long time ago but not tested. I don't have much interesting on this project now.
https://github.com/DragonBluep/uboot-mt7621/commits/staging/

> Current I can write H3C TX1801 bootloader without open it and no need TTL serial. If you can do it, please enable NMBM for u-boot (https://github.com/DragonBluep/uboot-mt7621) and OPENWRT firmware [...]

If TTL flashing can be avoided, it will be very useful for the future users. I can enable telnet on the vendor firmware, but I don't know how to use the H3C shell. If you are willing, you can share it here.

huytran123, 2024-12-16:
Hi @DragonBluep 

"it was added a long time ago but not tested. I don't have much interesting on this project now"
=> Your U-Boot is in this thread: https://www.right.com.cn/forum/thread-8286606-1-1.html, It has enabled NMBM right?. If not yet, can you help me fill informations bellow image? and then i will build it for self. Or you can build U-Boot with NMBM enabled help me?. I very happy with your help for me
[image]

"If TTL flashing can be avoided, it will be very useful for the future users. I can enable telnet on the vendor firmware, but I don't know how to use the H3C shell. If you are willing, you can share it here."
=> Very easy, you can connect it with Telnet, and then input password which used login on admin page. And then run command "debugshell"=> you can access ssh

DragonBluep, 2024-12-16:
The u-boot for tx1801 can not be built online. There is already a template for tx1801. Steps:
```
# prepare toolchain
wget -O - https://github.com/DragonBluep/uboot-mt7621/releases/download/20230517/openwrt-toolchain-ramips-mt7621_gcc-12.3.0_musl.Linux-x86_64.tar.xz | tar --xz -xf -

# clone repo
git clone -b staging --depth 1 https://github.com/DragonBluep/uboot-mt7621.git

# build
cd uboot-mt7621
./build.sh
```

huytran123, 2024-12-16:
Thank so much my friend. I will test it on tonight when i go back my home, and then i will give my result when i have tested it. Please keep in touch

huytran123, 2024-12-16:
> I found a backup on my computer. **untested!!** [u-boot-mt7621.tar.zip](https://github.com/user-attachments/files/18145669/u-boot-mt7621.tar.zip)

Thank so much my friend. I will test it on tonight when i go back my home, and then i will give my result when i have tested it. Please keep in touch

huytran123, 2024-12-16:
Hi @DragonBluep 
I have installed **u-boot-mt7621.bin** in folder **h3c_tx1801-plus-nmbm**, and then my H3C TX1801 can start success, that's great my friend.

Log by TTL serial:
Second info table with writecount 1 found in block 963
NMBM has been successfully attached
Loading Environment from NMBM... OK

Full log. You can read it to more detail
[Log_TTL_H3C_TX1801.txt](https://github.com/user-attachments/files/18150813/Log_TTL_H3C_TX1801.txt)

huytran123, 2024-12-16:
And I tried access http://192.168.1.1/. But cannot access it after install **u-boot-mt7621.bin** in folder **h3c_tx1801-plus-nmbm**
[image]

closed 2024-12-25 by huytran123