openwrt/openwrt #17237: APK failing to download packages with tilde (~) character in name (i.e. sedutil-2022.12.27~7a0cda7f-r2.apk)

Svaag, 2024-12-12:
### Describe the bug

When trying to install packages with the tilde (\~) character in name, such as commit hash stamped packages (sedutil-2022.12.27~7a0cda7f-r2.apk) the new installation code for APK does not send entire package name/url to wget as such:

```
23348 root      1216 S    /www/cgi-bin/cgi-exec
23349 root      1396 S    {package-manager} /bin/sh /usr/libexec/package-manager-call install sedutil
23350 root      1396 S    {package-manager} /bin/sh /usr/libexec/package-manager-call install sedutil
23352 root     12788 S    apk add sedutil
23362 root      7976 S    wget -q -T 60 https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/sedutil-2022.1
```

Error is present also in manual `apk add`:

```
root@BPI-R4:~# apk add sedutil
(1/2) Installing libstdcpp6 (13.3.0-r4)
Executing libstdcpp6-13.3.0-r4.post-install
(2/2) Installing sedutil (2022.12.27~7a0cda7f-r2)
 83% ############################################################################################################################################
```

Wget stalled in same way:
```
root@BPI-R4:~# ps w | grep sed
23422 root     12788 S    apk add sedutil
23432 root      7976 S    wget -q -T 60 https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/sedutil-2022.1
```

Just to confirm, manual download and apk add works:

`` 
root@BPI-R4:~# apk add --allow-untrusted sedutil-2022.12.27\~7a0cda7f-r2.apk
(1/2) Installing libstdcpp6 (13.3.0-r4)
Executing libstdcpp6-13.3.0-r4.post-install
(2/2) Installing sedutil (2022.12.27~7a0cda7f-r2)
Executing sedutil-2022.12.27~7a0cda7f-r2.post-install
OK: 121 MiB in 327 packages
`` 

### OpenWrt version

r28203-f4492b6b1c

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Svaag, 2024-12-12:
After purging apk cache and redownloading ADB lists the install works. I realized this after getting `APKE_ADB_INTEGRITY` "ADB integrity error" on apk commands. Closing case and reopening if I run into this again.

ynezz, 2024-12-12:
If you don't mind, I'll keep this open, looks valid to me.

Svaag, 2024-12-12:
After finding the workaround/fix I started thinking that the tilde naming was a coincidence/red herriing and the real underlying issue all along was the `APKE_ADB_INTEGRITY` issue that got solved by purging cached .adb files from repo.

Also noticed yesterday after fixing that `sedutil` package in SNAPSHOT repo is timestamped `Wed Dec 11 19:28:10 2024` and my first installation attempt was sometime around then but I can't say exactly when.

Another repo thing that caught my eye just now is `sha256sums` has timestamp `Thu Dec 12 02:39:50 2024` which is just a couple hours before I closed the issue with my workaround.

Maybe some race condition timing issue somewhere between `make world` on build server, repo sync, client package-list sync, and download of the actual package? First compiled .apk's turn up around `Wed Dec 11 18:32:18` and if `sha256sums` is only generated at end of buildserver job 8 hours later `Thu Dec 12 02:39:50 2024`  then I could imagine a few cases where a package manager might fail due to mismatch between client & repo states.

kvanbiesen, 2024-12-28:
I'm still bumping into this issue. 

Cleared /car/cache/apk and did an update and still it gives me the adb integrity error :(

sinancetinkaya, 2025-01-02:
Same
```
root@OpenWrt:~# apk update
fetch https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/video/packages.adb
 [https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/video/packages.adb]
OK: 9895 distinct packages available
root@OpenWrt:~#
root@OpenWrt:~# apk add hdparm
(1/1) Installing hdparm (9.65-r1)
ERROR: hdparm-9.65-r1: ADB integrity error
1 error; 54 MiB in 259 packages

```

Svaag, 2025-01-17:
> I'm still bumping into this issue.
> 
> Cleared /car/cache/apk and did an update and still it gives me the adb integrity error :(

I feel like this points to the problem somehow being related to repository state on the server side when apk client goes to the repo...

@kvanbiesen @sinancetinkaya Did this problem solve itself for you by any chance?

Svaag, 2025-01-17:
For reference here is the **only** location in apk-tools source where the APKE_ADB_INTEGRITY error is used:

https://github.com/alpinelinux/apk-tools/blob/f0e3aa4c139dcf03cec58df16dceb48ea7d4e9a4/src/adb.c#L133

Svaag, 2025-01-18:
I'm starting to think these issues we see are all bugs in APK v3 package/index code which is still in heavy development.
OpenWRT APK version is currently locked to a commit 8d93a79fb8958607f097be2c4ac97f4596faf91d which is around tag v3.0.0_pre5.

Latest version tagged is v3.0.0_rc3 and the diff includes heavy changes to both ADB cache handling and wget as well as signatures of packages and other things in mkpkg/mkndx:
https://github.com/alpinelinux/apk-tools/compare/v3.0.0_pre5...v3.0.0_rc3

I imagine upgrading the apk-tools package in the image would solve at least some of the things seen in this issue on either the client side or repo side through buildbot. I'm not sure if it's worth doing the work of upgrading to a rc3, but I'm pretty sure it's not worth troubleshooting this more unless we're running the latest apk-tools.

sinancetinkaya, 2025-01-18:
> I feel like this points to the problem somehow being related to repository state on the server side when apk client goes to the repo...
>
> [@kvanbiesen](https://github.com/kvanbiesen) [...]

I built a firmware with opkg

Svaag, 2025-01-19:
> I built a firmware with opkg

Yeah figures. I skimmed through the ~1k lines possibly related to anything mentioned in this issue and there are some pretty clear things that are very hard fail if they happen. Plus the whole code around the package format integrity verification is brand new since [ADB is the brand new package APKv3 format](https://github.com/alpinelinux/apk-tools/blob/master/doc/apk-v3.5.scd)) and not at all the same APK in all our Alpine containers; APKv3 is very much a work in progress.

Hence this is technically not really a single client or repo or client error but rather pretty much _anything_ that could make .apk APK_DIGEST_SHA256 change from what is written in the ADB metadata inside the APK, or be calculated/validated differently. Also add the complexity of there being a APK_DIGEST_SHA256_160 for, seemingly, backwards compability with clients running early 3.0.0_pre versions of APK. So it's either generating packages and indexes plus of course validating with SHA256, SHA256 hashes cut down to 160 bits, or simply SHA1 hashes (hence the 160 bits) depending on versions and where in the code it's being done. Somewhere deep down into `mkpkg`and `mkndx ` on build server, or anywhere along the chain of the APK client fetching from repo, checking downloaded package hash to see if it was altered in transit, then also checking package signature before finally allowing install.

Also this APKE_ADB_INTEGRITY error around digest handling we see was only added in upstream apk-tools Aug 28 in [fb74ef1c4e669ba0ce54c7fa8e5acffa05da003e](https://github.com/alpinelinux/apk-tools/commit/fb74ef1c4e669ba0ce54c7fa8e5acffa05da003e) to fix [issue 11017 on their gitlab](https://gitlab.alpinelinux.org/alpine/apk-tools/-/issues/11017) named _some changes past [05359b7c](https://gitlab.alpinelinux.org/alpine/apk-tools/-/commit/05359b7c233acbc4ae511da85d7fbd30ab407c48) break existing databases badly_.

And OpenWRT buildbot APK package signing was only first implemented in September in tag [V16](https://git.openwrt.org/?p=buildbot.git;a=tag;h=e658872603cb2cbc2c7e34f15c5719ea60ce53ec), then heavily patched in mid October in tag [V19](https://git.openwrt.org/?p=buildbot.git;a=tag;h=e3e3b83bd9b305771bd3d7139161bf1b1c063d2b) and [V20](https://git.openwrt.org/?p=buildbot.git;a=tag;h=f6d704aa86f7add3558c5f9caa867bb8c455dacc) to catch us up to 3.0.0-pre5 where we are now so that is likely also it's own can of worms that could hit us if we simply update apk-tools to 3.0.0-rc3 .

Maybe we just track this for now and watch upstream APK development since APK release in OpenWRT is blocked either way until a production release of APK 3.0...

Svaag, 2025-01-19:
Here is also a discussion between OpenWRT and APK devs specifically talking about this:

[https://gitlab.alpinelinux.org/alpine/apk-tools/-/issues/11008](https://gitlab.alpinelinux.org/alpine/apk-tools/-/issues/11008)

wirl, 2025-02-08:
Also running into this using the Image Builder (r28774-b61c192e75) for ipq40xx targets. No idea how to fix this on my end. I tried again a few times waiting a few hours between builds but the same packages fail each time.

```
(171/238) Installing cgi-io (2022.08.10~901b0f04-r21)
ERROR: cgi-io-2022.08.10~901b0f04-r21: ADB integrity error
(172/238) Installing liblucihttp0 (2023.03.15~9b5b683f-r1)
ERROR: liblucihttp0-2023.03.15~9b5b683f-r1: ADB integrity error
(173/238) Installing liblucihttp-ucode (2023.03.15~9b5b683f-r1)
ERROR: liblucihttp-ucode-2023.03.15~9b5b683f-r1: ADB integrity error
(174/238) Installing rpcd (2024.12.02~cc9a471c-r1)
ERROR: rpcd-2024.12.02~cc9a471c-r1: ADB integrity error
(175/238) Installing rpcd-mod-file (2024.12.02~cc9a471c-r1)
ERROR: rpcd-mod-file-2024.12.02~cc9a471c-r1: ADB integrity error
(176/238) Installing rpcd-mod-luci (20240305-r1)
ERROR: rpcd-mod-luci-20240305-r1: ADB integrity error
(177/238) Installing rpcd-mod-ucode (2024.12.02~cc9a471c-r1)
ERROR: rpcd-mod-ucode-2024.12.02~cc9a471c-r1: ADB integrity error
(178/238) Installing ucode-mod-html (1)
ERROR: ucode-mod-html-1: ADB integrity error
(179/238) Installing ucode-mod-math (2024.12.06~209f041f-r1)
ERROR: ucode-mod-math-2024.12.06~209f041f-r1: ADB integrity error
(180/238) Installing luci-base (25.037.68074~edc352e)
(181/238) Installing luci-app-package-manager (25.037.68074~edc352e)
(182/238) Installing luci-app-firewall (25.037.68074~edc352e)
(183/238) Installing rpcd-mod-iwinfo (2024.12.02~cc9a471c-r1)
ERROR: rpcd-mod-iwinfo-2024.12.02~cc9a471c-r1: ADB integrity error
(184/238) Installing luci-mod-network (25.037.68074~edc352e)
(185/238) Installing luci-mod-status (25.037.68074~edc352e)
(186/238) Installing luci-mod-system (25.037.68074~edc352e)
(187/238) Installing luci-mod-admin-full (25.037.68074~edc352e)
(188/238) Installing luci-proto-ipv6 (25.037.68074~edc352e)
(189/238) Installing luci-proto-ppp (25.037.68074~edc352e)
(190/238) Installing luci-theme-bootstrap (25.037.68074~edc352e)
(191/238) Installing rpcd-mod-rrdns (20170710)
ERROR: rpcd-mod-rrdns-20170710: ADB integrity error
(192/238) Installing uhttpd (2023.06.25~34a8a74d-r4)
ERROR: uhttpd-2023.06.25~34a8a74d-r4: ADB integrity error
(193/238) Installing uhttpd-mod-ubus (2023.06.25~34a8a74d-r4)
ERROR: uhttpd-mod-ubus-2023.06.25~34a8a74d-r4: ADB integrity error
```

Svaag, 2025-03-08:
> Update: Just tries again and this time it worked, very odd. Repeated the same thing for hours always failing and suddenly its fixed?! I'm glad the image is build but it's kind of weird it fixed itself.

Yeah this is definitely something characteristic of this issue. A hard fail-state somewhere in APK v3 ADB package structure verfification. 
OpenWRT seemingly uses some kind of special functionality where they can put `trusted = 1` on a whole APK repository index instead of having to go in and manually sign all packages when buildbot runs and rebuilds all packages every day. There is more information on that above.

I by the way noticed we updated to laterupstream git snapshot 
```root@BPI-R4:~# apk --version
apk-tools 3.0.0_pre20250208, compiled for aarch64.```

But I just got caught up with this error again. Or something like it. After manually installing packages one by one building something suddenly I got hit with a new error. `ADB block error`

```Executing package manager

apk add atlas-sw-probe-rpc

(1/8) Installing dropbearconvert (2024.86-r1)
(2/8) Installing libatomic1 (13.3.0-r4)
(3/8) Installing libnghttp2-14 (1.63.0-r1)
libnghttp2-14-1.63.0-r1.post-install: Executing script...
(4/8) Installing liburcu (0.15.1-r1)
liburcu-0.15.1-r1.post-install: Executing script...
(5/8) Installing libuv1 (1.48.0-r1)
libuv1-1.48.0-r1.post-install: Executing script...
(6/8) Installing bind-libs (9.20.5-r1)
bind-libs-9.20.5-r1.post-install: Executing script...
(7/8) Installing bind-dig (9.20.5-r1)
bind-dig-9.20.5-r1.post-install: Executing script...
(8/8) Installing atlas-sw-probe-rpc (5080-r2)
atlas-sw-probe-rpc-5080-r2.post-install: Executing script...
2 errors; 72 MiB in 317 packages

Errors

ERROR: dropbearconvert-2024.86-r1: ADB block error
ERROR: libatomic1-13.3.0-r4: ADB block error

The apk install command failed with code 2.```

Not sure if related or not. But we are running 3-4 months newer code on APK now so either way update yours if you run it and let's see where we are at with the `ADB integrity error` and if this one I just got is related somehow or separate issue.

Svaag, 2025-03-09:
Had my system end up in a pretty messed up state by apk breaking halfway through an upgrade, after that everything:

```
root@BPI-R4:~# apk fix
(1/6) Installing libgmp10 (6.3.0-r1)
ERROR: wget: exited with error 3
ERROR: libgmp10-6.3.0-r1: I/O error
(2/6) Installing libmpfr6 (4.2.1-r1)
libmpfr6-4.2.1-r1.post-install: Executing script...
(3/6) Installing libmount1 (2.40.2-r1)
ERROR: wget: exited with error 3
ERROR: libmount1-2.40.2-r1: I/O error
(4/6) Installing wpad-basic-mbedtls (2025.02.09~c8c7d56a-r1)
ERROR: wget: exited with error 3
ERROR: wpad-basic-mbedtls-2025.02.09~c8c7d56a-r1: I/O error
(5/6) Installing libwolfssl5.7.6.e624513f (5.7.6-r1)
ERROR: wget: exited with error 3
ERROR: libwolfssl5.7.6.e624513f-5.7.6-r1: I/O error
(6/6) Installing wpad-wolfssl (2025.02.09~c8c7d56a-r1)
ERROR: wget: exited with error 3
ERROR: wpad-wolfssl-2025.02.09~c8c7d56a-r1: I/O error
5 errors; 91 MiB in 381 packages
```

In this state I can manually wget each package add install it with `allow-untrusted`:

```
root@BPI-R4:~# apk --allow-untrusted add libmpfr6-4.2.1-r1.apk 
(1/6) Installing libgmp10 (6.3.0-r1)
ERROR: wget: exited with error 3
ERROR: libgmp10-6.3.0-r1: I/O error
(2/6) Installing libmpfr6 (4.2.1-r1)
libmpfr6-4.2.1-r1.post-install: Executing script...
(3/6) Installing libmount1 (2.40.2-r1)
ERROR: wget: exited with error 3
ERROR: libmount1-2.40.2-r1: I/O error
(4/6) Installing wpad-basic-mbedtls (2025.02.09~c8c7d56a-r1)
ERROR: wget: exited with error 3
ERROR: wpad-basic-mbedtls-2025.02.09~c8c7d56a-r1: I/O error
(5/6) Installing libwolfssl5.7.6.e624513f (5.7.6-r1)
ERROR: wget: exited with error 3
ERROR: libwolfssl5.7.6.e624513f-5.7.6-r1: I/O error
(6/6) Installing wpad-wolfssl (2025.02.09~c8c7d56a-r1)
ERROR: wget: exited with error 3
ERROR: wpad-wolfssl-2025.02.09~c8c7d56a-r1: I/O error
5 errors; 91 MiB in 381 packages
```

I even got gdb sideloaded, no symbols in my apk BUT /usr/libexec/wget-ssl getting killed with SIGILL:

```
(gdb) set follow-fork-mode child
(gdb) run fix
Starting program:  fix
No executable file specified.
Use the "file" or "exec-file" command.
(gdb) file /usr/bin/apk
Reading symbols from /usr/bin/apk...
(No debugging symbols found in /usr/bin/apk)
(gdb) run fix
Starting program: /usr/bin/apk fix
warning: Unable to determine the number of hardware watchpoints available.
warning: Unable to determine the number of hardware breakpoints available.
(1/2) Installing libmount1 (2.40.2-r1)
  0%                                                                                                                                          [Attaching after process 4635 vfork to child process 4636]
[New inferior 2 (process 4636)]
[Detaching vfork parent process 4635 after child exec]
[Inferior 1 (process 4635) detached]
process 4636 is executing new program: /usr/libexec/wget-ssl

Thread 2.1 "wget" received signal SIGILL, Illegal instruction.
[Switching to process 4636]
0x0000007ff7c5ae30 in ?? ()
```

And yes, running wget manually works:

```
root@BPI-R4:~# /usr/libexec/wget-ssl https://mirror-03.infra.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/wireless-regdb-2025.02.20-r
1.apk
--2025-03-09 18:48:41--  https://mirror-03.infra.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/wireless-regdb-2025.02.20-r1.apk
Resolving mirror-03.infra.openwrt.org... 2a03:4000:3b:221::1, 185.163.119.85
Connecting to mirror-03.infra.openwrt.org|2a03:4000:3b:221::1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3297 (3.2K) [application/octet-stream]
Saving to: 'wireless-regdb-2025.02.20-r1.apk'

wireless-regdb-2025.02.20-r1.apk    100%[==================================================================>]   3.22K  --.-KB/s    in 0s      

2025-03-09 18:48:42 (786 MB/s) - 'wireless-regdb-2025.02.20-r1.apk' saved [3297/3297]
```

Svaag, 2025-03-09:
ASM call at memory location 0x7ff7c5ae30:

`0x7ff7c5ae30    sha512su0       v0.2d, v0.2d`

arlexTech, 2025-04-22:
same with luci here

root@OpenWrt:~# apk add luci
(1/8) Installing liblucihttp0 (2023.03.15~9b5b683f-r1)
ERROR: liblucihttp0-2023.03.15~9b5b683f-r1: ADB block error
(2/8) Installing rpcd (2024.12.02~cc9a471c-r1)
ERROR: rpcd-2024.12.02~cc9a471c-r1: ADB block error
(3/8) Installing rpcd-mod-file (2024.12.02~cc9a471c-r1)
ERROR: rpcd-mod-file-2024.12.02~cc9a471c-r1: ADB block error
(4/8) Installing rpcd-mod-ucode (2024.12.02~cc9a471c-r1)
ERROR: rpcd-mod-ucode-2024.12.02~cc9a471c-r1: ADB block error
(5/8) Installing ucode-mod-math (2025.03.24~b27d70c9-r1)
ERROR: ucode-mod-math-2025.03.24~b27d70c9-r1: ADB block error
(6/8) Installing rpcd-mod-iwinfo (2024.12.02~cc9a471c-r1)
ERROR: rpcd-mod-iwinfo-2024.12.02~cc9a471c-r1: ADB block error
(7/8) Installing uhttpd (2023.06.25~34a8a74d-r4)
ERROR: uhttpd-2023.06.25~34a8a74d-r4: ADB block error
(8/8) Installing uhttpd-mod-ubus (2023.06.25~34a8a74d-r4)
ERROR: uhttpd-mod-ubus-2023.06.25~34a8a74d-r4: ADB block error
10 errors; 27 MiB in 175 packages

pktpls, 2025-07-28:
Have the same issue on my downstream buildbot using snapshot imagebuilder at the moment. Last night of 90 snapshot targets, 10 failed with a hanging APK install and were killed after 20 minutes without activity. All were hanging at busybox.

```
Building images for sunxi - Lamobo Lamobo R1
Packages: falter-standard apk-mbedtls base-files ca-bundle dnsmasq dropbear e2fsprogs firewall4 fstools iwinfo kernel kmod-ata-sunxi kmod-nft-offload kmod-rtl8192cu libc libgcc libustream-mbedtls logd mkf2fs mtd netifd nftables odhcp6c odhcpd-ipv6only partx-utils ppp ppp-mod-pppoe procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic-mbedtls
OK: 0 MiB in 0 packages
Installing packages...
(  1/276) Installing libgcc1 (14.3.0-r5)
(  2/276) Installing libc (1.2.5-r5)
(  3/276) Installing libmbedtls21 (3.6.4-r1)
(  4/276) Installing libubox20250723 (2025.07.23~49056d17-r1)
(  5/276) Installing libuclient20201210 (2024.10.22~88ae8f20-r1)
(  6/276) Installing uclient-fetch (2024.10.22~88ae8f20-r1)
(  7/276) Installing zlib (1.3.1-r1)
(  8/276) Installing apk-mbedtls (3.0.0_pre20250606-r2)
(  9/276) Installing busybox (1.37.0-r5)
```

The build workers are all the same on identical hardware, x86/64 Alpine podman containers, wrappen in a Debian 12 container based on the openwrt/buildbot.git Dockerfile.

I remember seeing this locally on x86/64 Fedora a few times too, and it was hanging on `wget` then.

pktpls, 2025-07-28:
I wonder if https://github.com/openwrt/openwrt/commit/e567f7b617b2a3c82d6122b1231f3324d1103b5e was supposed to fix this

jolness1, 2025-07-31:
> same with luci here
>
> root@OpenWrt:~# apk add luci (1/8) Installing liblucihttp0 (2023.03.15~9b5b683f-r1) ERROR: liblucihttp0-2023.03.15~9b5b683f-r1: ADB block error (2/8) Installing rpcd [...]

Am having similar issues with any package on a recent OpenWRT  [snapshot based build](https://github.com/pesa1234/MT6000_cust_build/tree/main/2025-07-17_r30972-34db0d03c8_next-r4.6.6.rss.mtk) . I was able to install `adblock-app-luci` from the GUI but it didn't install the adblock package. 

When trying to install from CLI I get this:
```
~# apk add adblock
(1/4) Installing coreutils (9.7-r1)
ERROR: coreutils-9.7-r1: ADB block error
(2/4) Installing coreutils-sort (9.7-r1)
ERROR: coreutils-sort-9.7-r1: ADB block error
(3/4) Installing gawk (5.3.2-r1)
 18% ##########################################
```

Haven't had issues until this build. Will try reverting tomorrow to the previous one and report back. If the version of `apk` being used by OpenWRT has been frozen at the same commit then it's behavior shouldn't change.. probably?

efahl, 2025-08-01:
> I've dealt with bugs like this before and they're maddening.

Haha, I tried to chase down a gcc optimizer bug decades ago (-O=spurious seg faults; no -O=runs fine).  Over like two years, I'd take a whack at it every Friday for an hour or two.  Debug prints would make it go away, so I finally just threw up my hands and left `printf("");` in the code to suppress it...

mio-19, 2026-02-22:
Does this issue also affect 25.12.0-rc5?

pktpls, 2026-02-22:
It is worth trying again with 25.12.0-rc5 now. There were a bunch of related APK fixes last week.

yrjyrj123, 2026-04-17:
Same issue here when upgrading from 25.12.1 to 25.12.2 via LuCI Attended Sysupgrade (ASU).

> Request Data:
> 
> {
>     "system_board": {
>         "kernel": "6.12.74",
>         "hostname": "Second",
>         "system": "ARMv8 Processor rev 4",
>         "model": "TP-Link TL-XDR6088",
>         "board_name": "tplink,tl-xdr6088",
>         "rootfs_type": "squashfs",
>         "release": {
>             "distribution": "OpenWrt",
>             "version": "25.12.1",
>             "firmware_url": "https://downloads.openwrt.org/",
>             "revision": "r32768-b21cfa8f8c",
>             "target": "mediatek/filogic",
>             "description": "OpenWrt 25.12.1 r32768-b21cfa8f8c",
>             "builddate": "1773711117"
>         }
>     },
>     "advanced_mode": "1",
>     "url": "https://sysupgrade.openwrt.org",
>     "branch": "25.12",
>     "revision": "r32768-b21cfa8f8c",
>     "efi": null,
>     "request_hash": "7c97cfcf0051fe5e0818298a1af49c34a9209a812b7afd9d3c03c55114060ab4",
>     "sha256_unsigned": "",
>     "client": "luci/26.075.80566~6efa8ae",
>     "packages": {
>         "apk-mbedtls": "3.0.5-r2",
>         "base-files": "1696~b21cfa8f8c",
>         "ca-bundle": "20250419-r2",
>         "collectd-mod-sensors": "5.12.0-r54",
>         "dnsmasq": "2.91-r2",
>         "dropbear": "2025.89-r1",
>         "ethtool-full": "6.15-r1",
>         "fitblk": "2",
>         "hostapd-utils": "2025.08.26~ca266cc2-r1",
>         "htop": "3.4.1-r1",
>         "iperf3": "3.20-r1",
>         "irqbalance": "1.9.5-r1",
>         "kernel": "6.12.74~60d938adcb727697d3015e4285d4c290-r1",
>         "kmod-crypto-hw-safexcel": "6.12.74-r1",
>         "kmod-fs-cifs": "6.12.74-r1",
>         "kmod-gpio-button-hotplug": "6.12.74-r5",
>         "kmod-leds-gpio": "6.12.74-r1",
>         "kmod-mt7915e": "6.12.74.2026.03.05~9f95baf9-r2",
>         "kmod-mt7986-firmware": "6.12.74.2026.03.05~9f95baf9-r2",
>         "kmod-phy-aquantia": "6.12.74-r1",
>         "kmod-usb3": "6.12.74-r1",
>         "libc": "1.2.5-r5",
>         "libustream-mbedtls": "2026.03.01~99f1c0db-r1",
>         "logd": "2025.10.30~6f78fa49-r1",
>         "luci": "26.075.80566~6efa8ae",
>         "luci-app-attendedsysupgrade": "26.075.80566~6efa8ae",
>         "luci-app-commands": "26.075.80566~6efa8ae",
>         "luci-app-statistics": "26.075.80566~6efa8ae",
>         "luci-app-usteer": "26.075.80566~6efa8ae",
>         "mt7986-wo-firmware": "20251125-r1",
>         "mtd": "27",
>         "odhcp6c": "2026.01.25~ee2949e3-r1",
>         "odhcpd-ipv6only": "2026.03.16~edf2e523-r1",
>         "openssh-sftp-server": "10.2_p1-r1",
>         "owut": "2026.01.13~2526d84b-r1",
>         "ppp": "2.5.2-r3",
>         "ppp-mod-pppoe": "2.5.2-r3",
>         "procd-ujail": "2026.03.13~58eb263d-r1",
>         "prometheus-node-exporter-lua-ethtool": "2025.11.22-r1",
>         "prometheus-node-exporter-lua-hostapd_stations": "2025.11.22-r1",
>         "prometheus-node-exporter-lua-hostapd_ubus_stations": "2025.11.22-r1",
>         "prometheus-node-exporter-lua-hwmon": "2025.11.22-r1",
>         "prometheus-node-exporter-lua-wifi": "2025.11.22-r1",
>         "prometheus-node-exporter-lua-wifi_stations": "2025.11.22-r1",
>         "screen": "5.0.1-r1",
>         "tcpdump": "4.99.6-r1",
>         "uboot-envtools": "2025.10-r1",
>         "uci": "2025.12.02~66127cd7-r1",
>         "urandom-seed": "3",
>         "urngd": "2025.10.03~f17e33d9-r1",
>         "wpad-wolfssl": "2025.08.26~ca266cc2-r1"
>     },
>     "profile": "tplink,tl-xdr6088",
>     "target": "mediatek/filogic",
>     "version": "25.12.1",
>     "diff_packages": true,
>     "filesystem": "squashfs"
> }
> STDERR:
> Generate local signing keys...
> WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
> WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
> read EC key
> writing EC key
> WARNING: opening /builder/packages/packages.adb: No such file or directory
> Package list missing or not up-to-date, generating it.
> 
> Building package index...
> ERROR: ethtool-lua-0.1.0-r1: ADB integrity error
> ERROR: prometheus-node-exporter-lua-ethtool-2025.11.22-r1: ADB integrity error
> ERROR: lua-bit32-5.3.0-r1: ADB integrity error
> ERROR: prometheus-node-exporter-lua-hostapd_stations-2025.11.22-r1: ADB integrity error
> ERROR: luabitop-1.0.3-r1: ADB integrity error
> ERROR: prometheus-node-exporter-lua-hostapd_ubus_stations-2025.11.22-r1: ADB integrity error
> ERROR: prometheus-node-exporter-lua-hwmon-2025.11.22-r1: ADB integrity error
> make[2]: *** [Makefile:254: package_install] Error 7
> make[1]: *** [Makefile:193: _call_manifest] Error 2
> make: *** [Makefile:369: manifest] Error 2
> 
> Traceback (most recent call last):
>   File "/app/.venv/lib/python3.14/site-packages/rq/worker/base.py", line 1463, in perform_job
>     return_value = job.perform()
>   File "/app/.venv/lib/python3.14/site-packages/rq/job.py", line 1318, in perform
>     self._result = self._execute()
>                    ~~~~~~~~~~~~~^^
>   File "/app/.venv/lib/python3.14/site-packages/rq/job.py", line 1378, in _execute
>     result = self.func(*self.args, **self.kwargs)
>   File "/app/asu/build.py", line 494, in build
>     result = _build(build_request, job)
>   File "/app/asu/build.py", line 302, in _build
>     report_error(job, check_package_errors(job.meta["stderr"]))
>     ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/app/asu/util.py", line 338, in report_error
>     raise RuntimeError(msg)
> RuntimeError: Impossible package selection