openwrt/openwrt #17239: Extreme 3935 fail to sysupgrade

L-Ances, 2024-12-12:
### Describe the bug

When flashing either the rc 24.10 or the snapshot, the ap refuses to boot back into openwrt after a sysupgrade.
pastebin:https://pastebin.com/AyffEhum

### OpenWrt version

24.10rc2

### OpenWrt release

24.10rc2

### OpenWrt target/subtarget

ipq806x/generic

### Device

Extreme Networks 3935i

### Image kind

Official downloaded image

### Steps to reproduce

sysupg command is sent to the device.

### Actual behaviour

https://pastebin.com/AyffEhum

### Expected behaviour

reboot and boot into openwrt

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vochong, 2024-12-15:
You must use the ":sysupgrade" image instead of the "factory" image when using the "sysupgrade" command.

openwrt-ipq806x-generic-extreme_ap3935-squashfs-nand-sysupgrade.bin

plaes, 2025-01-16:
One thing I had to do to get sysupgrade working (it always reverted back to default software):
1. Make sure to make copy of your uboot environment (`printenv`) 
```
RADIOADDR0=..
RADIOADDR1=..
eth1addr=
ethaddr=..
```
2. Reset the u-boot environment: `env default` and then `saveenv`
3. Reattempt the tftpboot + sysupgrade with `...nand-sysupgrade.bin`

stangri, 2025-01-22:
+1 on this with 24.10.0-rc5. I can get uImage to load, but sysupgrade doesn't take effect and the stock OS is loaded instead of OpenWrt after sysupgrade. I've tried @plaes suggestion to reset environment before tftp/sysupgrade and it didn't help.

I have two units of 3935, one flashed fine following the procedure at https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=3aef61060e3f51aa43fe494d5ff173e81dd43003, however the second one fails after sysupgrade. The links to some information I've captured on the affected system:
MTD: https://paste.ee/p/fD51vYQ8#section0
dmesg: https://paste.ee/p/fD51vYQ8#section1
some boot-up messages: https://paste.ee/p/fD51vYQ8#section2

Tagging @Ansuel @Hurricos, couldn't find username for Glen Lee by the email address from the commit sadly.

stangri, 2025-01-22:
@L-Ances do you mind fixing the OpenWrt version in your initial report? Maybe this is not getting attention because the posted OpenWrt version is invalid and an issue has invalid label.

L-Ances, 2025-01-22:
> +1 on this with 24.10.0-rc5. I can get uImage to load, but sysupgrade doesn't take effect and the stock OS is loaded instead of OpenWrt after sysupgrade. I've tried [@plaes](https://github.com/plaes) suggestion to reset environment before tftp/sysupgrade and it didn't help.
> 
> I have two units of 3935, one flashed fine following the procedure at https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=3aef61060e3f51aa43fe494d5ff173e81dd43003, however the second one fails after sysupgrade. The links to some information I've captured on the affected system: MTD: https://paste.ee/p/fD51vYQ8#section0 dmesg: https://paste.ee/p/fD51vYQ8#section1 some boot-up messages: https://paste.ee/p/fD51vYQ8#section2
> 
> Tagging [@Ansuel](https://github.com/Ansuel) [@Hurricos](https://github.com/Hurricos), couldn't find username for Glen Lee by the email address from the commit sadly.
> 
> PS. This is what happens after sysupgrade: https://paste.ee/p/epH1CqAJ vs a boot into stock OS: https://paste.ee/p/uak4hwFk. It's evident that after sysupgrade the system can't mount UBI device (https://paste.ee/p/epH1CqAJ#s=0&l=140) and goes into some sort of self-recovery (sorry, was using screen for console and some of the log scrolled up before I could capture it).

wow this is completely different, mine seems to have issues copying to PRI for some reason. What version number should I use and how should I fix the original post? sorry, this is all very new to me!

stangri, 2025-01-22:
> wow this is completely different, mine seems to have issues copying to PRI for some reason. What version number should I use and how should I fix the original post? sorry, this is all very new to me!

Oh, sorry, not sure if I need to create another issue then.

For OpenWrt version you need to put 24.10.0-rc2 probably.