openwrt/openwrt #17241: In 23.05.5 release, The network packets transmitted through the LAN port have not passed through the netfilter framework

henrypig, 2024-12-12:
### Describe the bug

My openwrt device has three LAN ports and one WAN port. I tried to monitor the network packets by the netfilter hooks. the hook setting as below:
this->routing_ops.hook = private_routing_hook;
this->routing_ops.pf = AF_INET;
this->routing_ops.hooknum = NF_INET_PRE_ROUTING;
this->routing_ops.priority = NF_IP_PRI_LAST;

The wan port ip is 192.168.1.201 and the other three lan port ips are 192.168.11.201/192.168.21.201/192.168.31.201

The netfilter hooks is called when I accessed the WAN port through my laptop via cmd "ssh  root@192.168.1.201" . But when i accessed the LAN port like 192.168.11.201, netfilter hooks  hook does not respond. It look like the netfilter framework only handle the network packets come form WAN port. I also test on the snapshot (mast) branch, it works fine for both wan port and lan port. 

Who knows what the reason is？

### OpenWrt version

r27859-f9da81d32f

### OpenWrt release

23.05.5 

### OpenWrt target/subtarget

x86/64

### Device

INTEL Corporation MAHOBAY

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-12:
Please show output of
```
ubus call system board
```
and the `nft list ruleset` section that "does not work"

henrypig, 2024-12-12:
root@OpenWrt:~# ubus call system board
{
	"kernel": "5.15.167",
	"hostname": "OpenWrt",
	"system": "Intel(R) Celeron(R) N5105 @ 2.00GHz",
	"model": "Techvision TVI7309X",
	"board_name": "techvision-tvi7309x",
	"rootfs_type": "ext4",
	"release": {
		"distribution": "OpenWrt",
		"version": "23.05.5",
		"revision": "r24106-10cc5fcd00",
		"target": "x86/64",
                "description": "OpenWrt 23.05.5 r24106-10cc5fcd00",
	}
}

brada4, 2024-12-12:
And?

brada4, 2024-12-12:
prerouting will not catch egress packets.

henrypig, 2024-12-13:
Sorry for not replying in a timely manner

root@OpenWrt:~# nft list ruleset
table inet fw4 {
	chain input {
		type filter hook input priority filter; policy drop;
		iifname "lo" accept comment "!fw4: Accept traffic from loopback"
		ct state established,related accept comment "!fw4: Allow inbound established and related flows"
		tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
		iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
		iifname "eth1" jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
		jump handle_reject
	}

	chain forward {
		type filter hook forward priority filter; policy drop;
		ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
		iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
		iifname "eth1" jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
		jump handle_reject
	}

	chain output {
		type filter hook output priority filter; policy accept;
		oifname "lo" accept comment "!fw4: Accept traffic towards loopback"
		ct state established,related accept comment "!fw4: Allow outbound established and related flows"
		oifname "br-lan" jump output_lan comment "!fw4: Handle lan IPv4/IPv6 output traffic"
		oifname "eth1" jump output_wan comment "!fw4: Handle wan IPv4/IPv6 output traffic"
	}

	chain prerouting {
		type filter hook prerouting priority filter; policy accept;
		iifname "br-lan" jump helper_lan comment "!fw4: Handle lan IPv4/IPv6 helper assignment"
	}

	chain handle_reject {
		meta l4proto tcp reject with tcp reset comment "!fw4: Reject TCP traffic"
		reject comment "!fw4: Reject any other traffic"
	}

	chain syn_flood {
		limit rate 25/second burst 50 packets return comment "!fw4: Accept SYN packets below rate-limit"
		drop comment "!fw4: Drop excess packets"
	}

	chain input_lan {
		jump accept_from_lan
	}

	chain output_lan {
		jump accept_to_lan
	}

	chain forward_lan {
		jump accept_to_wan comment "!fw4: Accept lan to wan forwarding"
		jump accept_to_lan
	}

	chain helper_lan {
	}

	chain accept_from_lan {
		iifname "br-lan" counter packets 3 bytes 164 accept comment "!fw4: accept lan IPv4/IPv6 traffic"
	}

	chain accept_to_lan {
		oifname "br-lan" counter packets 65 bytes 5330 accept comment "!fw4: accept lan IPv4/IPv6 traffic"
	}

	chain input_wan {
[... 200 lines cut ...]
		meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
	}

	chain mwan3_policy_wanb_only {
		meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
	}

	chain mwan3_policy_balanced {
		meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 54 bytes 5363 xt target "MARK"
	}

	chain mwan3_policy_wan_wanb {
		meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
	}

	chain mwan3_policy_wanb_wan {
		meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
	}

	chain mwan3_rule_https {
		meta mark & 0x00003f00 == 0x00000000 counter packets 0 bytes 0 jump mwan3_policy_balanced
		meta mark & 0x0000fc00 != 0x0000fc00 counter packets 0 bytes 0 xt target "SET"
		meta mark & 0x0000fc00 != 0x0000fc00 counter packets 0 bytes 0 xt target "SET"
	}
}

henrypig, 2024-12-13:
> prerouting will not catch egress packets.

I had add both prerouting and postrouting hooks. When network packets come from LAN ports, nothing happens.

henrypig, 2024-12-13:
> And?
Sorry for not replying in a timely manner
[nft list ruleset.txt](https://github.com/user-attachments/files/18119442/nft.list.ruleset.txt)

brada4, 2024-12-13:
It is your usage questions, not exposing any bug whatsoever.
Full nftables fiagram for you to understand which hook skips what traffic next time:
https://wiki.nftables.org/wiki-nftables/index.php/Netfilter_hooks