openwrt/openwrt #17245: [24.10.0-rc2] mediatek: bpi-r4: wifi led problem

MichelHanzen, 2024-12-12:
### Describe the bug

[image]
2g/5g/6g blue leds are not working with openwrt 24.10.0-rc2

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

configure leds with the Luci "System/LED configuration" menu
[image]

### Actual behaviour

2g/5g/6g blue leds are not working

### Expected behaviour

the thre blue LED's should blink when the relative wifi are up. It works with the BPI-R4-BE1350-WIFI_MP4_0-SDK-20240620 

### Additional info

Link to the BPI-R4-BE1350-WIFI_MP4_0-SDK-20240620 image:
https://drive.google.com/file/d/1U0tnvoaxDujO1q5QNxnw0i2d93X2zi9r/view?usp=sharing

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MichelHanzen, 2024-12-21:
Problem not solved with 24.10.0-rc3

MichelHanzen, 2024-12-23:
Same problem with 24.10.0-rc4. It looks like nobody cares...
Not taken i charge. I will not loose much time with testing. Back to BPI-R4-BE1350-WIFI_MP4_0-SDK-20240620 image

sasha-rnd, 2025-01-03:
24.10.0-rc4 The problem is not solved

MichelHanzen, 2025-01-20:
Not solved with 24.10.0-rc5

EPinci, 2025-01-26:
Uhm, not authoritative but I think the led configuration you see there is for the leds in the LAN ports (back side, at the side of each lan port), not for the leds of the wifi nic (that might not necessarily be installed at all).
I haven't seen any support at all for the wifi board leds.

MichelHanzen, 2025-01-27:
You're probably right but it's a pity that these front wifi leds are not managed by openwrt 24.10 though they are by BPI-R4-BE1350-WIFI_MP4_0-SDK-20240620 image

[image]

EPinci, 2025-01-27:
Yeah, I know. But since those leds are actually on the wifi board, is probably on the driver/firmware to expose them.

sasha-rnd, 2025-03-30:
24.10.0 r28427 The problem is not solved

Rahzadan, 2025-04-17:
The 2G led can by manipulated using **mt76-phy0** in the LED configuration. Unfortunately **mt76-phy1** and **mt76-phy2** don't exist anymore because bediatek changed the way the physical Wifi interfaces work - they merged them all into a single PHY - whereas they used to be separated.

You can see it here in this diagram:

[image]

You can see the difference in "Before MLO" vs "After MLO". Once they implemented MLO, the 3 existing PHY interfaces (0, 1, 2) became one single phy0. 

I'm not sure how to fix this, but someone would need to somehow make the other 2 LEDs available in the dropdown as something like **mt76-phy0.0** **mt76-phy0.1** **mt76-phy0.0** as these are the names of the 3 wireless interfaces now.

eduardocalfaia, 2026-01-16:
Hi guys, any update here? I have the same issue
Thanks