openwrt/openwrt #17269: Snashot error:  /etc/rc.d/S96led: sh: write error: Invalid argument

edrikk, 2024-12-14:
### Describe the bug

With the recent LED changes committed to Snapshot/main the log shows the message:

Sat Dec 14 12:42:08 2024 daemon.notice procd: /etc/rc.d/S96led: sh: write error: Invalid argument

This is seen on Xioami Redmi AX6S devices and Xiaomi Redmi AX3600 samples.

### OpenWrt version

r28330-15e173bf7e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Self-built image

### Steps to reproduce

Install latest snapshot such as https://github.com/openwrt/openwrt/commit/15e173bf7e873d004d6c665ee81d54a10ce1f868
View log file searching for "error" or "S96"

### Actual behaviour

Error message:  daemon.notice procd: /etc/rc.d/S96led: sh: write error: Invalid argument

### Expected behaviour

No error

### Diffconfig

```text
firmware-selector build.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

edrikk, 2024-12-14:
@blogic ping for awareness

blogic, 2024-12-14:
@edrikk fix is in main branch now, Thanks!

edrikk, 2024-12-17:
Fixed.  Thank you!

closed 2024-12-17 by edrikk