openwrt/openwrt #17272: Kernel panic 0000000096000004 (pc : 0xffffffc00818ff8c) on 23.05.5 for BPI-R3

Norde, 2024-12-14:
### Describe the bug

EDIT: New issue to avoid invisibilization by the github bot

Since upgrading to 23.05.5, my BPI-R3 randomly goes into initramfs mode following a kernel panic.

This happens very randomly (sometimes 2 times in the same week, sometimes 1 time a month), which makes it very difficult to identify the source of the problem.

It's always an Internal error “Oops: 0000000096000004 [#1] SMP” with the same pc: 0xffffffc00818ff8c and
lr: 0xffffffc00818ff1c.

Could someone give me some hints or suggestions for digging this further? (I'm unfortunately limited by the fact that this is my main router :/).

Below is the full pstore content:
`<6>[  570.731437] mt7530-mdio mdio-bus:1f lan4: configuring for phy/gmii link mode
<6>[  570.741434] br-lan: port 4(lan4) entered blocking state
<6>[  570.746654] br-lan: port 4(lan4) entered disabled state
<6>[  570.753749] device lan4 entered promiscuous mode
<6>[  570.762571] mt7530-mdio mdio-bus:1f sfp2: configuring for inband/2500base-x link mode
<6>[  570.772022] br-lan: port 5(sfp2) entered blocking state
<6>[  570.777242] br-lan: port 5(sfp2) entered disabled state
<6>[  570.784379] device sfp2 entered promiscuous mode
<6>[  570.792845] mtk_soc_eth 15100000.ethernet eth1: configuring for inband/2500base-x link mode
<6>[  570.801959] br-wan: port 1(eth1) entered blocking state
<6>[  570.807191] br-wan: port 1(eth1) entered disabled state
<6>[  570.812804] device eth1 entered promiscuous mode
<6>[  570.823433] mt7530-mdio mdio-bus:1f wan: configuring for phy/gmii link mode
<6>[  570.833372] br-wan: port 2(wan) entered blocking state
<6>[  570.838514] br-wan: port 2(wan) entered disabled state
<6>[  570.844844] device wan entered promiscuous mode
<6>[  571.193674] br-lan: port 6(phy1-ap0) entered blocking state
<6>[  571.199251] br-lan: port 6(phy1-ap0) entered disabled state
<6>[  571.205079] device phy1-ap0 entered promiscuous mode
<6>[  571.210221] br-lan: port 6(phy1-ap0) entered blocking state
<6>[  571.215799] br-lan: port 6(phy1-ap0) entered forwarding state
<6>[  571.454445] br-lan: port 7(phy0-ap0) entered blocking state
<6>[  571.460014] br-lan: port 7(phy0-ap0) entered disabled state
<6>[  571.465772] device phy0-ap0 entered promiscuous mode
<6>[  571.470831] br-lan: port 7(phy0-ap0) entered blocking state
<6>[  571.476388] br-lan: port 7(phy0-ap0) entered forwarding state
<6>[  571.482545] br-lan: port 6(phy1-ap0) entered disabled state
<6>[  571.488251] br-lan: port 7(phy0-ap0) entered disabled state
<6>[  571.494283] device phy0-ap0 left promiscuous mode
<6>[  571.499037] br-lan: port 7(phy0-ap0) entered disabled state
<6>[  571.582535] br-lan: port 7(phy0-ap0) entered blocking state
<6>[  571.588103] br-lan: port 7(phy0-ap0) entered disabled state
<6>[  571.593837] device phy0-ap0 entered promiscuous mode
<6>[  572.831432] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
<6>[  572.838208] br-lan: port 6(phy1-ap0) entered blocking state
<6>[  572.843817] br-lan: port 6(phy1-ap0) entered forwarding state
<6>[  573.185509] mt7530-mdio mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[  573.185542] br-lan: port 2(lan2) entered blocking state
<6>[  573.198652] br-lan: port 2(lan2) entered forwarding state
<6>[  573.606508] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[  573.606530] br-lan: port 1(lan1) entered blocking state
<6>[  573.619642] br-lan: port 1(lan1) entered forwarding state
<6>[  574.112840] mt7530-mdio mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[  574.112861] br-wan: port 2(wan) entered blocking state
<6>[  574.125796] br-wan: port 2(wan) entered forwarding state
<6>[  574.131618] IPv6: ADDRCONF(NETDEV_CHANGE): br-wan: link becomes ready
<6>[  574.363225] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
<6>[  574.369956] br-lan: port 7(phy0-ap0) entered blocking state
<6>[  574.375539] br-lan: port 7(phy0-ap0) entered forwarding state
<6>[  574.491785] br-guest: port 1(phy0-ap1) entered blocking state
<6>[  574.497583] br-guest: port 1(phy0-ap1) entered disabled state
<6>[  574.503583] device phy0-ap1 entered promiscuous mode
<6>[  574.509795] br-guest: port 1(phy0-ap1) entered blocking state
<6>[  574.515567] br-guest: port 1(phy0-ap1) entered forwarding state
<6>[  574.521747] IPv6: ADDRCONF(NETDEV_CHANGE): br-guest: link becomes ready
<6>[  574.528947] device phy0-ap1 left promiscuous mode
[... 185 lines cut ...]
<0>[79512.692930] Memory Limit: none`

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

njhsi, 2024-12-16:
same panic happened too on mvebu, disabling Adblock seems work around it ..

Norde, 2024-12-17:
> same panic happened too on mvebu, disabling Adblock seems work around it ..

I'm using Adguard Home but on a separate device (usb gadget). What Adblock do you use and with what kind of setup?