openwrt/openwrt #17273: DEVICE LAN IP above X.X.X.101 breaks build 24.10-rc2

entianbucali, 2024-12-14:
### Describe the bug

changing the device LAN IP to anything above X.X.X.101 will break the device. LAN will not reload correctly and DHCP will be inactive.

### OpenWrt version

openwrt-24.10.0-rc2-mvebu-cortexa9-linksys_wrt3200acm

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

openwrt-24.10.0-rc2-mvebu-cortexa9-linksys_wrt3200acm

### Device

wrt3200acm

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-12-15:
It is probably within the IP address range of its DHCPd server, which is prohibited.

brada4, 2024-12-15:
What is logged on `logread -e dnsmasq` ?

entianbucali, 2024-12-20:
impossible to connect to the router, it needs to be reset. Once the router IP is set to anything X.X.X.101-255 router loses connectivity. It's gone. No more DHCP, no more router IP. It starts broadcasting unidentified network 169.254.7.4

entianbucali, 2024-12-20:
> It is probably within the IP address range of its DHCPd server, which is prohibited.

that's not it, because in previous firmwares you can set the router IP to any range you want and it doesn't conflict. This is a bug specific to firmware 24.10-rc2

entianbucali, 2025-01-01:
issue persists in build 24.10-rc4

kilesduli, 2025-01-02:
I think it's just an issue with the script generating the dnsmasq.conf, but I'm not familiar with OpenWRT and don't know how it was caused. At the same time, I don't think setting the LAN IP in the dhcp-range should cause any errors, but I'm not sure what would happen if the IP allocation is left with only the LAN IP.

TwizzyDizzy, 2025-02-07:
> What is logged on logread -e dnsmasq ?

**log lines don't match due to masked config** but it should be clear, that it refers to the crippled `dhcp-range` setting further down below.
```
Fri Feb  7 22:46:56 2025 daemon.crit dnsmasq[1]: bad dhcp-range at line 38 of /var/etc/dnsmasq.conf.cfg01411c
Fri Feb  7 22:46:56 2025 daemon.crit dnsmasq[1]: FAILED to start up
Fri Feb  7 22:46:56 2025 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 7 crashes, 1 seconds since last crash
```

**/var/etc/dnsmasq.conf.cfg01411c**
```
# auto-generated config file from /etc/config/dhcp                                                                                                                                                                                                                                                                           
conf-file=/etc/dnsmasq.conf                                                                                                                                                                                                                                                                                                  
dhcp-authoritative                                                                                                                                                                                                                                                                                                           
domain-needed                                                                                                                                                                                                                                                                                                                
localise-queries                                                                                                                                                                                                                                                                                                             
read-ethers                                                                                                                                                                                                                                                                                                                  
enable-ubus                                                                                                                                                                                                                                                                                                                  
expand-hosts                                                                                                                                                                                                                                                                                                                 
bind-dynamic                                                                                                                                                                                                                                                                                                                 
local-service                                                                                                                                                                                                                                                                                                                
port=0                                                                                                                                                                                                                                                                                                                       
dhcp-leasefile=/tmp/dhcp.leases                                                                                                                                                                                                                                                                                              
servers-file=server=192.168.1.10                                                                                                                                                                                                                                                                                            
resolv-file=/tmp/resolv.conf.auto                                                                                                                                                                                                                                                                                            
stop-dns-rebind                                                                                                                                                                                                                                                                                                              
[... cut ...]
config dhcp 'GUESTS'
	option interface 'GUESTS'
	option leasetime '1h'
	option limit '60'
	option start '30'
	list dhcp_option '6,192.168.1.10'
	list dhcp_option '42,192.168.1.10'

config dhcp 'GUESTS_LAN'
	option interface 'GUESTS_LAN'
	option leasetime '1h'
	option limit '60'
	option start '30'
	list dhcp_option '6,192.168.1.10'
	list dhcp_option '42,192.168.1.10'
```

interface ip GUESTS: 192.168.2.1
interface ip GUESTS_LAN: 192.168.3.1

Let me know if I can provide any further information.

Cheers
Thomas

Neustradamus, 2025-02-23:
To follow this ticket.

xiaoliandan, 2025-03-08:
### Issue Context
In the **OpenWrt /etc/init.d/dnsmasq** script, the original code generated the DHCP address pool range using:

`
eval "$(ipcalc.sh "${subnet%%/*}" $netmask $start $limit)"
`

**Original Logic:** Split the subnet into an IP address (${subnet%%/*}) and mask ($netmask), passed separately to ipcalc.sh.
**Problem:** When the subnet was in CIDR format (e.g., 192.168.1.0 24), $netmask might be misinterpreted as a dotted-decimal mask (e.g., 24 treated as 192.168.1.0 255.255.255.0), causing ipcalc.sh to throw a **Not a dotted quad** error.

### Fix Implementation
The code was updated to pass the full CIDR-formatted subnet directly:

`
eval "$(ipcalc.sh $subnet $start $limit)"
`

### Verification
Validated with the command:

`
ipcalc.sh 192.168.1.1/24 2 19
`
**Output:**
`
IP=192.168.1.1
NETMASK=255.255.255.0
BROADCAST=192.168.1.255
NETWORK=192.168.1.0
PREFIX=24
COUNT=256
START=192.168.1.2
END=192.168.1.20
`
This confirms the DHCP range is now correctly calculated.

### Summary
This fix ensures ipcalc.sh receives the subnet in CIDR notation, resolving mask parsing errors. Thanks to the community support! 🙌