openwrt/openwrt #17283: Internet disconnect on ZBT Z800AX qmi_wwan 2-1:1.4 wwan0: NETDEV WATCHDOG: CPU: 3: transmit queue 0 timed out 5400 m

5gfreak, 2024-12-15:
### Describe the bug

On ZBT Z800AX using QMI to connect to internet, the connection disconnects with the error:
kern.crit kernel: [ 5450.069735] qmi_wwan 2-1:1.4 wwan0: NETDEV WATCHDOG: CPU: 3: transmit queue 0 timed out 5400 m

Modem RM520N-GL

is there any way to find out what causes the disconnections?

Thank you,

### OpenWrt version

r28330-15e173bf7e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

 Zbtlink ZBT-Z800AX

### Image kind

Official downloaded image

### Steps to reproduce

after a period of time qmi disconnects from internet protocolling the error kern.crit kernel: [ 6156.254276] qmi_wwan 2-1:1.4 wwan0: NETDEV WATCHDOG: CPU: 3: transmit queue 0 timed out 5010 ms
a reboot reconnects to internet for about half an hour and then disconnects again.
or restarting the qmi interface reconnects and then disconnects after some period with the same error

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

SuperMitic, 2024-12-25:
I am getting similar issue on MBIM with latest snapshot on x86 and internet stop working. 
`cdc_mbim 9-1:1.2 wwan0: NETDEV WATCHDOG: CPU: 0: transmit queue 0 timed out 5010 ms`

To fix I need to restart the interface. 
Do you know from which version this issue appeared? I think in older kernel I didn't have this issue

5gfreak, 2024-12-27:
I think it has started from ver. r28200 onwards.

for some reason I reverted the module firmware to ver. RM520NGLAAR03A03M4G_01.200.01.200
and there is no internet breaks anymore. Probably is from the firmware of the moduel I don't know
The module had the ver RM520NGLAAR03A04M4G_01.202.01.202 installed.