openwrt/openwrt #17284: OpenWRT 24.10: kmod-rtl8xxxu missing firmware rtl8192fufw.bin

ZerBea, 2024-12-15:
### Describe the bug

Firmware for rtl8192fu is missing:
```
[  285.859316] usb 1-1.4: rtl8xxxu: Loading firmware rtlwifi/rtl8192fufw.bin
[  285.868825] usb 1-1.4: Direct firmware load for rtlwifi/rtl8192fufw.bin failed with error -2
[  285.882259] usb 1-1.4: Falling back to sysfs fallback for: rtlwifi/rtl8192fufw.bin
[  285.930215] usb 1-1.4: request_firmware(rtlwifi/rtl8192fufw.bin) failed
[  285.939476] usb 1-1.4: Fatal - failed to load firmware
[  285.946993] rtl8xxxu: probe of 1-1.4:1.0 failed with error -11
```

After copying missing firmware, everything is fine:
```
[   12.168041] usb 1-1.4: RTL8192FU rev B (UMC) romver 0, 2T2R, TX queues 3, WiFi=1, BT=0, GPS=0, HI PA=0
[   12.181481] usb 1-1.4: RTL8192FU MAC: 08:be:ac:1c:5c:ac
[   12.190695] usb 1-1.4: rtl8xxxu: Loading firmware rtlwifi/rtl8192fufw.bin
[   12.206368] usb 1-1.4: Firmware revision 7.3 (signature 0x92f1)
```
Missing firmware is from here:
https://gitlab.com/kernel-firmware/linux-firmware

### OpenWrt version

r28178-28de3ffefa

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2708

### Device

Raspberry Pi Model B Plus Rev 1.2

### Image kind

Self-built image

### Steps to reproduce

connect EW-7722UTn V3 to run into the issue mentioned above

### Actual behaviour

firmware is not available

### Expected behaviour

```
iw list
Wiphy phy1
	wiphy index: 1
	max # scan SSIDs: 4
	max scan IEs length: 2257 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	RTS threshold: 2347
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
[... 263 lines cut ...]
CONFIG_MBEDTLS_NIST_KW_C=y
# CONFIG_MBEDTLS_PLATFORM_C is not set
CONFIG_MBEDTLS_PSA_CRYPTO_CLIENT=y
# CONFIG_MBEDTLS_RIPEMD160_C is not set
CONFIG_MBEDTLS_RSA_NO_CRT=y
# CONFIG_MBEDTLS_SELF_TEST is not set
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=y
CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y
# CONFIG_MBEDTLS_SSL_RENEGOTIATION is not set
CONFIG_MBEDTLS_SSL_TLS1_3_COMPATIBILITY_MODE=y
CONFIG_MBEDTLS_SSL_TLS1_3_KEY_EXCHANGE_MODE_EPHEMERAL_ENABLED=y
CONFIG_MBEDTLS_SSL_TLS1_3_KEY_EXCHANGE_MODE_PSK_ENABLED=y
CONFIG_MBEDTLS_SSL_TLS1_3_KEY_EXCHANGE_MODE_PSK_EPHEMERAL_ENABLED=y
CONFIG_MBEDTLS_THREADING_C=y
CONFIG_MBEDTLS_THREADING_PTHREAD=y
# CONFIG_MBEDTLS_VERSION_C is not set
# CONFIG_MBEDTLS_VERSION_FEATURES is not set
CONFIG_PACKAGE_libmbedtls=y
CONFIG_PACKAGE_openwrt-keyring=y
CONFIG_PACKAGE_usign=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-15:
Can you check if firmware file from https://gitlab.com/kernel-firmware/linux-firmware (clone repo, copy file to displayed path relative to /lib/firmware) is sufficient?

ZerBea, 2024-12-15:
Confirmed:
```
[   13.256147] usb 1-1.4: RTL8192FU MAC: 08:be:ac:1c:5c:ac
[   13.265380] usb 1-1.4: rtl8xxxu: Loading firmware rtlwifi/rtl8192fufw.bin
[   13.281042] usb 1-1.4: Firmware revision 7.3 (signature 0x92f1)
...
[  164.160524] rtl8xxxu 1-1.4:1.0 wlan0: entered promiscuous mode
[  178.049601] rtl8xxxu 1-1.4:1.0 wlan0: left promiscuous mode
```

brada4, 2024-12-15:
Make a similar section to this one via PR? Or wait for someb(dy to do it for you.
In general kmod packages do not include fw as one driver can support dozen devices each with unique megabyte for firmware. 
https://github.com/openwrt/openwrt/blob/96eb12e961748373a38738de92f5ef1a87f0cc75/package/firmware/linux-firmware/realtek.mk#L42

ZerBea, 2024-12-16:
Done:
https://github.com/openwrt/openwrt/pull/17286/commits/78949ed206f8c0f7ff04931483ffd6b73786a23d

ZerBea, 2024-12-16:
A driver/device test is here:
https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-10428718

brada4, 2024-12-16:
I dont own the device, just shared the recipe how to make error go away, I see you made serious testing, once you force-push to PR also add extra background sharing tests you did in regard.

ZerBea, 2024-12-16:
My tests are limited to:
- latest longterm and stable [Linux kernel](https://www.kernel.org/)
- netlink (nl80211) and rtnetlink support
- monitor mode
- frame injection
- penetration test successfully done

Other modes are not tested:
```
$ iw list
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
```