openwrt/openwrt #17285: Cannot write to Ubiquiti devices in 24.10.0-rc2. Flash mx25l12805d and mx25l6405d both fail.

aanon4, 2024-12-15:
### Describe the bug

Cannot write the flash of the Rocket AC Lite, the Bullet M XW nor the Rocket M XW. It does work on the Rocket M XM.

This problem came up in the 23.05 series also, but the fixes were merged into OpenWRT and are present in this build. They dont appear to help now.

I see the problem with two macronix chips - the mx25l12805d and the mx25l6405d

### OpenWrt version

24.10.0-rc2

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti Rocket AC Lite (amongst others)

### Image kind

Official downloaded image

### Steps to reproduce

Flash the factory image onto a new device. Try to write anything to the file system where it should be preserved across reboots, and note that it doesn't persist.

### Actual behaviour

No changes to the filesystem are preserved across reboot.

### Expected behaviour

Changes should persist.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aanon4, 2024-12-15:
I'll stick a 'v' on the version ... perhaps that will make you happy bot?

aanon4, 2024-12-15:
See also https://github.com/openwrt/openwrt/issues/17234

aanon4, 2024-12-16:
And here's the fix if you're looking for it.  The function 'spi_nor_init' and the various things it calls does in fact make quite extensive use of the mtd data structure, which wasn't initialized. So moving the initialization of the mtd earlier fixes the problem by actually letting the code unlock the flash.

```
--- /dev/null
+++ b/target/linux/generic/backport-6.6/420-earlier-mtd-setup.patch
@@ -0,0 +1,20 @@
+--- a/drivers/mtd/spi-nor/core.c
++++ b/drivers/mtd/spi-nor/core.c
+@@ -3540,14 +3540,14 @@
+ 	if (ret)
+ 		return ret;
+ 
++	/* No mtd_info fields should be used up to this point. */
++	spi_nor_set_mtd_info(nor);
++
+ 	/* Send all the required SPI flash commands to initialize device */
+ 	ret = spi_nor_init(nor);
+ 	if (ret)
+ 		return ret;
+ 
+-	/* No mtd_info fields should be used up to this point. */
+-	spi_nor_set_mtd_info(nor);
+-
+ 	dev_info(dev, "%s (%lld Kbytes)\n", info->name,
+ 			(long long)mtd->size >> 10);
+ 
```

namiltd, 2024-12-16:
It is not fixed since there is no patch added to the repository yet. Please create a PR with the patch.

aanon4, 2024-12-16:
Yes .. I'm aware that this fix is not part of OpenWRT at this moment. The title is purely to allow people with this problem to find the solution.

Happy to create a PR also.

nfontan, 2025-06-04:
@aanon4 Do you have any firmware compiled with this fix you could share with us? I don't have the knowledge to create it myself, and I'm having this problem. Thank you very much.

aanon4, 2025-06-04:
I work on the AREDN project - [arednmesh.org](http://arednmesh.org) - so the only firmwares we have are for that - sorry.

nfontan, 2025-06-05:
@aanon4 
I was able to do it and created a small repository with the steps I followed, and I added some working images to the releases. Thanks for the fix!

https://github.com/nfontan/openwrt-loco5ac/tree/main

hauke, 2025-09-02:
The fix was integrated in Linux kernel [6.12.44](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?h=v6.12.44&id=264e99d), [6.6.103](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?h=v6.6.103&id=b9e9afc) merged into main in b92bab633fe8cf3c3a72345c08d1ab6a8dd0abd2 and 1c92e468d544db98018c55d5d18e41fc1a3af20b and openwrt-24.10 in c676281e7e26c532b53908f84c59786fde189a76.

closed 2025-09-02 by hauke