openwrt/openwrt #17288: Proton VPN via Wireguard

SirNewis, 2024-12-16:
Hello,

as the title above says I am trying to get the Proton VPN working with the wireguard VPN protocol. I have used it with openVPN but I have experienced network timeouts for my wifi connection as i was using PBR to route all my PC traffic directly to my ISP and all other devices that are connected to the network via WiFi to go through VPN.

Now I have switched to Wireguard VPN protovol but i have no network acces on my wifi AP. I have only network on my PC as it is routed via the PBR to my ISP directly.

Here is my configuration: 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

And Logs
[image]

Best regards,

brada4, 2024-12-17:
Hello,
please fill the issue template , and use code boxes to format text configurations.  3rd fw zone rule makes no sense at all

SirNewis, 2024-12-17:
Hello, 

yes i know it doesn't make sense but i found it on some forum as a fix for someone also wirh no network access.

I configured it now as it should be but still no network access:
[image]

brada4, 2024-12-17:
You ¡hould close your usage questions, they are not bugs warranting code change to fix.

closed 2024-12-17 by Jaeg3rr

brada4, 2024-12-17:
Check in openwrt forum for living person to assist you.