openwrt/openwrt #17291: Unknown kernel command line parameter on Linksys WRT1900ACS1

buboleck, 2024-12-17:
### Describe the bug

There is message about invalid command line parameter in the kernel log:

> 
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mtdblock5 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,256K(u_env),256K(s_env),1m@9m(devinfo),40m@10m(kernel),34m@16m(rootfs),40m@50m(alt_kernel),34m@56m(alt_rootfs),80m@10m(ubifs),-@90m(syscfg)
[    0.000000] Bootloader command line (ignored): console=ttyS0,115200 root=/dev/mtdblock5 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,256K(u_env),256K(s_env),1m@9m(devinfo),40m@10m(kernel),34m@16m(rootfs),40m@50m(alt_kernel),34m@56m(alt_rootfs),80m@10m(ubifs),-@90m(syscfg)
[    0.000000] Unknown kernel command line parameters "earlyprintk mangled_fs=jffs2 mtdparts=armada-nand:2048K(uboot)ro,256K(u_env),256K(s_env),1m@9m(devinfo),40m@10m(kernel),34m@16m(rootfs),40m@50m(alt_kernel),34m@56m(alt_rootfs),80m@10m(ubifs),-@90m(syscfg)", will be passed to user space.

> 

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900ACS

### Image kind

Official downloaded image

### Steps to reproduce

Flash OpenWrt 24.10.0-rc2 and boot the device. 

### Actual behaviour

There is message about unknown kernel command line parameters 

### Expected behaviour

All kernel command line parameters to be valid.

### Additional info

There is no other issue, the device boots and I haven't noticed anything else.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

IdWV, 2024-12-19:
AFAIK this is normal. When I read the installation instructions for this box, it seems to me that neither u-boot nor it's environment is changed. So the OpenWrt kernel still gets the same kernel command line as the firmware kernel did. But it doesn't recognize all of them (because the functionality is deprecated, or not compiled in), and logs that. And somehow passes it to user space either. (Maybe as command line option for init?).

buboleck, 2024-12-19:
Yea, just rebooted it to 23.05.5 and it is the same unknown parameter. I guess I never looked carefully at the logs.