openwrt/openwrt #17292: IGMP snooping on a bridge breaks sending ipv6 multicasts over a vlan

mnlipp, 2024-12-17:
### Describe the bug

I have setup a Fritz!Box 4040 as router with OpenWRT. Instead of using the wan interface for connecting to the upstream router, I use VLAN 98 on br-lan (device `br-lan.98`). I'm tracking the traffic with wireshark on `br-lan` (`vlan.id==98 and ((icmpv6 and icmpv6.type==134) or dhcpv6)`) and I'm capturing the traffic on the upstream router's interface with a second wireshark.

The wireshark that tracks `br-lan.98` shows the SOLICIT messages for requesting a PD being sent to `ff02::1:2`. However, when IGMP snooping is enabled on `br-lan`, these messages never arrive at the upstream router, i.e. the second wireshark does not show them -- obviously they don't leave the box.

With IGMP snooping disabled, the same messages arrive at the upstream's router's interface as expected and everything works find.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ipq40xx/generic

### Device

AVM FRITZ!Box 4040

### Image kind

Official downloaded image

### Steps to reproduce

Described above.

### Actual behaviour

Described above.

### Expected behaviour

As this is a dhcpv6 message, the IGMP setting shouldn't have any effect, the message should be sent.

### Additional info

```
config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        option ipv6 '1'
```

```
config device                                  
        option name 'br-lan.98'                
        option type '8021q'            
        option ifname 'br-lan'         
        option vid '98'                
        option ipv6 '1'                
        option macaddr 'f0:b0:14:76:61:98'
```

```
config interface 'Vdf6'                   
        option proto 'dhcpv6'             
        option device 'br-lan.98'         
        option reqaddress 'none'          
        option reqprefix 'auto'           
        option peerdns '0'                
        option norelease '1'              
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mkrle, 2025-04-02:
I am troubleshooting similar issues with my setup (23.05), my main VLAN is 10, and it's untagged on `lan1` (access port), and tagged on `lan3` (trunk). MLD(v1) queries are generated by OpenWrt and sent out untagged on `lan1` however they are also sent untagged on `lan3` (according to tcpdump) so they get black holed by the switch on the other side of `lan3`.  On `lan1` side everything works fine, but on `lan3` it learns no IPv6 multicast addresses. The switch connected to `lan3` has MLD querier enabled but that doesn't help in any way.

devfaz, 2025-05-25:
Have seen the same issue on a fb4040 with non vlan:1 br-lan.

As soon as I enable igmp snooping ipv6 ra pkts will be dropped (most of the time).

If I keep it disabled ipv6 ra is working perfectly.

The same setup running for years with default no vlan br-lan and igmp snooping worked perfectly.

May not be limited to fb4040 as I see the same behavior with Edge router X.

mkrle, 2025-07-08:
@nbd168 any chance of backporting https://github.com/openwrt/openwrt/commit/38c21271d2b5749f135d662c2b28ef2bbdf607b8 to 24.10 to fix the multicast/ipv6  issues?