openwrt/openwrt #17294: Can you guide me EX5700 mtd3_Factory.bin

Doanduy09, 2024-12-17:
Can you guide me EX5700 mtd3_Factory.bin, I copied it to the dev folder and it was an error for EX5700, I must have done something wrong, I hope you can guide me, thank you

brada4, 2024-12-17:
Try forum
https://forum.openwrt.org/t/looking-for-a-zyxel-telenor-ex5700-wifi-router-2/173942
File with such n`me is not produced here

brada4, 2024-12-18:
You are using ancient immortalwrt, your issue does not involve OpenWRT. 
https://forum.openwrt.org/t/zyxel-ex5700-change-wifi-6gh-to-5ghz/218766

@Ansuel what was wrong atlabeller side here?

Doanduy09, 2024-12-19:
> You are using ancient immortalwrt, your issue does not involve OpenWRT. https://forum.openwrt.org/t/zyxel-ex5700-change-wifi-6gh-to-5ghz/218766
> 
> @Ansuel what was wrong atlabeller side here?

I have upgraded to openwrt, thanks for your support

root@OpenWrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/11280000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '2g'
        option channel '1'
        option htmode 'HE20'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'soc/11280000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
        option band '6g'
        option channel 'auto'
        option htmode 'HE80'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio2'
        option type 'mac80211'
        option path 'platform/soc/18000000.wifi'
        option band '5g'
        option channel '36'
        option htmode 'HE80'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

root@OpenWrt:~#