openwrt/openwrt #17295: SFP module OEM SFP-2.5G(-T-R-RM) based on RTL8221B-VB-CG PHY intermittently failing to establish connection

Betonmischer86, 2024-12-17:
### Describe the bug

SFP module identified as OEM SFP-2.5G-T-R-RM would often lose connectivity (with no errors reported by dmesg). After reinserting the module or several retries of `ip link set eth2 down && ip link set eth2 up` link is restored.

There is a known and previously tested fix for this problem, which consists of the already upstreamed patch set to `drivers/net/phy/mediatek.c` and yet to be merged patch to `drivers/net/phy/sfp.c`. The latter is easy to implement, but leads to the module not functioning altogether.

### OpenWrt version

r28339-8a7239009c

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Steps to reproduce

1. Build OpenWrt from unmodified source.
1.2 Soft reboot or hard reboot the router or pass `ip link set eth2 down && ip link set eth2 up`.
2. Build OpenWrt with the patch to `drivers/net/phy/sfp.c`, which would enable the realtek driver to talk to the module.
2.2. Enable dynamic kernel debugging with `echo "file drivers/net/* +p" > /sys/kernel/debug/dynamic_debug/control`.
2.3. Observe kernel log with `dmesg`.
3. Build OpenWrt with the patch to `drivers/net/phy/sfp.c` and also with the patch to `drivers/net/phy/mediatek.c`, which would revert it to Linux MASTER branch.
3.2. Enable dynamic kernel debugging with `echo "file drivers/net/* +p" > /sys/kernel/debug/dynamic_debug/control`.
3.3. Observe kernel log with `dmesg`.

### Actual behaviour

1. Often no link would be established. No errors in kernel log produced (with or without dynamic debugging)
2. Attempts to set link up end with the error `sfp sfp1: sfp_add_phy failed: -EOPNOTSUPP`. Kernel log:
```
[ 1263.978037] mtk_soc_eth 15100000.ethernet eth2: configuring for inband/sgmii link mode
[ 1263.985986] mtk_soc_eth 15100000.ethernet eth2: major config sgmii
[ 1263.992168] mtk_soc_eth 15100000.ethernet eth2: phylink_mac_config: mode=inband/sgmii/none adv=00,00000000,00000000,000060ef pause=01
[ 1264.007331] sfp sfp1: SM: enter present:down:down event dev_up
[ 1264.013165] sfp sfp1: tx disable 1 -> 0
[ 1264.017019] sfp sfp1: SM: exit present:up:wait
[ 1267.731616] sfp sfp1: los 1 -> 0
[ 1267.734861] sfp sfp1: SM: enter present:up:wait event los_low
[ 1267.740605] sfp sfp1: SM: exit present:up:wait
[ 1289.183300] sfp sfp1: SM: enter present:up:wait event timeout
[ 1289.189871] mdio_bus i2c:sfp1: probed
[ 1289.896408] mtk_soc_eth 15100000.ethernet eth2: interface 2 (mii) rate match none supports 0-3,6-7,13-14
[ 1289.906040] mtk_soc_eth 15100000.ethernet eth2: interface 3 (gmii) rate match none supports 0-3,5-7,13-14
[ 1289.915692] mtk_soc_eth 15100000.ethernet eth2: interface 4 (sgmii) rate match none supports 0-3,5-7,13-14
[ 1289.925428] mtk_soc_eth 15100000.ethernet eth2: interface 22 (1000base-x) rate match none supports 5-7,13-14
[ 1289.935331] mtk_soc_eth 15100000.ethernet eth2: interface 23 (2500base-x) rate match none supports 6-7,13-14,47
[ 1289.945538] mtk_soc_eth 15100000.ethernet eth2: interface 29 (usxgmii) rate match none supports 0-3,5-7,13-14,47
[ 1289.955798] mtk_soc_eth 15100000.ethernet eth2: requesting link mode inband/sgmii with support 00,00000000,00008000,000060ef
[ 1290.043974] sfp sfp1: sfp_add_phy failed: -EOPNOTSUPP
[ 1290.049036] sfp sfp1: SM: exit present:up:fail
[ 1290.053499] sfp sfp1: los 0 -> 1
[ 1290.056725] sfp sfp1: SM: enter present:up:fail event los_high
[ 1290.062551] sfp sfp1: SM: exit present:up:fail
```
3. The module is recognized by the driver and is able to bring link up. However, RX/TX counters sit at 0, no connection can be established. Kernel log:
```
[  112.717325] mtk_soc_eth 15100000.ethernet eth2: configuring for inband/sgmii link mode
[  112.725271] mtk_soc_eth 15100000.ethernet eth2: major config sgmii
[... 229 lines cut ...]
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_ucode-mod-uclient=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wireless-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_SINGBOX_WITH_CLASH_API=y
CONFIG_SINGBOX_WITH_ECH=y
CONFIG_SINGBOX_WITH_GVISOR=y
CONFIG_SINGBOX_WITH_QUIC=y
CONFIG_SINGBOX_WITH_REALITY_SERVER=y
CONFIG_SINGBOX_WITH_UTLS=y
CONFIG_SINGBOX_WITH_WIREGUARD=y
CONFIG_TARGET_ROOTFS_PARTSIZE=960
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2025-04-04:
Got same issue on RTL8261 with Mediatek Feed (assuming you are using also MTK feed)

Betonmischer86, 2025-04-04:
> Got same issue on RTL8261 with Mediatek Feed (assuming you are using also MTK feed)

I'm on a snapshot OpenWrt. I tried different forks, but the issue is there regardless. I got used to resetting the interface when needed, though. I guess nothing can be done until the necessary patch is properly implemented in the kernel.

danpawlik, 2025-04-05:
Odd.
In my case, I plugged SFP modules, then I need to go via UART and use option "Reset all settings to factory defaults" and then reboot. I have too short that module to see whats going on, but very strange behavior.