openwrt/openwrt #17315: Unable to build video modules

graysky2, 2024-12-20:
### Describe the bug

I want to have some of the kernel video modules included (`kmod-video-videobuf2` for example) but they are not. What happens is their respective apk packages get built, but they are empty.

### OpenWrt version

r28354-31e45f62cafb

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Self-built image

### Steps to reproduce

1. Build with diff config shown in the diffconfig section below
2. Flash the image

### Actual behaviour

The apk files corresponding to the modules built are empty.  From the build system you see tiny apk files:
```
% ls -lh bin/targets/bcm27xx/bcm2712/packages/|grep video
-rw-r--r-- 1 facade users  714 Dec 20 07:55 kmod-video-async-6.6.66-r1.apk
-rw-r--r-- 1 facade users  851 Dec 20 07:55 kmod-video-core-6.6.66-r1.apk
-rw-r--r-- 1 facade users  740 Dec 20 07:55 kmod-video-fwnode-6.6.66-r1.apk
-rw-r--r-- 1 facade users  762 Dec 20 07:55 kmod-video-videobuf2-6.6.66-r1.apk
```

On the image you find no modules at all:
```
# ls /lib/modules/6.6.66 | grep video
#
```

### Expected behaviour

The modules should be built and installed

### Diffconfig

```text
[diffconfig](https://github.com/user-attachments/files/18211025/diffconfig.txt)
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

graysky2, 2024-12-20:
@Noltari - I see you have pushed some commits to `target/linux/bcm27xx/modules/video.mk` so pinging you

graysky2, 2024-12-27:
This is true for x86/64 also, so not just an ARM thing

hauke, 2024-12-27:
This is expected.

The kernel configuration options do not have a title, see:
```
config VIDEOBUF2_CORE
	select DMA_SHARED_BUFFER
	tristate
```
See: drivers/media/common/videobuf2/Kconfig

We can not select these options directly, it is only getting selected when some other Kconfig option selects it. This is not happening with your OpenWrt configuration. 

I am getting these warnings when building the configuration:
```
logs/package/kernel/linux/compile.txt:WARNING: kmod-thermal is not available in the kernel config - generating empty package
logs/package/kernel/linux/compile.txt:WARNING: kmod-video-videobuf2 is not available in the kernel config - generating empty package
logs/package/kernel/linux/compile.txt:WARNING: kmod-video-async is not available in the kernel config - generating empty package
logs/package/kernel/linux/compile.txt:WARNING: kmod-video-fwnode is not available in the kernel config - generating empty package
```

graysky2, 2024-12-27:
I am trying to write a module for RPi5B camera facing interface which requires these.  I am not sure what I need to do in order to get them to build.  Do you have a suggestion for what I need in the `DEPENDS=`
```

diff --git a/target/linux/bcm27xx/modules/video.mk b/target/linux/bcm27xx/modules/video.mk
index 8e2b19bd8711..d0df1117f70a 100644
--- a/target/linux/bcm27xx/modules/video.mk
+++ b/target/linux/bcm27xx/modules/video.mk
@@ -117,6 +117,29 @@ endef
 $(eval $(call KernelPackage,vchiq-mmal-bcm2835))
 
 
+define KernelPackage/rp1-cfe
+  SUBMENU:=$(VIDEO_MENU)
+  TITLE:=RP1 Camera Front-End
+  KCONFIG:= \
+    CONFIG_VIDEO_RP1_CFE \
+    CONFIG_VIDEO_V4L2_SUBDEV_API=y \
+    CONFIG_MEDIA_CONTROLLER=y \
+    CONFIG_VIDEOBUF2_DMA_CONTIG \
+    CONFIG_V4L2_FWNODE
+  FILES:=$(LINUX_DIR)/drivers/media/platform/raspberrypi/rp1_cfe/rp1-cfe.ko
+  AUTOLOAD:=$(call AutoLoad,67,rp1-cfe)
+  DEPENDS:= \
+   @TARGET_bcm27xx_bcm2712 +kmod-video-core +kmod-video-fwnode +kmod-video-dma-contig \
+   +kmod-video-videobuf2 +kmod-dma-buf
+endef
+
+define KernelPackage/rp1-cfe/description
+  Driver for the Camera Serial Interface (CSI) to capture video
+  streams from connected cameras.
+endef
+
+
+$(eval $(call KernelPackage,rp1-cfe))
```

hauke, 2024-12-28:
If `CONFIG_VIDEO_RP1_CFE` needs `CONFIG_VIDEOBUF2_CORE` it should select it.

graysky2, 2024-12-28:
I added it but the build is still failing due to `videobuf2-vmalloc.ko` not getting built.

> See: drivers/media/common/videobuf2/Kconfig
Inspecting that file led me to add `CONFIG_VIDEOBUF2_VMALLOC` also but still, when building, that module is not present:
```
ERROR: module '/scratch/stable-openwrt/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/linux-6.6.67/drivers/media/common/videobuf2/videobuf2-vmalloc.ko' is missing.
```