openwrt/openwrt #17316: Cudy TR3000 usb net rndis error

ztj1993, 2024-12-20:
### Describe the bug

I encountered an error while using Cudy TR3000 to share my phone network via USB RNDIS. I can confirm that it can be shared on other OpenWRT devices, but not on this device.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy TR3000 v1

### Image kind

Official downloaded image

### Steps to reproduce

install
```
opkg install kmod-usb-net-rndis
```
or more
```
opkg install kmod-usb-net kmod-usb-net-rndis kmod-usb-net-cdc-ether usbutils
opkg install kmod-nls-base kmod-usb-core kmod-usb-net kmod-usb-net-cdc-ether kmod-usb2
```

usb devices
```
root@OpenWrt:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.15
S:  Manufacturer=Linux 5.15.167 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=11200000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.15
S:  Manufacturer=Linux 5.15.167 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=11200000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

```

log
```
Fri Dec 20 13:51:54 2024 kern.err kernel: [ 2112.403033] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Fri Dec 20 13:51:55 2024 kern.err kernel: [ 2113.682658] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Fri Dec 20 13:51:55 2024 kern.info kernel: [ 2113.689293] usb usb1-port1: attempt power cycle
Fri Dec 20 13:51:56 2024 kern.info kernel: [ 2115.122223] usb 1-1: new full-speed USB device number 67 using xhci-mtk
Fri Dec 20 13:51:56 2024 kern.warn kernel: [ 2115.128965] usb 1-1: Device not responding to setup address.
Fri Dec 20 13:51:57 2024 kern.warn kernel: [ 2115.352225] usb 1-1: Device not responding to setup address.
Fri Dec 20 13:51:57 2024 kern.err kernel: [ 2115.572097] usb 1-1: device not accepting address 67, error -71
Fri Dec 20 13:51:57 2024 kern.info kernel: [ 2115.722079] usb 1-1: new full-speed USB device number 68 using xhci-mtk
Fri Dec 20 13:51:57 2024 kern.warn kernel: [ 2115.728898] usb 1-1: Device not responding to setup address.
Fri Dec 20 13:51:57 2024 kern.warn kernel: [ 2115.952055] usb 1-1: Device not responding to setup address.
Fri Dec 20 13:51:57 2024 kern.err kernel: [ 2116.171957] usb 1-1: device not accepting address 68, error -71
Fri Dec 20 13:51:57 2024 kern.err kernel: [ 2116.178012] usb usb1-port1: unable to enumerate USB device

```

### Actual behaviour

error

### Expected behaviour

Can access the internet through USB

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-20:
Most likely USB power issues, or device you do not mention is not working with usb3. 
Test it against pc and check power subscription as usb2 and usb3

samed, 2024-12-20:
I think it's a cable problem, I had the same issue and replacing the cable solved the issue for me.

ztj1993, 2024-12-23:
We are using the official power supply, 5V3A.
There is no problem with the cable, Windows can share internet access.

ztj1993, 2024-12-30:
According to the understanding, it is caused by insufficient power supply and requires a separate jumper for power supply.

closed 2024-12-30 by ztj1993