openwrt/openwrt #17317: RPCError

shuning0128, 2024-12-20:
### Describe the bug

RPCError
RPC call to luci/getFeatures failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (https://10.69.106.254/luci-static/resources/rpc.js?v=24.354.68373~24edaa2:11:3)

### OpenWrt version

r28354-31e45f62ca

### OpenWrt release

 SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-20:
What addidional packages you had and did not yet install from old version? Full clean install certainly works.
https://forum.openwrt.org/t/rpcerror-with-archer-c7v2-and-openwrt-24-10-0rc2/218966

anomeome, 2024-12-20:
[resolved](https://git.openwrt.org/?p=project/luci.git;a=commit;h=7fe0218bcf20a38423685cad622b18267d304384)

jsdfhasuh, 2025-01-02:
> [resolved](https://git.openwrt.org/?p=project/luci.git;a=commit;h=7fe0218bcf20a38423685cad622b18267d304384)

how to resolve

albertux, 2025-01-21:
a never mind I did a config route that did not like it in /etc/config/network removed and back to normal