openwrt/openwrt #17320: WAN not working after update from 23.05.5 to 24.10.0-rc3

selevn-1, 2024-12-21:
### Describe the bug

after updating from 23.05.5 to 24.10.0-rc3 WAN not working even i setup static ip but still WAN now working  

### OpenWrt version

r28202-8667ca841b

### OpenWrt release

24.10.0-rc3

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

selevn-1, 2024-12-21:
> At least attach the syslog to give us a chance to find a solution to the problem. Check if the problem also occurred in older releases, e.g. 24.10.0-rc2. Have you tried restoring the default configuration?

sorry i didn't take syslog
but yes i have restore my config from v 23.05.5 and didn't work

selevn-1, 2024-12-21:
> At least attach the syslog to give us a chance to find a solution to the problem. Check if the problem also occurred in older releases, e.g. 24.10.0-rc2. Have you tried restoring the default configuration?

i have add logread and yes i restore my config but wan still not working

BUG-reporter, 2025-02-05:
Try the following workaround:
`ethtool -s wan autoneg off speed 100 duplex full`

BUG-reporter, 2025-02-05:
This but is the same as mine: https://github.com/openwrt/openwrt/issues/17711 and it is not fixed in release.

YakBizzarro, 2025-02-14:
Same for me, WAN not working after attended sysupgrade from 23.05.5 to 24.10.0 on my AX6. Solved by restoring a backup. Thanks @przemyslaw-latoch !

agru, 2025-04-15:
I can confirm that I had the same problem when upgrading from 23.05.5 to 24.10.1 (the brand new current version). And I also was able to fix that by restoring my configuration backup. Device is a TP-Link Archer A7 v5.

dicreat, 2025-05-13:
> Try the following workaround: `ethtool -s wan autoneg off speed 100 duplex full`

Router Xiaomi Redmi Router AC2100, same problem after update OpenWrt 23.05.5 -> 24.10.1 

Option "autoneg on" works fine for me, wan was up:
ethtool -s wan autoneg on speed 100 duplex full