openwrt/openwrt #17324: Cudy AX3000 - Timeout waiting for mcu reset state 10

FrozenGamesYT, 2024-12-21:
### Describe the bug

Cudy ax3000 crashes after some time, could be memory leak or firmware issue. wifi becomes unusable and webui breaks.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy WR3000 v1

### Image kind

Official downloaded image

### Steps to reproduce

I don't know

### Actual behaviour

something related to the cpu or wifi chip seems to be crashing after a while

### Expected behaviour

not crashing

### Additional info

`Sat Dec 21 20:07:11 2024 kern.warn kernel: [148464.643870] ------------[ cut here ]------------
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.648581] Timeout waiting for MCU reset state 10
Sat Dec 21 20:07:11 2024 kern.warn kernel: [148464.653489] WARNING: CPU: 0 PID: 23171 at 0xffffffc000aead38 [mt7915e@00000000e9f35ff1+0x22000]
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.662276] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha compat crypto_safexcel sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.662467]  cls_basic act_skbedit act_mirred act_gact ifb ip6_udp_tunnel udp_tunnel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio gpio_button_hotplug aquantia crc_ccitt
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.766679] CPU: 0 PID: 23171 Comm: kworker/u4:2 Not tainted 5.15.167 #0
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.773455] Hardware name: Cudy WR3000 v1 (DT)
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.777973] Workqueue: mt76 0xffffffc000aecdf0 [mt7915e@00000000e9f35ff1+0x22000]
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.785534] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.792570] pc : 0xffffffc000aead38 [mt7915e@00000000e9f35ff1+0x22000]
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.799172] lr : 0xffffffc000aead38 [mt7915e@00000000e9f35ff1+0x22000]
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.805774] sp : ffffffc00c30bc40
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.809162] x29: ffffffc00c30bc40 x28: ffffff8003d42ff8 x27: ffffff8003d4a920
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.816375] x26: ffffff8003d42920 x25: ffffff8003d468e8 x24: ffffff8003d42920
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.823586] x23: 0000000000000000 x22: ffffff8003d4e920 x21: 0000000000000010
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.830797] x20: ffffff8003d503e0 x19: 0000000000000000 x18: ffffffc008a5a2a8
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.838008] x17: 0000000000000000 x16: 0000000000000000 x15: 00000000000003b7
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.845219] x14: 000000000000013d x13: ffffffc00c30b978 x12: ffffffc008ab22a8
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.852430] x11: 434d20726f662067 x10: ffffffc008ab22a8 x9 : 0000000000000000
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.859642] x8 : ffffffc008a5a258 x7 : ffffffc008a5a2a8 x6 : 0000000000000001
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.866852] x5 : ffffff800ff2b6f8 x4 : 0000000000000000 x3 : 0000000000000027
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.874063] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000026
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.881274] Call trace:
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.883795]  0xffffffc000aead38 [mt7915e@00000000e9f35ff1+0x22000]
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.890051]  0xffffffc000aedb44 [mt7915e@00000000e9f35ff1+0x22000]
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.896307]  0xffffffc0080508e0
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.899523]  0xffffffc0080516d0
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.902739]  0xffffffc008058e6c
Sat Dec 21 20:07:11 2024 kern.debug kernel: [148464.905956]  0xffffffc00801581c
Sat Dec 21 20:07:11 2024 kern.warn kernel: [148464.909172] ---[ end trace 906a92d43ba62bf4 ]---
`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-22:
Does it change anything enabling/disabling WED, most likely by means of changing fw4 hw to sw offload or vice versa.

Could you format monospace block with 3x backticks?

FrozenGamesYT, 2024-12-22:
> Does it change anything enabling/disabling WED, most likely by means of changing fw4 hw to sw offload or vice versa.
> 
> Could you format monospace block with 3x backticks?

format fixed.

I have Routing/NAT Offloading setting off because i am using SQM.

brada4, 2024-12-22:
My idea is to shake driver somehow to alt€r code paths and maybe find the setting triggering issue.

brada4, 2024-12-22:
If 24.10-rc works easier to use that.