openwrt/openwrt #17333: 2 bugs in 24.10.0-rc3

jazzyshk, 2024-12-22:
### Describe the bug

1. Channel analysis is not working
2. Sotfware/Hardware NAT setting is not saved on clean setup for rc3 (but previous NAT setting is maintained)

### OpenWrt version

24.10.0-rc3

### OpenWrt release

-

### OpenWrt target/subtarget

-

### Device

xiaomi redmi ax6000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-12-22:
You didn't even provide the hardware on which you found these errors, but I confirm that channel analysis doesn't work on ramips/mt7620/TP-Link-Archer-C5-v4.

jazzyshk, 2024-12-22:
> I can confirm that channel analysis also does not work on ramips/mt7620/TP-Link-Archer-C5-v4.

oh sorry! i'm not a developer... and i'm not good at English...

mediatek/filogic/xiaomi redmi ax6000(openwrt u-boot layout) -> my device!

namiltd, 2024-12-22:
I copied the original modules\luci-mod-status\htdocs\luci-static\resources\view\status\channel_analysis.js file from the luci repository and the scan works. It looks like a problem with the minification of the channel_analysis.js file.

jazzyshk, 2024-12-22:
> I copied the original resources/view/status/channel_analysis.js file from the luci repository and the scan works. Fixed in [openwrt/luci@b43aea7](https://github.com/openwrt/luci/commit/b43aea743e78bb478495d2bfca046a0417dbdfe7) In the next rc the code will be updated so the problem will be fixed.

thanks very much!

namiltd, 2024-12-22:
Second problem fixed in https://github.com/openwrt/luci/commit/8a422af6a0e97cdb7b3df1049898731a4b2d153d

hauke, 2024-12-23:
The problem only appeared in 24.10.0-rc3. It should be fixed in 24.10.0-rc4.

closed 2024-12-23 by hauke