openwrt/openwrt #17337: Wifi Device Configuration does not save Settings via LUCI web Interface

MichaelUray, 2024-12-22:
### Describe the bug

**The operating frequency mode is empty per default:**

[image]

**If I try to change any settings, it only keeps rotating the circle in the save button, but it does not change anything in the configuration.**
[image]

### OpenWrt version

r28202-8667ca841b

### OpenWrt release

24.10.0-rc3

### OpenWrt target/subtarget

ipq40xx/mikrotik

### Device

MikroTik hAP ac3

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

samara1532, 2024-12-22:
MT7921 - rockchip/armv8 - 24.10.0.rc3

Wifi Device Configuration does not save Settings

es86de, 2024-12-22:
Same problem with saving on TP-Link WDR3600 running 24.10.0-rc3.

MichaelUray, 2024-12-22:
I just tested it with `OpenWrt 24.10.0-rc2 r28161-ea17e958b9`:
The mode is also not set to `N` per default, but I can change settings and save/apply them.

namiltd, 2024-12-22:
Similar issue with configuration not being saved: https://github.com/openwrt/openwrt/issues/17333

namiltd, 2024-12-22:
Fixed in https://github.com/openwrt/luci/commit/8a422af6a0e97cdb7b3df1049898731a4b2d153d
The change will be included in the next rc version.

MichaelUray, 2024-12-23:
> Fixed in [openwrt/luci@8a422af](https://github.com/openwrt/luci/commit/8a422af6a0e97cdb7b3df1049898731a4b2d153d)

Thanks, this could get used as workaround for the moment in -rc3:
```
sed -i 's/if(c\[conf\]?\.\[sid\]\[opt\]!=null)/if(c\[conf\]?\.\[sid\]?\.\[opt\]!=null)/g' /www/luci-static/resources/uci.js
```

You might have to clear the browser cache afterwards or open it in an anonymous window.

namiltd, 2024-12-23:
OpenWrt 24.10-rc4 is ready (https://github.com/openwrt/openwrt/commit/1e530e5831f3a0e10576d106236f1fa6f48c39e4). Now all you have to do is wait for the images to be generated.

hauke, 2024-12-23:
The problem only appeared in 24.10.0-rc3. It should be fixed in 24.10.0-rc4.

closed 2024-12-23 by hauke