openwrt/openwrt #17347: Issue with HTMODE Configuration in STA Mode on OpenWrt 24.10.0-rc3

skyo0910061077, 2024-12-23:
### Describe the bug

In OpenWrt 24.10.0-rc3, when using STA mode and specifying htmode, strange behavior occurs. For example, if I set htmode to 'HE80' on 5G, iwinfo shows EHT80. Similarly, if I set htmode to 'VHT40', iwinfo shows VHT80. However, in the LuCI interface, it always displays EHT.

Could this be a version issue with my installed packages? How should I resolve it?

Thank you.

Package name                                | Version
iw                                                    | 6.9-r1
iwinfo                                             | 2024.10.20~b94f066e-r1
mod-mt76-connac                         | 6.6.66.2024.10.28~c246fa54-r1 
kmod-mt76-core                           | 6.6.66.2024.10.28~c246fa54-r1 
kmod-mt7996-233-firmware         | 6.6.66.2024.10.28~c246fa54-r1 
kmod-mt7996-firmware                 | 6.6.66.2024.10.28~c246fa54-r1 
kmod-mt7996-firmware-common | 6.6.66.2024.10.28~c246fa54-r1 
kmod-mt7996e                              | 6.6.66.2024.10.28~c246fa54-r1

### OpenWrt version

r28202-8667ca841b

### OpenWrt release

24.10.0-rc3

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-23:
Same in older versions, you are forced to 2.4 HT40 if relaying from clueless OEM AP, multiple architectures tried. Also wds and mesh. affected.

skyo0910061077, 2024-12-25:
The issue remains unresolved even after upgrading to version 24.10.0-Rc4.