openwrt/openwrt #17350: mbedtls: unable to access Meta websites

LGA1150, 2024-12-23:
### Describe the bug

It looks like the current mbedtls configuration lacks some cipher modes required by Meta websites: neither `uclient-fetch` nor `curl` works.

### OpenWrt version

r28354-31e45f62ca

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

HiWiFi HC5962

### Image kind

Official downloaded image

### Steps to reproduce

`uclient-fetch` (with libustream-mbedtls) to access a Meta website, such as https://www.instagram.com 

### Actual behaviour

```
# uclient-fetch https://www.instagram.com
Downloading 'https://www.instagram.com'
Connecting to 157.240.31.174:443
SSL error: SSL - The requested feature is not available
Connection error: Connection failed
```

### Expected behaviour

`uclient-fetch` downloads index.html successfully. Can be done with libustream-openssl.
```
# uclient-fetch https://www.instagram.com
Downloading 'https://www.instagram.com'
Connecting to 157.240.31.174:443
Writing to 'index.html'
index.html             387k --:--:-- ETA
Download completed (396883 bytes)
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2024-12-23:
I am running into this error message: `TLS handshake fragmentation not supported` when activating `CONFIG_MBEDTLS_DEBUG_C`. 

<details>
<summary>
Detailed debug log
</summary>
```
root@OpenWrt:~# uclient-fetch -vvv https://www.instagram.com
Downloading 'https://www.instagram.com'
library/ssl_tls.c:4606: 0xffffbcff9fe0: => handshake
library/ssl_msg.c:2353: 0xffffbcff9fe0: => flush output
library/ssl_msg.c:2362: 0xffffbcff9fe0: <= flush output
library/ssl_tls.c:4525: 0xffffbcff9fe0: client state: MBEDTLS_SSL_HELLO_REQUEST
library/ssl_msg.c:2353: 0xffffbcff9fe0: => flush output
library/ssl_msg.c:2362: 0xffffbcff9fe0: <= flush output
library/ssl_tls.c:4525: 0xffffbcff9fe0: client state: MBEDTLS_SSL_CLIENT_HELLO
library/ssl_client.c:0919: 0xffffbcff9fe0: => write client hello
library/ssl_client.c:0486: 0xffffbcff9fe0: dumping 'client hello, random bytes' (32 bytes)
library/ssl_client.c:0486: 0xffffbcff9fe0: 0000:  9d c6 6a f0 79 8d 86 b4 bb f7 60 cf 4d f6 2d 5f  ..j.y.....`.M.-_
library/ssl_client.c:0486: 0xffffbcff9fe0: 0010:  cb 06 cb e6 b8 f4 88 11 2c db e7 0a 0f a3 f9 b9  ........,.......
library/ssl_client.c:0511: 0xffffbcff9fe0: dumping 'session id' (32 bytes)
library/ssl_client.c:0511: 0xffffbcff9fe0: 0000:  04 df 4e 85 84 5a 00 4b 24 48 07 cd 1a 1f 03 03  ..N..Z.K$H......
library/ssl_client.c:0511: 0xffffbcff9fe0: 0010:  f2 87 bc 3a fc 1d 5e f6 59 c3 9d d7 f1 1f 66 fb  ...:..^.Y.....f.
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: 1303, TLS1-3-CHACHA20-POLY1305-SHA256
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: 1302, TLS1-3-AES-256-GCM-SHA384
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: 1301, TLS1-3-AES-128-GCM-SHA256
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: cca9, TLS-ECDHE-ECDSA-WITH-CHACHA20-POLY1305-SHA256
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: c02b, TLS-ECDHE-ECDSA-WITH-AES-128-GCM-SHA256
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: c02c, TLS-ECDHE-ECDSA-WITH-AES-256-GCM-SHA384
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: cca8, TLS-ECDHE-RSA-WITH-CHACHA20-POLY1305-SHA256
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: c02f, TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: c030, TLS-ECDHE-RSA-WITH-AES-256-GCM-SHA384
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: c009, TLS-ECDHE-ECDSA-WITH-AES-128-CBC-SHA
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: c00a, TLS-ECDHE-ECDSA-WITH-AES-256-CBC-SHA
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: c013, TLS-ECDHE-RSA-WITH-AES-128-CBC-SHA
library/ssl_client.c:0369: 0xffffbcff9fe0: client hello, add ciphersuite: c014, TLS-ECDHE-RSA-WITH-AES-256-CBC-SHA
library/ssl_client.c:0387: 0xffffbcff9fe0: adding EMPTY_RENEGOTIATION_INFO_SCSV
library/ssl_client.c:0396: 0xffffbcff9fe0: client hello, got 14 cipher suites
library/ssl_client.c:0040: 0xffffbcff9fe0: client hello, adding server name extension: www.instagram.com
library/ssl_tls13_client.c:0057: 0xffffbcff9fe0: client hello, adding supported versions extension
library/ssl_tls13_client.c:0080: 0xffffbcff9fe0: supported version: [3:4]
library/ssl_tls13_client.c:0086: 0xffffbcff9fe0: supported version: [3:3]
library/ssl_tls13_client.c:0572: 0xffffbcff9fe0: no cookie to send; skip extension
library/ssl_tls13_client.c:0285: 0xffffbcff9fe0: client hello: adding key share extension
library/ssl_tls13_generic.c:1552: 0xffffbcff9fe0: Perform PSA-based ECDH/FFDH computation.
library/ssl_tls13_client.c:0356: 0xffffbcff9fe0: dumping 'client hello, key_share extension' (42 bytes)
library/ssl_tls13_client.c:0356: 0xffffbcff9fe0: 0000:  00 33 00 26 00 24 00 1d 00 20 e2 a9 da 05 50 b4  .3.&.$... ....P.
library/ssl_tls13_client.c:0356: 0xffffbcff9fe0: 0010:  c2 53 a2 c6 e4 7d 83 e7 93 c0 b6 4d ee 09 2b 1e  .S...}.....M..+.
library/ssl_tls13_client.c:0356: 0xffffbcff9fe0: 0020:  a2 9f 23 28 d6 dd 6f e9 a9 56                    ..#(..o..V
library/ssl_tls13_client.c:0633: 0xffffbcff9fe0: client hello, adding psk_key_exchange_modes extension
library/ssl_client.c:0230: 0xffffbcff9fe0: client hello, adding supported_groups extension
library/ssl_client.c:0249: 0xffffbcff9fe0: got supported group(001d)
library/ssl_client.c:0281: 0xffffbcff9fe0: NamedGroup: x25519 ( 1d )
library/ssl_client.c:0249: 0xffffbcff9fe0: got supported group(0017)
library/ssl_client.c:0281: 0xffffbcff9fe0: NamedGroup: secp256r1 ( 17 )
library/ssl_client.c:0249: 0xffffbcff9fe0: got supported group(0018)
library/ssl_client.c:0281: 0xffffbcff9fe0: NamedGroup: secp384r1 ( 18 )
library/ssl_client.c:0249: 0xffffbcff9fe0: got supported group(0019)
library/ssl_client.c:0281: 0xffffbcff9fe0: NamedGroup: secp521r1 ( 19 )
library/ssl_client.c:0249: 0xffffbcff9fe0: got supported group(0100)
library/ssl_client.c:0281: 0xffffbcff9fe0: NamedGroup: ffdhe2048 ( 100 )
library/ssl_client.c:0249: 0xffffbcff9fe0: got supported group(0101)
library/ssl_client.c:0281: 0xffffbcff9fe0: NamedGroup: ffdhe3072 ( 101 )
library/ssl_client.c:0249: 0xffffbcff9fe0: got supported group(0102)
library/ssl_client.c:0281: 0xffffbcff9fe0: NamedGroup: ffdhe4096 ( 102 )
library/ssl_client.c:0249: 0xffffbcff9fe0: got supported group(0103)
library/ssl_client.c:0281: 0xffffbcff9fe0: NamedGroup: ffdhe6144 ( 103 )
[... cut ...]
library/ssl_tls.c:5521: 0xffffbcff9fe0: => free
library/ssl_tls.c:5583: 0xffffbcff9fe0: <= free
root@OpenWrt:~# 
```
</details>

It looks like mbedts does not support handshake fragmentation. See this ticket: https://github.com/Mbed-TLS/mbedtls/issues/1840 maybe this PR could fix it: https://github.com/Mbed-TLS/mbedtls/pull/8981 

I do not know why Meta uses handshake fragmentation in their TLS handshake. The handshake looks very similar to the one from Google. Google just puts everything into one record.

hauke, 2024-12-23:
This patch does *not* help:
```
diff --git a/ustream-mbedtls.c b/ustream-mbedtls.c
index 361ff99..ed18e9b 100644
--- a/ustream-mbedtls.c
+++ b/ustream-mbedtls.c
@@ -241,6 +241,10 @@ __ustream_ssl_context_new(bool server)
 		mbedtls_ssl_conf_ciphersuites(conf, default_ciphersuites_client);
 	}
 
+#ifdef MBEDTLS_SSL_PROTO_TLS1_3
+	mbedtls_ssl_conf_tls13_key_exchange_modes(conf, MBEDTLS_SSL_TLS1_3_KEY_EXCHANGE_MODE_EPHEMERAL);
+#endif
+
 #if defined(MBEDTLS_SSL_CACHE_C)
 	mbedtls_ssl_conf_session_cache(conf, &ctx->cache,
 				       mbedtls_ssl_cache_get,
```

Neustradamus, 2024-12-27:
Hope that a solution will be find :/