openwrt/openwrt #17351: Periodic link down+up since 24.10.0-rc*

bagbag, 2024-12-23:
### Describe the bug

Since I updated from 23.05 to 24.10.0-rc2 (and now rc3) I get periodic (minutes to hours apart) link downs where I notice it as if I have pulled the cable from my computer.

### OpenWrt version

r28202-8667ca841b

### OpenWrt release

24.10.0-rc3

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

Unfortunately, I don't know what is causing this behavior as I have not made any configuration changes from 23.05 to 24.10.

### Actual behaviour

Syslog:
```
Mon Dec 23 12:03:24 2024 daemon.notice netifd: Network device 'lan1' link is down
Mon Dec 23 12:03:24 2024 kern.info kernel: [ 3141.980711] mt7530-mdio mdio-bus:1f lan1: Link is Down
Mon Dec 23 12:03:24 2024 kern.info kernel: [ 3141.986162] br-lan: port 1(lan1) entered disabled state
Mon Dec 23 12:03:27 2024 kern.info kernel: [ 3144.841462] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
Mon Dec 23 12:03:27 2024 kern.info kernel: [ 3144.841537] br-lan: port 1(lan1) entered blocking state
Mon Dec 23 12:03:27 2024 daemon.notice netifd: Network device 'lan1' link is up
Mon Dec 23 12:03:27 2024 kern.info kernel: [ 3144.854694] br-lan: port 1(lan1) entered forwarding state
```

### Expected behaviour

Keep link up

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2024-12-24:
Have you tried downgrading to 23.05.5 to confirm that that is still good? If it is, that suggests there might be some difference in the switch port configuration/behaviour between 23.05.x and 24.10 is interacting badly with your PC's ethernet port.  Perhaps using something like `ethtool` (on the router) to evaluate the port setup in both working and problematic states might be helpful.

LGA1150, 2024-12-24:
Try disabling EEE with ethtool. MT7530 is known to have a faulty EEE function.
`ethtool --set-eee lan1 eee off tx-lpi off`

tapper82, 2024-12-24:
@LGA1150 Hi what if you would like to do this for all interfaces? All so would you have to run this at start up?

LGA1150, 2024-12-25:
@tapper82 
This is only a temporary fix, to check if it's really EEE that cause this unstable link.
I was a maintainer of the MediaTek PHY driver. I already [disabled EEE](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/phy/mediatek-ge.c?h=v5.15#n26) on startup, but @arinc9 found out that it is somehow enabled afterwards.

Anyway, it's not standard to open-code EEE disable in PHY drivers. To permanently disable it, add `eee-broken-1000t` and `eee-broken-100tx` to PHY nodes in devicetree.

arinc9, 2024-12-25:
@bagbag please let us know whether disabling EEE prevents unwanted link downs. I will send a patch to disable it in the way that it is not automatically enabled afterwards.

trm2sgm, 2024-12-28:
Same issue here with a Ubiquity EdgeRouter-X.
EEE settings are not supported by ethtool so I cannot try that quick fix.

kberanek, 2024-12-28:
I upgraded to rc4 yesterday on my banana pi r4 and started seeing similar issues.  I was previously on rc2 for a while and didn't notice any issues.  The bpi-r4 also appears to use MT7530 so I tried disabling eee as shown above, but I received error messages of either `Cannot get EEE settings: Not supported` or `eee unmodified, ignoring` depending on which interfaces I ran the command on.  I just downgraded back to rc2 and will monitor, but given how noticeable it was on rc4 I'm pretty sure it wasn't happening on rc2.

I also tried switching to rc3 using owut, but I was getting errors at the time.  If that starts working again I'll also give rc3 a shot and report back.

I've included a redacted excerpt of my logs below.  I have a dual wan setup with wan/wan6 on the eth2 port and wanb/wanb6 on the wan port.  From the logs I have it appears to only be affecting the eth2 port.

```
Sat Dec 28 14:10:11 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 14:10:11 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
Sat Dec 28 14:25:12 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 14:25:12 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
Sat Dec 28 14:35:09 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan.99) X.X.X.X xx:xx:xx:xx:xx:xx
Sat Dec 28 14:35:09 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan.99) X.X.X.X xx:xx:xx:xx:xx:xx
Sat Dec 28 14:40:12 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 14:40:12 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
Sat Dec 28 14:55:12 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 14:55:12 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
Sat Dec 28 15:10:12 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 15:10:12 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
Sat Dec 28 15:25:12 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 15:25:12 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
Sat Dec 28 15:40:12 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 15:40:12 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
Sat Dec 28 15:55:12 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 15:55:12 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
Sat Dec 28 16:10:13 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 16:10:13 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
Sat Dec 28 16:14:21 2024 daemon.notice netifd: Network device 'eth2' link is down
Sat Dec 28 16:14:21 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sat Dec 28 16:14:21 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Sat Dec 28 16:14:21 2024 kern.info kernel: [36280.833988] mtk_soc_eth 15100000.ethernet eth2: Link is Down
Sat Dec 28 16:14:21 2024 daemon.notice netifd: wan6 (12339): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan6" } (Permission denied)
Sat Dec 28 16:14:21 2024 daemon.notice netifd: wan (12337): udhcpc: received SIGTERM
Sat Dec 28 16:14:21 2024 daemon.notice netifd: wan (12337): udhcpc: unicasting a release of X.X.X.X to X.X.X.X
Sat Dec 28 16:14:21 2024 daemon.notice netifd: wan (12337): udhcpc: sending release
Sat Dec 28 16:14:21 2024 daemon.notice netifd: wan (12337): udhcpc: entering released state
Sat Dec 28 16:14:21 2024 daemon.notice netifd: wan (12337): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Sat Dec 28 16:14:21 2024 daemon.notice netifd: Interface 'wan' is now down
Sat Dec 28 16:14:21 2024 daemon.notice netifd: Interface 'wan6' is now down
Sat Dec 28 16:14:21 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Dec 28 16:15:06 2024 daemon.notice netifd: Network device 'eth2' link is up
Sat Dec 28 16:15:06 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Sat Dec 28 16:15:06 2024 daemon.notice netifd: Interface 'wan' is setting up now
Sat Dec 28 16:15:06 2024 daemon.notice netifd: Interface 'wan6' has link connectivity
Sat Dec 28 16:15:06 2024 daemon.notice netifd: Interface 'wan6' is setting up now
Sat Dec 28 16:15:06 2024 kern.info kernel: [36326.234889] mtk_soc_eth 15100000.ethernet eth2: Link is Up - 10Gbps/Full - flow control off
Sat Dec 28 16:15:06 2024 daemon.notice netifd: wan (5900): udhcpc: started, v1.36.1
Sat Dec 28 16:15:06 2024 daemon.notice netifd: wan (5900): udhcpc: broadcasting discover
Sat Dec 28 16:15:09 2024 daemon.notice netifd: wan (5900): udhcpc: broadcasting discover
Sat Dec 28 16:15:09 2024 daemon.notice netifd: wan (5900): udhcpc: broadcasting select for X.X.X.X, server X.X.X.X
[... cut ...]
Sat Dec 28 18:10:25 2024 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan.99) X.X.X.X xx:xx:xx:xx:xx:xx
Sat Dec 28 18:10:25 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan.99) X.X.X.X xx:xx:xx:xx:xx:xx
Sat Dec 28 18:10:25 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan.99) X.X.X.X xx:xx:xx:xx:xx:xx
Sat Dec 28 18:25:14 2024 daemon.notice netifd: wanb (6018): udhcpc: sending renew to server X.X.X.X
Sat Dec 28 18:25:14 2024 daemon.notice netifd: wanb (6018): udhcpc: lease of X.X.X.X obtained from X.X.X.X, lease time 1800
```

bagbag, 2024-12-28:
Sorry, haven't had time yet to test `ethtool --set-eee lan1 eee off tx-lpi off`.
I just ran it. It did not print anything. It also didn't throw an error like it did for the others.  I will observe if these disconnections are still there and report back.

arinc9, 2024-12-28:
> The bpi-r4 also appears to use MT7530

The SoC on BPI-R4 is MT7988. This SoC has a switch component which uses part of the MT7530 switch intellectual property (IP), but the switch PHYs on this SoC is not part of the IP. So MT7988 is irrelevant here.

> From the logs I have it appears to only be affecting the eth2 port.

The MAC that pertains to the eth2 port is not part of the switch component of MT7988. Your issue seems to be different than the one reported here.

kberanek, 2024-12-28:
> The MAC that pertains to the eth2 port is not part of the switch component of MT7988. Your issue seems to be different than the one reported here.

Got it.  Thanks.  I'll open another issue.

Djfe, 2025-01-08:
someone at Freifunk Aachen just observed this behavior on his D-Link COVR-X1860 on the WAN Port. (maybe LAN was also affected not sure, he switched back to release to have a stable OS for now)

I personally own an Asus RT-AX53U and didn't run into this issue, yet. so it might be a form of race condition. is it maybe related to the home network?

pktpls, 2025-01-08:
I observed this regularly at one of my locations over the past year with snapshot (pre-24.10). The link would go down and recover after a few seconds or minutes. The location doesn't exist anymore, but I'll keep an eye out at other locations.

The setup was as follows: USW-Flex as router in the middle, 2x Zyxel NWA55AXE as dump APs. The Zyxels showed the issue, but not simultaneously. The USW-Flex didn't show the issue IIRC.

bagbag, 2025-01-08:
Sorry for my late feedback, but got time now to investigate further.

@LGA1150
I'm on rc5 now, disabled EEE and it seems to work. Since disabling that, I havn't seen that problem again.

As an addition to what @Djfe said: I have 2 ports connected on my ASUS RT-AX53U, and only 1 port shows that symptom, so it might really be only in specific cases.

maurerle, 2025-01-08:
So I have a D-Link COVR-X1860 here, which does have this issue continuously and does not establish a proper link.
As wifi (and mesh) works fine

Setting `ethtool --set-eee internet eee off tx-lpi off` works and fixes the problem after ~5 seconds.

@LGA1150 should we add something like eee-broken-1000t to this device then?

arinc9, 2025-01-09:
@bagbag @maurerle please spend a week with EEE off and let me know if you encounter the issue at all. I will send a patch to disable EEE on MT7530 by default after I'm certain.

GoetzGoerisch, 2025-01-17:
not seeing this on `TP-Link ER605 v2`  currently `-rc5`

maurerle, 2025-01-18:
@arinc9 this issue was is completeley resolved by having `ethtool --set-eee internet eee off tx-lpi off`, though it only appeared on 1 of 6 COVR-X1860.
The issue appears with every reboot, while it never appeared on the other devices.

We also had the problem that the link continuously goes down and up without ever establishing a proper connection, unlike others reporting in this issue.

So I think to
> disable EEE on MT7530 by default 
seems to be a good idea. Can you send a patch to test?

Djfe, 2025-01-18:
important:
the other 5 covr devices were installed in completely different locations/wired setups

Djfe, 2025-01-18:
I forgot: is there a way to dump the wee registers of the affected device?
If so: how? :)

rmilecki, 2025-01-19:
Was EEE support added recently (which would make this problem a bug in a new feature) or was it present in old releases too (which would make this problem a regression)?

Djfe, 2025-01-19:
@dangowrt do you maybe know what could've added this? (wasn't the ethernet driver updated/adjusted in the past months?)
The issue is not present on the COVR device when we run it with OpenWrt 23 in the same setup. It appears to be a regression.

matjon, 2025-01-21:
Hello,

Like the bug reporter, I also have an Asus RT-AX53U and can confirm that on OpenWRT 23.05.5, EEE is disabled out of the box. `ethtool --show-eee lan1` gives me:

```
Cannot get EEE settings: Not supported
EEE Settings for lan1:
	EEE status: disabled
	Tx LPI: 30 (us)
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  Not reported
	Link partner advertised EEE link modes:  Not reported
```

After `ethtool --set-eee lan1 eee on`, I get:
```
EEE Settings for lan1:
	EEE status: enabled - inactive
	Tx LPI: 30 (us)
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  100baseT/Full 
	                            1000baseT/Full 
	Link partner advertised EEE link modes:  Not reported
```

I get `EEE status: enabled - inactive` even for disconnected ports.

maurerle, 2025-01-22:
I also have a Netgear WAX206 which has a `PHY [mdio-bus:07] driver [RTL8221B-VB-CG 2.5Gbps PHY (C45)] (irq=POLL)` WAN port which does not work.

As soon as I trigger `ethtool --set-eee wan eee off tx-lpi off` - it works.

There are also other people who have this working without this, so it looks like it is an issue which does not occur on all devices:
https://forum.openwrt.org/t/wax206-24-10-0-rc4-wan-port-ooo/220279

arinc9, 2025-01-23:
I'm going to refrain from disabling EEE for now as I see reports here that suggest that it is not EEE that is the cause of the problem, but rather it reveals an issue of race condition of sorts when EEE is enabled. Perhaps the cause is in the MediaTek ethernet driver...

maurerle, 2025-01-26:
Regarding the WAX206, I can say, that the Fix in #17701 did fix the broken WAN port.

The WAX206 did never spam the logs with link down+up messages, so it is a slightly different error behavior than in this issue, but was fixed through EEE being turned off as well (which is not needed anymore).

So the issue on mt7530 might be timing related as well.

maurerle, 2025-01-26:
We further investigated the issue and found out, that this issue occured in combination with a 100MBit/s port.
While it was working fine in Gigabit scenarios all along.

The 100MBit/s port was identified as Gigabit/FullDuplex with ethtool on versions from 38d79aca0eb22218b5dcc2ada98cde4bb734bc9c.
We then had the behavior as described in the initial posting.

I can say, that the issue was not reproducible for me with a firmware from 939550bfbfc8423873348f890d2507eb6732ffec
While it was reproducible with 38d79aca0eb22218b5dcc2ada98cde4bb734bc9c

ref: https://github.com/openwrt/openwrt/compare/38d79aca0eb22218b5dcc2ada98cde4bb734bc9c...939550bfbfc8423873348f890d2507eb6732ffec

So for me this was fixed with [v24.10.0-rc6](https://github.com/openwrt/openwrt/releases/tag/v24.10.0-rc6).
The 100MBit/s link is identified correctly on this firmware using ethtool.
So I would wait for negative reports on v24.10-0-rc6, otherwise this looks fixed.

bagbag, 2025-01-28:
> While it was working fine in Gigabit scenarios all along.

I'm only using Gigabit, so I guess it's another issue.

I enabled (or disabled the disabling of) EEE and will test it with rc6.

bagbag, 2025-01-28:
A few minutes in... and there it is.

```
Tue Jan 28 12:23:41 2025 kern.info kernel: [  749.773876] mt7530-mdio mdio-bus:1f lan1: Link is Down
Tue Jan 28 12:23:41 2025 kern.info kernel: [  749.779189] br-lan: port 1(lan1) entered disabled state
Tue Jan 28 12:23:41 2025 daemon.notice netifd: Network device 'lan1' link is down
Tue Jan 28 12:23:44 2025 kern.info kernel: [  752.725705] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
Tue Jan 28 12:23:44 2025 kern.info kernel: [  752.725811] br-lan: port 1(lan1) entered blocking state
Tue Jan 28 12:23:44 2025 kern.info kernel: [  752.738909] br-lan: port 1(lan1) entered forwarding state
Tue Jan 28 12:23:44 2025 daemon.notice netifd: Network device 'lan1' link is up
```

lexmark3200, 2025-02-01:
rc7 on ER-X the same issue. Link Is Down and Up, a few times...
So I make :
`root@Router:~# ethtool --show-eee eth1
EEE Settings for eth1:
        EEE status: enabled - inactive
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  100baseT/Full
                                    1000baseT/Full
        Link partner advertised EEE link modes:  Not reported
root@Router:~# ethtool --set-eee eth1 eee off
root@Router:~# ethtool --show-eee eth1
EEE Settings for eth1:
        EEE status: disabled
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  Not reported
root@Router:~#`
I will test it a few days

i300220, 2025-02-05:
Same issue with ERL3 (octeon).  Eee not supported.  Details in https://forum.openwrt.org/t/octeon-24-10-0-rc7-periodic-lan-connectivity-loss/223802

rubictus, 2025-02-06:
I just saw I've the same issue on my Xiaomi AX6000 still in 23.05.5 :

```shell
root@ax6000:~# logread | grep mt7530-mdio | tail -10
Thu Feb  6 18:00:45 2025 kern.info kernel: [716806.012471] mt7530-mdio mdio-bus:1f lan3: Link is Down
Thu Feb  6 18:00:48 2025 kern.info kernel: [716808.682819] mt7530-mdio mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Feb  6 18:00:49 2025 kern.info kernel: [716809.511874] mt7530-mdio mdio-bus:1f lan3: Link is Down
Thu Feb  6 18:00:51 2025 kern.info kernel: [716812.179077] mt7530-mdio mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Feb  6 19:54:39 2025 kern.info kernel: [723632.570761] mt7530-mdio mdio-bus:1f lan3: Link is Down
Thu Feb  6 19:54:42 2025 kern.info kernel: [723635.251591] mt7530-mdio mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Feb  6 19:54:42 2025 kern.info kernel: [723636.090091] mt7530-mdio mdio-bus:1f lan3: Link is Down
Thu Feb  6 19:54:45 2025 kern.info kernel: [723638.747861] mt7530-mdio mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Feb  6 19:54:46 2025 kern.info kernel: [723639.567501] mt7530-mdio mdio-bus:1f lan3: Link is Down
Thu Feb  6 19:54:49 2025 kern.info kernel: [723642.244133] mt7530-mdio mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx

root@ax6000:~# ethtool --show-eee lan3
EEE Settings for lan3:
        EEE status: disabled
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full 
                                   1000baseT/Full 
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  100baseT/Full 
                                                 1000baseT/Full

root@ax6000:~# cat /etc/*release* | head
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.5'
DISTRIB_REVISION='r24106-10cc5fcd00'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.5 r24106-10cc5fcd00'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="23.05.5"
ID="openwrt"
```

lexmark3200, 2025-02-08:
I have installed 24.10.0, and EEE off, and now is working without issue.

```
root@Router:~# ubus call system board
{
        "kernel": "6.6.73",
        "hostname": "Router",
        "system": "MediaTek MT7621 ver:1 eco:3",
        "model": "Ubiquiti EdgeRouter X",
        "board_name": "ubnt,edgerouter-x",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "ramips/mt7621",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}

root@Router:~# ethtool --show-eee eth1
EEE Settings for eth1:
        EEE status: disabled
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  Not reported

root@Router:~# uptime
 12:27:35 up 2 days, 13:37,  load average: 0.08, 0.02, 0.00
```

danpawlik, 2025-02-08:
Linking solution for issue: https://github.com/openwrt/openwrt/issues/15348

Lyceris-chan, 2025-02-12:
im on 24.10 stable and turning of EEE seemed to have fixed it here fingers crossed I'm on a ASUS RT-AX53U <html>

opty77, 2025-03-08:
Do you use ethtool or ethtool-full?

opty77, 2025-03-21:
No problem on

```
root@strong:~# ubus call system board
{
        "kernel": "6.6.73",
        "hostname": "strong",
        "system": "MediaTek MT7621 ver:1 eco:3",
        "model": "MTC Wireless Router WR1201",
        "board_name": "mtc,wr1201",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "ramips/mt7621",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

(actually Strong 1200):

```
root@strong:~# for iface in $(echo $(for i in `seq 4`; do echo lan$i; done) wan); do ethtool --show-eee $iface; done
EEE Settings for lan1:
        EEE status: enabled - inactive
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  100baseT/Full
                                    1000baseT/Full
        Link partner advertised EEE link modes:  Not reported
EEE Settings for lan2:
        EEE status: enabled - inactive
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  100baseT/Full
                                    1000baseT/Full
        Link partner advertised EEE link modes:  Not reported
EEE Settings for lan3:
        EEE status: enabled - inactive
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  100baseT/Full
                                    1000baseT/Full
        Link partner advertised EEE link modes:  Not reported
EEE Settings for lan4:
        EEE status: enabled - inactive
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  100baseT/Full
                                    1000baseT/Full
        Link partner advertised EEE link modes:  Not reported
EEE Settings for wan:
        EEE status: enabled - inactive
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  100baseT/Full
                                    1000baseT/Full
        Link partner advertised EEE link modes:  Not reported
```

since the first 24.10.0 boot after upgrade from 23.05.4 at

```
root@strong:~# date -d @`grep btime /proc/stat | cut -d \  -f 2` -Is
2025-03-09T12:06:26+01:00
```

opty77, 2025-03-21:
Just for the record: I didn't have to use ethtool-full.

matte3560, 2025-03-25:
> No problem on
> 
> ...
> 
> ```
> EEE Settings for lan1:
>         Link partner advertised EEE link modes:  Not reported
> EEE Settings for lan2:
>         Link partner advertised EEE link modes:  Not reported
> EEE Settings for lan3:
>         Link partner advertised EEE link modes:  Not reported
> EEE Settings for lan4:
>         Link partner advertised EEE link modes:  Not reported
> EEE Settings for wan:
>         Link partner advertised EEE link modes:  Not reported
> ```

To me this indicates that you are not seeing the problem because none of the systems connected to your device actually support EEE, and this has the same effect as disabling it on the device (EEE is not used).

jfsgithub2020, 2025-03-30:
Just in case it helps with debugging, I almost bricked 2 out 3 routers with most likely this bug. Only a downgrade to 23.05.5 helped. 

Devices (both ramips platform): 
* Netgear R6220  (mt7621)
* Xiaomi Gigabit Router 4A (mt7621)

Symptoms:
* during boot of 24.10.0 LAN LEDs come on as expected
* after some seconds they go off and remain off
* no network connections to LAN ports are possible
* wifi is active but not usable (due to mssing DHCP access to my DHCP server outside of openwrt)

Tried workaround:
* go into rescue mode
* modify EEE settings in /etc/rc.local with: ethtool --set-eee lan1 eee off tx-lpi off
* this did not help

What I do not know is, if some part of the boot procedure will enable eee again after running rc.local.

Djfe, 2025-03-30:
have you tried it for wan?

lexmark3200, 2025-03-30:
Can you check the log? In my case (er-x) I saw on log, the problem was with eth0 - WAN and I turn off EEE only on this port. Maybe in log you find wich port is off...

jfsgithub2020, 2025-03-30:
> have you tried it for wan?

No, as I do not use the wan port, just one of the lan ports. (sorry, I must correct my initial post: the command was only shutting of the eee feature, not the tx-lpi).

Djfe, 2025-03-30:
if you ever get stuck again:
recovering netgear devices is easy, use https://github.com/jclehner/nmrpflash

jfsgithub2020, 2025-03-30:
> Symptoms:
> 
>    * during boot of 24.10.0 LAN LEDs come on as expected
>    * after some seconds they go off and remain off
>    * no network connections to LAN ports are possible
>    * wifi is active but not usable (due to mssing DHCP access to my
>      DHCP server outside of openwrt)
> 
>  Message ID: ***@***.***>
> I suspect that this is a separate bug. I think that if there was a
> problem with EEE, the routers in rescue mode would also be impossible to
> connect to.

Good point. So somehow setting up normal operation creates the problem, 
while rescue mode still works.

> Did you try setting a static IP address on the Ethernet / WiFi clients?

It already has a static IP, but the link is down according to the LED. IMHO, that
shows that it has nothing to do with IP address allocation.

> Did you try erasing the configuration while doing sysupgrade?

No. I love keeping my setup ;-)

> EEE could be disabled in /etc/config/network like so (note: this is just
> an example which must be adapted to your needs):
> 
> config device
>          option name 'lan1'
>          option eee '0'

Ah, ok. I can try that. I see no equivalent option in Luci though.

I have 3 openwrt devices with roughly the same setup - just access points for wifi, nothing
fancy. The ramips devices failed the upgrade and the TP-Link CP210 just runs fine.

I will think about it, on how to figure out the root cause.

jfsgithub2020, 2025-03-30:
> if you ever get stuck again: recovering netgear devices is easy, use https://github.com/jclehner/nmrpflash

Just to be sure: does this work with openwrt images as well? In the text it just mentions the Netgear firmware....

Djfe, 2025-03-30:
with specific factory images for netgear devices
don't use the sysupgrades with nmrpflash

yes, it works like a charm :)

matjon, 2025-03-30:
Hello,

Regarding this bug, there was a very informative discussion on the 
openwrt-devel mailing list: 
https://lists.openwrt.org/pipermail/openwrt-devel/2025-January/043671.html 
(see also following messages).

I'd like to ask: would it be possible to disable EEE by default for the 
time being on affected devices, only for OpenWRT?
It could be reenabled once the proper solution is known.
This bug affects many people and is serious when it hits. The workaround 
is not simple to
perform especially for less experienced users - there is no respective 
option in LuCI AFAIK.
There has been no major progress on fixing this issue and the cause 
seems elusive.
On the other hand, EEE is not a critical function of the router.

Additionally, do you think it would be helpful to try to bisect this 
problem? (Disclaimer: I don't have resources currently for this).

matjon, 2025-03-30:
> Can you check the log? In my case (er-x) I saw on log, the problem was with eth0 - WAN and I turn off EEE only on this port. Maybe in log you find wich port is off...

Using a rescue mode, you could in `/etc/config/system` set something like:

```
config system
        ...
	option log_file '/system.log'

```
Then, during a normal boot, the log will be stored in `/system.log` to be accessible with a rescue mode just after that. Just don't forget to clear this setting after debugging, in order not to trash your flash and exhaust flash space. (note: `/var/` is on a tmpfs by default so do not place your log file there).

jfsgithub2020, 2025-03-30:
> Can you check the log? In my case (er-x) I saw on log, the problem was with eth0 - WAN and I turn off EEE only on this port. Maybe in log you find wich port is off...

Good point, but as I only connect 1 port (lan1) and that should work, it doesn't matter what the others are doing.

jfsgithub2020, 2025-03-30:
With the following config changes, to disable eee and tx-lpi:
...
config device
        option name 'lan1'
        option ipv6 '0'
        option eee '0'
        option tx-lpi '0'
...

I did add the logging now as suggested by matjon:

> config system
>         ...
> 	option log_file '/system.log'
> ```

I let the Xiaomi Mi Router 4A Gigabit Edition boot into 24.10.0, waited a while, checked if it is accessible,
then power down, wait, power up and push the button for rescue mode. Then login to 192.168.1.1
and copy the system.log file to my PC. Here now the result:

Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.306122] mt7530-mdio mdio-bus:1f lan1: configuring for phy/gmii link mode
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.315331] br-lan: port 2(lan1) entered blocking state
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.320829] br-lan: port 2(lan1) entered disabled state
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.326237] mt7530-mdio mdio-bus:1f lan1: entered allmulticast mode
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.337555] mt7530-mdio mdio-bus:1f lan1: left allmulticast mode
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.345391] br-lan: port 2(lan1) entered blocking state
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.350944] br-lan: port 2(lan1) entered disabled state
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.356290] mt7530-mdio mdio-bus:1f lan1: entered allmulticast mode
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.365015] mt7530-mdio mdio-bus:1f lan1: left allmulticast mode
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.373465] br-lan: port 2(lan1) entered blocking state
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.378993] br-lan: port 2(lan1) entered disabled state
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.384367] mt7530-mdio mdio-bus:1f lan1: entered allmulticast mode
Sun Mar 30 11:22:05 2025 kern.info kernel: [   21.394198] mt7530-mdio mdio-bus:1f lan1: left allmulticast mode

This keeps going on for lan2 and wan.

After this, hostapd tries to attach to the ports as well and the same repeats with the phy devices:

Sun Mar 30 11:22:09 2025 daemon.notice hostapd: Restart interface for phy phy0
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.132926] br-lan: port 2(phy0-ap0) entered blocking state
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.138739] br-lan: port 2(phy0-ap0) entered disabled state
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.144589] mt7603e 0000:02:00.0 phy0-ap0: entered allmulticast mode
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.154010] mt7603e 0000:02:00.0 phy0-ap0: left allmulticast mode
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.162532] br-lan: port 2(phy0-ap0) entered blocking state
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.168552] br-lan: port 2(phy0-ap0) entered disabled state
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.174402] mt7603e 0000:02:00.0 phy0-ap0: entered allmulticast mode
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.189117] mt7603e 0000:02:00.0 phy0-ap0: left allmulticast mode
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.210070] br-lan: port 2(phy0-ap0) entered blocking state
Sun Mar 30 11:22:09 2025 kern.info kernel: [   25.215761] br-lan: port 2(phy0-ap0) entered disabled state

Anybody can make sense of this?

jfsgithub2020, 2025-03-30:
Additional info. In rescue mode, I see that eee is enabled, but not used:
root@OpenWrt-failsafe:~# ethtool --show-eee lan1
EEE settings for lan1:
        EEE status: enabled - inactive
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  100baseT/Full
                                    1000baseT/Full
        Link partner advertised EEE link modes:  Not reported

And in this state, I can login and do things.

jfsgithub2020, 2025-03-30:
Update: I decided to reset to factory defaults. Result:

No problem with first boot.
{
        "kernel": "6.6.73",
        "hostname": "OpenWrt",
        "system": "MediaTek MT7621 ver:1 eco:3",
        "model": "Xiaomi Mi Router 4A Gigabit Edition",
        "board_name": "xiaomi,mi-router-4a-gigabit",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "ramips/mt7621",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}

jfsgithub2020, 2025-03-30:
Update: manually setting up my changes from the factory defaults, still does not reproduce the problem.
* changed the IP address - still static
* added a couple of wifi access points
* added an ssh key
* disabled ipv6
* disabled wan interfaces
* disabled DHCP server on openwrt

So it seems an upgrade issue with settings from old versions. Sorry for the trouble. Still, it would be interesting what caused it. I will check with diff, if I can pin point anything in the config files.

opty77, 2025-03-30:
Probably doesn't matter as @matte3560 suggested in [2751158994](https://github.com/openwrt/openwrt/issues/17351#issuecomment-2751158994) but I (had to) reset the configuration too due to migration to DSA.

barcell1, 2025-04-02:
i think MAYBE a lost my router (archer c6 v3.2) ethernet connections because of this bug. I had a fully functional router, and then after trying installing 24.10 openwrt from stock tplink firmware, all ethernet ports stoped to work, even in uboot.

 uboot and openwrt only recognize eth0 ( No lan1, lan2 .. wan ). 

I tried also to install 23.05.05 but didnt make any difference, i have others exactly similar routers that had 23.05 without any problem.

matjon, 2025-04-05:
I have been looking further into this issue.

In `target/linux/generic/backport-6.6`, there are 51 patches backported from later kernels that touch the mt7530 driver. Their names begin with `790-` and 7 of them touch the EEE code.

OpenWRT 23.05.5 uses Linux 5.15.167  ( see: https://openwrt.org/releases/23.05/notes-23.05.5 ) and it contains the Linux kernel's

commit 06dfcd4098cf ("net: dsa: mt7530: fix enabling EEE on MT7531 switch on all boards")

This patch has been backported to Linux v5.15.158 ( see https://kernel.dance/#06dfcd4098cf ) . According to the commit message:

> My testing on MT7531 shows a certain amount of traffic loss when EEE is
> enabled. That said, I haven't come across a board that enables EEE. So
> enable EEE on the switch MACs but disable EEE advertisement on the switch
> PHYs. This way, we don't change the behaviour of the majority of the boards
> that have this switch. The mediatek-ge PHY driver already disables EEE
> advertisement on the switch PHYs but my testing shows that it is somehow
> enabled afterwards. Disabling EEE advertisement before the PHY driver
> initialises keeps it off.

According to my previous tests, on OpenWRT 23.05.5 EEE was *disabled* out of the box (on Asus RT-AX53U), which aligns with the commit message. So something with disabling it on the PHY layer apparently fails on OpenWRT 24.10.

I'd try to check if disabling the mt7530 patch series on `target/linux/generic/backport-6.6` (by removing `790-*.patch` files and recompiling) fixes this issue.

wenfu929, 2025-04-06:
[ 2110.089899] mt7530-mdio mdio-bus:1f lan2: Link is Down
[ 2110.100359] br-lan: port 2(lan2) entered disabled state
[ 2113.984437] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[ 2114.000442] br-lan: port 2(lan2) entered blocking state
[ 2114.010902] br-lan: port 2(lan2) entered forwarding state
[ 2293.084619] mt7530-mdio mdio-bus:1f lan2: Link is Down
[ 2293.095072] br-lan: port 2(lan2) entered disabled state
[ 2295.586853] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[ 2295.586958] br-lan: port 2(lan2) entered blocking state
[ 2295.613131] br-lan: port 2(lan2) entered forwarding state
[ 2317.383065] mt7530-mdio mdio-bus:1f lan2: Link is Down
[ 2317.393531] br-lan: port 2(lan2) entered disabled state
[ 2321.342435] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[ 2321.342543] br-lan: port 1(lan1) entered blocking state
[ 2321.368699] br-lan: port 1(lan1) entered forwarding state
[ 2337.090153] mt7530-mdio mdio-bus:1f lan1: Link is Down
[ 2337.100591] br-lan: port 1(lan1) entered disabled state
[ 2339.592213] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[ 2339.592326] br-lan: port 1(lan1) entered blocking state
[ 2339.618498] br-lan: port 1(lan1) entered forwarding state
[ 2375.772041] mt7530-mdio mdio-bus:1f lan1: Link is Down
[ 2375.782585] br-lan: port 1(lan1) entered disabled state
[ 2378.178363] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[ 2378.184537] br-lan: port 1(lan1) entered blocking state
[ 2378.204620] br-lan: port 1(lan1) entered forwarding state
[ 2413.774265] mt7530-mdio mdio-bus:1f lan1: Link is Down
[ 2413.784791] br-lan: port 1(lan1) entered disabled state
[ 2417.732305] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[ 2417.732413] br-lan: port 1(lan1) entered blocking state
[ 2417.758605] br-lan: port 1(lan1) entered forwarding state

wenfu929, 2025-04-06:
[ 2110.089899] mt7530-mdio mdio-bus:1f lan2: Link is Down
[ 2110.100359] br-lan: port 2(lan2) entered disabled state
[ 2113.984437] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[ 2114.000442] br-lan: port 2(lan2) entered blocking state
[ 2114.010902] br-lan: port 2(lan2) entered forwarding state
[ 2293.084619] mt7530-mdio mdio-bus:1f lan2: Link is Down
[ 2293.095072] br-lan: port 2(lan2) entered disabled state
[ 2295.586853] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[ 2295.586958] br-lan: port 2(lan2) entered blocking state
[ 2295.613131] br-lan: port 2(lan2) entered forwarding state
[ 2317.383065] mt7530-mdio mdio-bus:1f lan2: Link is Down
[ 2317.393531] br-lan: port 2(lan2) entered disabled state
[ 2321.342435] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[ 2321.342543] br-lan: port 1(lan1) entered blocking state
[ 2321.368699] br-lan: port 1(lan1) entered forwarding state
[ 2337.090153] mt7530-mdio mdio-bus:1f lan1: Link is Down
[ 2337.100591] br-lan: port 1(lan1) entered disabled state
[ 2339.592213] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[ 2339.592326] br-lan: port 1(lan1) entered blocking state
[ 2339.618498] br-lan: port 1(lan1) entered forwarding state
[ 2375.772041] mt7530-mdio mdio-bus:1f lan1: Link is Down
[ 2375.782585] br-lan: port 1(lan1) entered disabled state
[ 2378.178363] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[ 2378.184537] br-lan: port 1(lan1) entered blocking state
[ 2378.204620] br-lan: port 1(lan1) entered forwarding state
[ 2413.774265] mt7530-mdio mdio-bus:1f lan1: Link is Down
[ 2413.784791] br-lan: port 1(lan1) entered disabled state
[ 2417.732305] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[ 2417.732413] br-lan: port 1(lan1) entered blocking state
[ 2417.758605] br-lan: port 1(lan1) entered forwarding state
我这个也是遇到这个问题，是不是和你们一样的？

matjon, 2025-04-06:
@barcell1 
That snippet from the log seems the most important:

```
[    1.761182] mt7530-mdio mdio-bus:1f: MT7530 adapts as multi-chip module
[    2.774046] mt7530-mdio mdio-bus:1f: reset timeout
[    2.778946] mt7530-mdio: probe of mdio-bus:1f failed with error -145
[    2.786205] clk: Disabling unused clocks
```

There is a timeout after 1 second above.

Please disconnect all cables from the router (incl. power) and leave it so for several hours. Maybe this will fix it.

barcell1, 2025-04-06:
> Please disconnect all cables from the router (incl. power) and leave it so for several hours. Maybe this will fix it.

I did , but problem still persist, do you think will be a patch to correct this issue ?

what i dont get is if was a 24.10  bug which look it was, why rollback to 23.05 dint make it work again ? 
i have others exactly same routers in 23.05 working

[125 more replies]