openwrt/openwrt #17353: configuring smart plug triggers router crash

ErikHorn, 2024-12-23:
### Describe the bug

Attempting to configure a new smart plug on my network appears to reliably crash the router.

I upgraded to 24.10.0-rc2 several weeks ago in hopes that it may fix some general unreliability that I've been having with my smart plugs. The router has been stable since the upgrade, until I attempted to set up some newly purchased smart plugs to my network. 

While attempting to set them up, I could reliably crash the router.

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Dynalink DL-WRX36

### Image kind

Official downloaded image

### Steps to reproduce

1. Put smart plug into setup mode
2. use smartlife app on a wifi connected phone to set up device

### Actual behaviour

router crashed and reboots

The following log output was captured using "logread -f" in a ssh session. There were no abnormal messages prior to this:

```
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: fatal error received:
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] QC Image Version: QC_IMAGE_VERSION_STRING=WLAN.HK.2.9.0.1-02146-QCAHKSWPL_SILICONZ-1
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] Image Variant : IMAGE_VARIANT_STRING=8074.wlanfw.eval_v2Q
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992]
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] wal_peer_control.c:2904 Assertion is_graceful_to_handle failedparam0 :zero, param1 :zero, param2 :zero.
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] Thread ID      : 0x00000062  Thread name    : WLAN RT1  Process ID     : 0
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] Register:
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] SP : 0x4bfd5a60
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] FP : 0x4bfd5a68
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] PC : 0x4b107f58
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] SSR : 0x00000008
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] BADVA : 0x00020000
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] LR : 0x4b1076f4
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992]
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] Stack Dump
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] from : 0x4bfd5a60
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992] to   : 0x4bfd62c0
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.888992]
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.938023] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.960338] remoteproc remoteproc0: handling crash #1 in cd00000.q6v5_wcss
Mon Dec 23 10:17:30 2024 kern.err kernel: [  464.968456] remoteproc remoteproc0: recovering cd00000.q6v5_wcss
Mon Dec 23 10:17:30 2024 kern.info kernel: [  465.000934] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
Mon Dec 23 10:17:30 2024 kern.warn kernel: [  465.248268] ath11k c000000.wifi: failed to transmit frame -108
Mon Dec 23 10:17:30 2024 kern.warn kernel: [  465.248319] ath11k c000000.wifi: failed to transmit frame -108
Mon Dec 23 10:17:30 2024 kern.warn kernel: [  465.343839] ath11k c000000.wifi: failed to find peer 18:de:50:99:71:03 on vdev 1 after creation
Mon Dec 23 10:17:30 2024 kern.warn kernel: [  465.343905] ath11k c000000.wifi: failed to find peer vdev_id 1 addr 18:de:50:99:71:03 in delete
Mon Dec 23 10:17:30 2024 kern.info kernel: [  465.347108] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
Mon Dec 23 10:17:30 2024 kern.warn kernel: [  465.351364] ath11k c000000.wifi: failed peer 18:de:50:99:71:03 delete vdev_id 1 fallback ret -22
Mon Dec 23 10:17:30 2024 kern.warn kernel: [  465.367709] ath11k c000000.wifi: Failed to add peer: 18:de:50:99:71:03 for VDEV: 1
Mon Dec 23 10:17:30 2024 kern.warn kernel: [  465.376456] ath11k c000000.wifi: Failed to add station: 18:de:50:99:71:03 for VDEV: 1
Mon Dec 23 10:17:30 2024 kern.warn kernel: [  465.391976] ath11k c000000.wifi: qmi ignore invalid mem req type 3
Mon Dec 23 10:17:30 2024 kern.info kernel: [  465.398428] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
Mon Dec 23 10:17:30 2024 kern.info kernel: [  465.398468] ath11k c000000.wifi: fw_version 0x290b84a5 fw_build_timestamp 2024-09-23 11:32 fw_build_id WLAN.HK.2.9.0.1-02146-QCAHKSWPL_SILICONZ-1
Mon Dec 23 10:17:30 2024 daemon.notice hostapd: phy1-ap0: STA 18:de:50:99:71:03 IEEE 802.11: Could not add STA to kernel driver
Mon Dec 23 10:17:30 2024 daemon.notice hostapd: phy1-ap1: STA 18:de:50:99:71:03 IEEE 802.11: did not acknowledge authentication response
```

### Expected behaviour

client activity should not crash router

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ErikHorn, 2026-08-25:
> I hit the same assertion and the same firmware build on a Linksys MX4200v2 (qualcommax/ipq807x, OpenWrt 25.12.5):
> 
> ```
> wal_peer_control.c:2904 Assertion is_graceful_to_handle failed
> QC Image Version: WLAN.HK.2.9.0.1-02146-QCAHKSWPL_SILICONZ-1
> ```
> 
> I have filed [#24864](https://github.com/openwrt/openwrt/issues/24864) with a full reproduction, since the trigger turned out to be narrower and more specific than "configuring a smart plug": the crash requires **two BSSes on the same 2.4 GHz radio** plus a client that repeatedly fails peer creation.
> 
> With two BSSes on the 2.4 GHz radio the device resets ~40 s after `wifi reload`, reproduced three times. With either BSS alone it is stable - including one run with 19 associated clients, which rules out load as a factor. The two-BSS control was re-run immediately after the single-BSS passes and crashed again, so the variable is the BSS count, not elapsed time. Dropping to a single BSS on 2.4 GHz resolved it completely.
> 
> Like your case, the failing client is a Tuya device (OUI `a8:80:55`), which matches the SmartLife smart plug you described - I suspect the plug is the peer that fails, and the second BSS is what turns that failure into a firmware assertion. If you can check whether your 2.4 GHz radio was also running a second SSID at the time, that would be a useful confirmation.
> 
> [#24864](https://github.com/openwrt/openwrt/issues/24864) has the full bisection matrix, the complete assertion dump, environment details and a working workaround, so I would suggest tracking the investigation there.

I do have 2 ssids configured on the radio.

Since I originally created this ticket, I have upgraded to openwrt 25.12.5. I just plugged in the smart plug, which started up in pairing/configuration mode from previous attempts, and the router immediately crashed and rebooted two or three times until pairing timed out. 

I disabled one ssid, unplugged the smartplug, plugged it back in. It entered pairing/configuration mode and the router did not reset.

juandiegotroconis, 2026-08-25:
> I do have 2 ssids configured on the radio.
>
> Since I originally created this ticket, I have upgraded to openwrt 25.12.5. I just plugged in the smart plug, which started up in pairing/configuration [...]

Great confirmation, and worth noting this is actually cross-device — you're on a Dynalink DL-WRX36, I'm on a Linksys MX4200v2, different boards but the same ipq807x-family target and the same firmware blob/assertion. The fact that "two SSIDs + pairing-mode plug → immediate crash" and "drop to one SSID → clean pairing" reproduces identically on completely different hardware makes it pretty hard to argue this is a board-specific quirk — it's the firmware blob's peer-handling path on multi-VDEV 2.4GHz radios. Good to have this pinned down across two device families now.