openwrt/openwrt #17354: Typo in LuCi config page for 2.4g wireless

JustAnotherEndUser, 2024-12-23:
### Describe the bug

In LuCi wireless config, there are AX references in 2.4Ghz radio area.  Both in the description and drop-down.

[image]

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ipq807x/generic

### Device

Dynalink DL-WRX36

### Image kind

Official downloaded image

### Steps to reproduce

Navigate to the Network>Wireless configuration page via LuCi.  I'm not sure if this is a typo, or would actually set conflicting / non-valid options.

### Actual behaviour

"AX" is select-able in the Wireless 2.4 setup, with channels 1-11.

### Expected behaviour

I would not expect AX to be a select-able configuration option for the 2.4g radio. 

### Additional info

See screenshot.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-12-23:
Where is the typo? The AX standard also works on the 2.4GHz band.

JustAnotherEndUser, 2024-12-23:
@namiltd - Apologies, my error.

closed 2024-12-23 by JustAnotherEndUser