openwrt/openwrt #17355: Channel scanning broken in BPI r4

EPinci, 2024-12-23:
### Describe the bug

In BPI r4, after the device path rework, channel scanning from LuCI is broken.
Might be related to https://github.com/openwrt/luci/issues/7483

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Steps to reproduce

Open LuCI, overview, channel analisys. Notice scanning result in nothing displayed on any channel.
Might be related to https://github.com/openwrt/luci/issues/7483

### Actual behaviour

Channels are not populated with present SSIDs.

### Expected behaviour

Expect to see channels populated with detected SSIDs.

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_collectd=y
CONFIG_PACKAGE_collectd-mod-cpu=y
CONFIG_PACKAGE_collectd-mod-interface=y
CONFIG_PACKAGE_collectd-mod-ipstatistics=y
CONFIG_PACKAGE_collectd-mod-iwinfo=y
CONFIG_PACKAGE_collectd-mod-load=y
CONFIG_PACKAGE_collectd-mod-memory=y
CONFIG_PACKAGE_collectd-mod-network=y
CONFIG_PACKAGE_collectd-mod-rrdtool=y
CONFIG_PACKAGE_collectd-mod-thermal=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_ddns-scripts=y
CONFIG_PACKAGE_ddns-scripts-services=y
CONFIG_PACKAGE_dns-over-https=y
CONFIG_PACKAGE_drill=y
CONFIG_PACKAGE_ethtool-full=y
CONFIG_PACKAGE_f2fs-tools=y
CONFIG_PACKAGE_ip-tiny=y
CONFIG_PACKAGE_ip6tables-nft=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_ipset=y
CONFIG_PACKAGE_iptables-mod-conntrack-extra=y
CONFIG_PACKAGE_iptables-mod-ipopt=y
CONFIG_PACKAGE_iptables-nft=y
CONFIG_PACKAGE_kmod-fs-exfat=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-fs-ntfs3=y
CONFIG_PACKAGE_kmod-fs-vfat=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ipt-conntrack=y
CONFIG_PACKAGE_kmod-ipt-conntrack-extra=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipopt=y
CONFIG_PACKAGE_kmod-ipt-ipset=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-mii=y
CONFIG_PACKAGE_kmod-nf-conncount=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-ipt6=y
CONFIG_PACKAGE_kmod-nft-compat=y
CONFIG_PACKAGE_kmod-nls-cp437=y
CONFIG_PACKAGE_kmod-nls-iso8859-1=y
CONFIG_PACKAGE_kmod-nls-utf8=y
CONFIG_PACKAGE_kmod-nvme=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_kmod-usb-net-qmi-wwan=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-wdm=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libcap-ng=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libevdev=y
CONFIG_PACKAGE_libiperf3=y
CONFIG_PACKAGE_libipset=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libldns=y
CONFIG_PACKAGE_libltdl=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_librrd1=y
CONFIG_PACKAGE_libspeedtestcpp=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libudev-zero=y
CONFIG_PACKAGE_libusb-1.0=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-mwan3=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-app-statistics=y
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-dashboard=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-qmi=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_minicom=y
CONFIG_PACKAGE_miniupnpd-nftables=y
CONFIG_PACKAGE_mwan3=y
CONFIG_PACKAGE_nano=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_rrdtool1=y
CONFIG_PACKAGE_speedtestcpp=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_usbids=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_wget-ssl=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mbedtls=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_TARGET_ROOTFS_PARTSIZE=1024
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-25:
brcmfmac radio can not scan when multiple APs are enabled. Check "combinations" in "iw list" for possible modes.

MichelHanzen, 2024-12-25:
I can confirm this too. I already opened a case but I closed it because I didn't receive any feedback.
@brada4 Can you develop your remark? It is a little bit cryptic for me.

brada4, 2025-01-06:
Check radio capabilities in `iw list` -> valid interface combinations, channel maps etc.

MichelHanzen, 2025-01-07:
@brada4 Can you develop your remark? It is a little bit cryptic for me.

brada4, 2025-01-07:
Does your device show capñbilities bits that it supports scanning while running AP.

EPinci, 2025-01-07:
...not entirely sure how to "parse" this, can you help? @brada4

```
root@OpenWrt:~# iw list
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 2131 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0xff RX 0xff
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 2 usec (0x04)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x2270ce926d09f30e4e3f00):
                                HE40/2.4GHz
                                242 tone RUs/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
                                Partial Bandwidth UL MU-MIMO
                                DCM Max Constellation: 2
                                DCM Max Constellation Rx: 2
                                SU Beamformer
                                SU Beamformee
                                Beamformee STS <= 80Mhz: 3
                                Beamformee STS > 80Mhz: 3
                                Sounding Dimensions <= 80Mhz: 1
                                Sounding Dimensions > 80Mhz: 1
                                Codebook Size SU Feedback
                                Codebook Size MU Feedback
                                Triggered CQI Feedback
[... 995 lines cut ...]
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ FILS_DISCOVERY ]: FILS discovery frame transmission support
                * [ UNSOL_BCAST_PROBE_RESP ]: unsolicated broadcast probe response transmission support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
                * [ BSS_COLOR ]: BSS coloring support
                * [ POWERED_ADDR_CHANGE ]: can change MAC address while up
        wiphy radio 0:
                freq range: 2400.0 MHz - 2500.0 MHz
                valid interface combinations:
                         * #{ AP, mesh point } <= 16, #{ managed } <= 19,
                           total <= 19, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

        wiphy radio 1:
                freq range: 5000.0 MHz - 5900.0 MHz
                valid interface combinations:
                         * #{ AP, mesh point } <= 16, #{ managed } <= 19,
                           total <= 19, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

        wiphy radio 2:
                freq range: 5900.0 MHz - 7200.0 MHz
                valid interface combinations:
                         * #{ AP, mesh point } <= 16, #{ managed } <= 19,
                           total <= 19, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }
```

brada4, 2025-01-07:
It is a bit wild to have 19 client sessions, but ok, it should be able to scan. 19-number_of_ap-s times...

brada4, 2025-01-08:
Interface combinations numbers are far from typical with only one client permitted usually. I dont know whether it is a bug or feature. Either way mt76 was updated in 2410-rc5 again

EPinci, 2025-01-08:
Yeah, I'm building and updating daily but is still broken. :(

brada4, 2025-01-08:
Device capabilities indicate "it works", the command line equivalent of luci is
```
ubus call iwinfo survey ' {"device":"phy0-ap0"} '
```
Other is scan to connect in Network/Wireless
```
ubus call iwinfo scan ' {"device":"phy0-ap0"} '
```

EPinci, 2025-01-08:
Uhm...

```
root@OpenWrt:~# ubus call iwinfo survey ' {"device":"phy0-ap0"} '
root@OpenWrt:~# ubus call iwinfo scan ' {"device":"phy0-ap0"} '
Command failed: Not found
```

MichelHanzen, 2025-01-13:
@EPinci  I succeded to make the "Channel Analysis' work.
[image]
Initially, my wirelles config was made country code 'JP'. I changed the country to 'BE' (I live in Belgium) and I had to restrict the 6Ghz channel width to 160 Mkz. Please find herafter my /etc/config/wireless:

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/11300000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '2g'
        option channel '6'
        option htmode 'EHT20'
        option cell_density '0'
        option country 'BE'
        option disabled '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Mhanzen'
        option encryption 'sae-mixed'
        option key 'xxxxxxxx'
        option ocv '0'
        option wds '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'soc/11300000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option radio '1'
        option band '5g'
        option channel '36'
        option htmode 'EHT160'
        option cell_density '0'
        option country 'BE'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Mhanzen'
        option encryption 'sae'
        option key 'xxxxxxxx'
        option ocv '0'

config wifi-device 'radio2'
        option type 'mac80211'
        option path 'soc/11300000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option radio '2'
        option band '6g'
        option channel '37'
        option htmode 'EHT160'
        option cell_density '0'
        option country 'BE'
        option disabled '0'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'Mhanzen'
        option encryption 'sae'
        option key 'xxxxxxxx'
        option ocv '0'

MichelHanzen, 2025-01-24:
Channel scan doesn't work anymore with RC6.

brada4, 2025-01-24:
A bit strange observation - unless you have AP in particular regdom up you only "scan" ch36 in 5ghz.

EPinci, 2025-01-24:
Yeah, something is definitely off for BPI r4 and has been ever since the driver splitted the radios.
I don't think it is related to configuration and certainly not related to country selection (IT in my case).

I have 2 openwrt devices running:
- **they both run my own daily snapshot of the openwrt-24.10 branch** - so same context
- they are on different floors so actual results can be different

In my **GL MT3000** (Filogic AX device) channel scanning is working.

[image] 

In my **BPI r4** (Filogic BE device) is not: messy radios identification and empty charts (not even its own SSID)

(same snapshot as above from @MichelHanzen)

I have noticed though that scanning is never immediately responsive after a config change - no, I'm not using any "auto" channel selection - so tests need to be made "a while" after any wireless config changes.

fridtjof, 2025-01-26:
This is due to the linux wireless subsystem's architecture changes for MLO, which mandates combining all the radios under a single PHY.

With that came a few changes to interface names, which explains why the ubus calls suggested by @brada4 failed. (luci also seems to set off these calls using the UCI names like radio0,1,2 which iwinfo will try to convert to interface names)

iwinfo currently does not handle this UCI -> interface name lookup correctly, and doing it correctly seems to require information the kernel currently does not give. Otherwise it would need to resort to OpenWRT specific measures like duplicating interface naming logic from netifd/wifi-scripts in order to figure out the correct interface name.

For more detail see here, where I wrote up my findings: https://github.com/openwrt/luci/issues/7483#issuecomment-2614047082

brada4, 2025-01-26:
I dug up the calls tapping on 23.05 luci ;-)