openwrt/openwrt #17366: Bridge port isolation broken on Zyxel GS1900-24E

fr-em, 2024-12-24:
### Describe the bug

Started with two of these switches, trying to create a guest and IOT network with isolated clients. 
Used bridge isolation to achive this. Configured the switches, but the ports do not get in an isolated state.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

realtek/rtl838x

### Device

ZyXEL GS1900-24E

### Image kind

Official downloaded image

### Steps to reproduce

Installed additional package kmod-br-netfilter
Rebooted switch
activate isolation on multiple ports from default bridge (switch)

### Actual behaviour

Connected devices to these switches on the defined ports, devices can talk to each other, independed from the state of the isolation option.

Also tested this with bridged VLANS (single VLAN for every port), bridged together and added isolate option. Works as aspected but seems to be done on the CPU. The network speed is terrible with this solution. Only arround 5MB/s in single directional traffic possible.

Used config:

```
config device 'switch'
        option name 'switch'
        option type 'bridge'
        option macaddr 'xx:xx...'
        option ipv6 '0'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'lan5'
        list ports 'lan6'
        list ports 'lan7'
        list ports 'lan8'
        list ports 'lan9'
        list ports 'lan10'
        list ports 'lan11'
        list ports 'lan12'
        list ports 'lan13'
        list ports 'lan14'
        list ports 'lan15'
        list ports 'lan16'
        list ports 'lan17'
        list ports 'lan18'
        list ports 'lan19'
        list ports 'lan20'
        list ports 'lan21'
        list ports 'lan22'
        list ports 'lan23'
        list ports 'lan24'

config bridge-vlan
        option device 'switch'
        option vlan '30'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'lan4:t'
        list ports 'lan15'
        list ports 'lan16'

config device
        option name 'lan15'
        option isolate '1'

config device
        option name 'lan16'
        option isolate '1'

```

### Expected behaviour

The 'isolated' option should isolate the bridge ports and inhibit traffic between isolated ports

### Additional info

Also tested with original switch firmware, works as espected

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2024-12-27:
@RussellSenior, @0xharshal, @bmork, @jwh, @plappermaul, @andi-at-1, @casipw, @Hurricos, @dangowrt, @svanheule, @joe2xyz, @luizluca, @brada4, @GoetzGoerisch, @paulmenzel, @KanjiMonster, @cryptage21, @ben-tiky, @davischw, @thomas-mm, @kvuorine, @Borromini, @MaxRower, @ynezz, @DragonBluep, @robimarko, @hauke, @mirko, @musashino205, @rudolfvesely, @blogic, @RaylynnKnight, @adschm, @oliv3r, @zambelly, @mrnuke, @hardfalcon, @sudoBash418, @zambelly, @pkgadd, @andyboeh, @DavideFioravanti,  : What do you think?

Neustradamus, 2025-01-01:
@svanheule: What do you think about @fr-em ticket?

svanheule, 2025-01-02:
The VLAN/bridge support is known to be a bit rough, but I don't know the networking code well enough to find out where the problem may be.

plappermaul, 2025-01-05:
https://github.com/plappermaul/realtek-doc might give some more insights.

fr-em, 2025-02-12:
I tested a little bit more and migrated meanwhile to OpenWrt 24.10.0 r28427. 
I found also, that on the switch, the port isolation can be choosen in LUCI without installing packages like kmod-br-netfilter. But as in 23.05 it has no effect on the communication if port isolation is switchted on. 

I created new VLANs like the default vlan 1 as shown below. This is different from creating via LUCI, which will create the VLAN as 802.1q interface and not as plain network interface.

```
# new VLAN configured like VLAN 1
config bridge-vlan 'guest_vlan'
        option device 'switch'
        option vlan '31'
        list ports 'lan1:t'
        list ports 'lan21'
        list ports 'lan22'
        list ports 'lan23'
        list ports 'lan24'

config device
        option name 'switch.31'
        option macaddr 'e8:37:7a:a2:0a:9f'
        option ipv6 '0'
        option mtu '1500'
        option txqueuelen '1000'
        option promisc '0'

# vlan created via Luci
config bridge-vlan
        option device 'switch'
        option vlan '32'

```
Further I found out, that the option of any new created VLAN (as network device or 802.1q interface) has no value for promiscuous mode shown in Luci, while for default vlan 1 its shown.

Disabling the promiscuous mode on the vlan results in the wanted result, that there is no communication possible on isolated ports. 

Is it possible that something went wrong here?

Default VLAN: 
[image]

New created VLAN:
[image]