openwrt/openwrt #17368: x86_x64 builds requires CONFIG_TARGET_SERIAL be set!.  Stop.

mullengers, 2024-12-24:
### Describe the bug

make[3]: Entering directory '/openwrt/package/base-files'
/openwrt/target/linux/x86/base-files.mk:3: *** This platform requires CONFIG_TARGET_SERIAL be set!.  Stop.
make[3]: Leaving directory '/openwrt/package/base-files'

OpenWrt 22 and 23 has no problem with the serial port being set to "" (blank)
CONFIG_GRUB_SERIAL=""

This requirement is no good for Dell Wyse 3040 clients as they do not have PC serial port and GRUB will not boot.

### OpenWrt version

24.10-rc4 (all RCs actually)

### OpenWrt release

24.10-rc4 (all RCs actually)

### OpenWrt target/subtarget

target-x86_64_musl/root-x86

### Device

/dev/ttyS0 does not exist on all PCs platforms!

### Image kind

Self-built image

### Steps to reproduce

in OpenWrt 22 and 23 was:
.config to contain CONFIG_GRUB_SERIAL="" => PASS and all OK

Now in OpenWrt 24.10
.config to contain CONFIG_TARGET_SERIAL="" => FAIL and build does not finish

### Actual behaviour

time: package/system/usign/compile#0.27#0.12#0.34
make[3]: Entering directory '/openwrt/package/base-files'
/openwrt/target/linux/x86/base-files.mk:3: *** This platform requires CONFIG_TARGET_SERIAL be set!.  Stop.
make[3]: Leaving directory '/openwrt/package/base-files'
time: package/base-files/compile#0.54#0.03#0.58
    ERROR: package/base-files failed to build.
make[2]: *** [package/Makefile:179: package/base-files/compile] Error 1
make[2]: Leaving directory '/openwrt'
make[1]: *** [package/Makefile:173: /openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/openwrt'
make: *** [/openwrt/include/toplevel.mk:248: world] Error 2

### Expected behaviour

Should have built an x86_x64 PC image

### Additional info

Dell Wyse 3040 Thin Clients can't boot if /dev/ttyS0 is required - Openwrt 22 and 23.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vincejv, 2024-12-25:
did you turn that flag off through make menuconfig or you manually edited the .config file?
because manually editing it could cause issues like these as it won't disable the dependencies that require that.

mullengers, 2024-12-25:
Hi,I did everything by the book I promise.Yes used the make menu config method.To be exact I normally use the diff config thing (where you can use the tooling add a tiny bit of desired config to the main .config) this is how I make different builds. But it failed. Works fine on 22 and 23 as I said.So then I tried make menuconfig directly on the 24rc4 and it failed.I built a second Debian 12 host PC, with the normal build environment and tried menu config again on a clean git hub pull of 24.10-rc4 and it failed too.I think someone has turned this non-blank / needs “ttyS0” requirement on in 24 but I don’t know where to look to definitely say that is the case. Also another cause might be, note that the variable name changed from 22/23 to 24, maybe an allow-to-be-blank rule for the old variable name is not working any more…Best regardsSimonSent from my iPad

mullengers, 2024-12-25:
Testing: 24.10rc4 - I left the CONFIG_TARGET_SERIAL as its default “ttyS0” and the images all built. I then installed a squashfs efi imagine onto a Dell Wyse 3040 and rebooted. The GRUB hangs with a one liner “GRUB booting” or something - is an expected FAIL. During the initial grub 5 second count down, I can press ‘e’ and temporarily edit the grub config and delete the console=ttyS0 settings and the device will then boot. Once in OpenWrt 24, one then needs to use nano to permanently edit the grub.cfg settings to do the same. The device then boots normally every time without issue on OpenWrt 24.10rc4.So the setting should not be made mandatory - blank “” is OK.Best regardsSimonSent from my iPad