openwrt/openwrt #17369: Linksys MX4200 failed to boot after from 24.10.0-rc2 to 24.10.0-rc4

katon888, 2024-12-25:
### Describe the bug

Linksys MX4200 v2 doesn't boot after upgrade from 24.10.0-rc2 to 24.10.0-rc4

from:
openwrt-24.10.0-rc2-qualcommax-ipq807x-linksys_mx4200v2-squashfs-sysupgrade.bin
to:
openwrt-24.10.0-rc4-qualcommax-ipq807x-linksys_mx4200v2-squashfs-sysupgrade.bin

below serial logs are observed with sysupgrade failed

`Wed Dec 25 06:22:40 UTC 2024 upgrade: Performing system upgrade...
verifying sysupgrade tar file integrity
Unlocking kernel ...

Writing from <stdin> to kernel ...
[   65.736485] ubi1: attaching mtd22
[   66.564198] ubi1: scanning is finished
[   66.572287] ubi1: attached mtd22 (name "rootfs", size 144 MiB)
[   66.572333] ubi1: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[   66.577044] ubi1: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[   66.583884] ubi1: VID header offset: 2048 (aligned 2048), data offset: 4096
[   66.590749] ubi1: good PEBs: 1152, bad PEBs: 0, corrupted PEBs: 0
[   66.597543] ubi1: user volume: 1, internal volumes: 1, max. volumes count: 128
[   66.603763] ubi1: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 808907754
[   66.610903] ubi1: available PEBs: 0, total reserved PEBs: 1152, PEBs reserved for bad PEB handling: 80
[   66.619848] ubi1: background thread "ubi_bgt1d" started, PID 5446
UBI device number 1, total 1152 LEBs (146276352 bytes, 139.5 MiB), available 0 LEBs (0 bytes), LEB size 126976 bytes (124.0 KiB)
ubimkvol: error!: UBI device does not have free logical eraseblocks
cannot create rootfs volume
sysupgrade failed
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource busy
[   67.135616] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
[   67.634292] reboot: Restarting system`

### OpenWrt version

unable to check, device unable to boot

### OpenWrt release

 24.10.0-rc4

### OpenWrt target/subtarget

unable to check, device unable to boot

### Device

Linksys MX4200v2

### Image kind

Official downloaded image

### Steps to reproduce

Linksys MX4200v2 running 24.10.0-rc2 "openwrt-24.10.0-rc2-qualcommax-ipq807x-linksys_mx4200v2-squashfs-sysupgrade.bin", using LuCI to upgrade to 24.10.0-rc4 "openwrt-24.10.0-rc4-qualcommax-ipq807x-linksys_mx4200v2-squashfs-sysupgrade.bin"

### Actual behaviour

Device unable to boot to system

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

katon888, 2024-12-26:
> Invalid Version reported. `unable to check, device unable to boot` Is this from a clean repository?

yes, 24.10.0-rc2 r28161-ea17e958b9

> Invalid Target/Subtarget reported.. `unable to check, device unable to boot` Is this from a supported device?

yes, qualcommax/ipq807x, Linksys MX4200v2

robimarko, 2024-12-26:
I doubt this has anything to do with 24.10rc4 in particular.
If you try and sysupgrade to a snapshot, does it work then?

Also, can you grab the output of `ubinfo -a`?

testuser7, 2024-12-27:
@katon888 Did you installed OpenWrt on both partitions before updating?

innovara, 2024-12-29:
For what it's worth, I've just built and flashed the factory image that I've built from main (OpenWrt SNAPSHOT r28443-9ea174c7bf / LuCI Master 24.360.56502~2da1efb) to both partitions of my MX4200v2 test unit. Then flashed `openwrt-24.10.0-rc4-qualcommax-ipq807x-linksys_mx4200v2-squashfs-sysupgrade.bin` with LuCI from one of the partitions without preserving settings and it's worked as expected.

The issue reported might have to do with flashing from rc2, or some settings carried forward, but it doesn't seem to be directly related to, or introduced by, rc4.

katon888, 2024-12-30:
The result is still the same on my device:

1. I flashed back to OEM stock firmware (2.0.6.211172)
2. Flash snapshot fw (r28242-1eff737906) in OEM FW's GUI.
3. Install LuCI after r28242-1eff737906 boot up
4. flash 24.10.0-rc4 sysupgrade fw from LuCI.

One question, in OEM stock fw, how to flash both partition from GUI?

innovara, 2024-12-30:
> One question, in OEM stock fw, how to flash both partition from GUI?

You can't. OEM will flash to the other partition that is not currently running from. That's the point of dual partitions, leaving a partition available that is known to work. Follow the installation instructions on OpenWrt's website for this device (https://openwrt.org/toh/linksys/mx4200_v1_and_v2) and make sure that you have OpenWrt installed on both partitions. Note that you use '-factory' images when you flash with mtd. Once you are positive that you have it installed on both partitions and can switch to either of them and the router boots, attempt the sysupgrade installation from one of them.

One thing I've done in the past when things got a bit strange while flashing as in your case, was to flash OEM, boot OEM, flash OEM again. That way the router would have OEM on both partitions which is the closest you can get to restoring things as they were. Added a reset for good measure to clear the settings partition as well. Then just followed the steps as per above.

katon888, 2025-01-02:
@innovara Thanks for your explanation.
This issue can be closed as invalid, because I overlooked an step in the installation guideline, which is an important step.

My device works flawlessly with below steps:

1. I flashed back to OEM stock firmware (2.0.6.211172)
2. Flash snapshot fw (r28242-1eff737906) in OEM FW's GUI.
3. ssh into DUT running openwrt snapshot fw image.
4. Follow installation guideline to flash snapshot fw image to alterative partition.
5. Install LuCI 
6. flash 24.10.0-rc4 sysupgrade fw from LuCI.

closed 2025-01-02 by katon888