openwrt/openwrt #17370: Kernel Warning on 2.4GHz driver for Archer C5 v4 (mt7620)

mrkabali, 2024-12-25:
### Describe the bug

There are frequent kernel warning messages as seen below on the logs. Although it doesn't seem to have any dterminental effect on the funtioning of the device.

Wed Dec 25 11:42:53 2024 kern.warn kernel: [14138.552053] ieee80211 phy1: rt2800_watchdog_dma_busy: Warning - Watchdog TX DMA busy detected
Wed Dec 25 11:42:53 2024 kern.info kernel: [14138.563433] ieee80211 phy1: Hardware restart was requested
Wed Dec 25 11:42:53 2024 kern.warn kernel: [14138.981095] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 0 failed to flush
Wed Dec 25 11:42:54 2024 kern.warn kernel: [14139.549121] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 2 failed to flush
Wed Dec 25 11:52:59 2024 kern.warn kernel: [14408.317824] ieee80211 phy1: rt2800_watchdog_dma_busy: Warning - Watchdog TX DMA busy detected
Wed Dec 25 11:52:59 2024 kern.info kernel: [14408.326570] ieee80211 phy1: Hardware restart was requested
Wed Dec 25 11:53:00 2024 kern.warn kernel: [14408.469212] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 0 failed to flush
Wed Dec 25 11:53:00 2024 kern.warn kernel: [14408.637834] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 2 failed to flush

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

ramips/mt7620

### Device

TP-Link Archer C5 v4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-12-25:
There is already a fix for this: https://github.com/openwrt/openwrt/pull/16845. I tested it on this router. It should be committed before the next release.

mrkabali, 2024-12-28:
@namiltd Thanks, will try the fix. This issue has become a bit severe now with multiple restarts a day.

mrkabali, 2025-03-10:
Just so if it could be useful.

After upgrading to 24.0.10 official build, noticed that the "hardware start" error still occuring and it triggers the restart of the whole router.

namiltd, 2025-03-10:
Try: `echo 0 > /proc/sys/kernel/randomize_va_space`

The problem is described in: https://github.com/openwrt/openwrt/issues/16396

mrkabali, 2025-03-10:
Thank you so much @namiltd  for the lighting response. Much appriciated.

Will try this and see how it goes.

namiltd, 2025-03-10:
You can also use an additional script that will reset the WiFi only:
```
#!/bin/sh

# A simple script to workaround MT7620 wireless hang issue via
# crontab schedule:
# * * * * * /sbin/check_radio.sh 
# * * * * * sleep 30 && /sbin/check_radio.sh

minilog=`logread 2>/dev/null | grep "Hardware restart was requested"`

(echo $minilog | grep -q "phy0: Hardware restart was requested") && phy0=true 
(echo $minilog | grep -q "phy1: Hardware restart was requested") && phy1=true

if [ "$phy0" == "true" ] || [ "$phy1" == "true" ]; then

    /etc/init.d/log restart

    [ "$phy0" == "true" ] && date >>/tmp/radio0_restart.log
    [ "$phy1" == "true" ] && date >>/tmp/radio1_restart.log

    [ "$phy0" == "true" ] && wifi down radio0
    [ "$phy1" == "true" ] && wifi down radio1

    sleep 2

    [ "$phy0" == "true" ] && wifi up radio0
    [ "$phy1" == "true" ] && wifi up radio1
fi
```

Add `/sbin/check_radio.sh` to /etc/sysupgrade.conf

mrkabali, 2025-03-12:
Thank you, will try this.

And with the workaround, no crash/restart for the last 2 days, thats a relief.
However time to time below still happens. Is it a known issue on driver / kernel?

Tue Mar 11 22:03:27 2025 kern.warn kernel: [46361.732629] ieee80211 phy1: rt2800_watchdog_dma_busy: Warning - Watchdog TX DMA busy detected
Tue Mar 11 22:03:27 2025 kern.info kernel: [46361.732676] ieee80211 phy1: Hardware restart was requested

namiltd, 2025-03-12:
The problem with "Hardware restart was requested" is known, it was partially solved with patches but sometimes it still occurs. There is a script for this from: https://github.com/openwrt/openwrt/issues/17370#issuecomment-2710217242 [Hardware restart was requested](https://github.com/openwrt/openwrt/issues/17370#issuecomment-2710217242). Thanks to it, WiFi reloads completely and goes back to work.

namiltd, 2025-03-12:
The problem with "Hardware restart was requested" is known, it was partially solved with patches but sometimes it still occurs. There is a script for this from: https://github.com/openwrt/openwrt/issues/17370#issuecomment-2710217242 [Hardware restart was requested](https://github.com/openwrt/openwrt/issues/17370#issuecomment-2710217242). Thanks to it, WiFi reloads completely and goes back to work. Additionally, the date of this reload is recorded in the files /tmp/radio0_restart.log and /tmp/radio1_restart.log.

mrkabali, 2025-03-12:
Thanks, hopefuly the last question.

Why do we need two entries in crontab for the script?

mrkabali, 2025-03-13:
Closing with the workaround script in play for automatic PHY restart. Hopefuly there will be a permenant fix someday. Thank you @namiltd for the support.