openwrt/openwrt #17371: [24.10.0-rc4] mt7621/TP-Link AX23 No Internet after update from 23.05.5

vadim-brave, 2024-12-25:
### Describe the bug

Router: TP-Link AX23

After update from 23.0.5.5 to 24.10.0 RC4 there is no connection to the Internet on one of the PCs. Its Windows 10 PC (network is not recognized). On Linux with EndeauvourOS Internet is working after update.

Reset to defaults either via Luci or via command line does not resolve this problem.

After downgrading back to version 23.05.5, Internet starts working as it should.

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

ramips/mt762

### Device

TP-Link Archer AX23 v1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-12-25:
If you can, please attach kernel logs from v24.10 and 23.05 (as attachment).

vadim-brave, 2024-12-25:
> If you can, please attach kernel logs from v24.10.0-rc4 and v23.05.5 (as attachment). Additionally, output of commands (anonymize addresses): ifconfig ip route

Where to enter commands?

vadim-brave, 2024-12-25:
> If you can, please attach kernel logs from v24.10.0-rc4 and v23.05.5 (as attachment). Additionally, output of commands (anonymize addresses): ifconfig ip route

https://pastebin.com/Yg04GN4y

23.0.5.5 kernel log

namiltd, 2024-12-25:
> Where to enter commands?

In the SSH terminal

vadim-brave, 2024-12-26:
> > Where to enter commands?
> 
> In the SSH terminal

23.05.5

**kernel log**

https://0.0g.gg/?563e70da99b72598#-7AiEaUnZRnJzqp2MBXa33vGm3WLNQF1qTBnBpiX9eUSo

**ifconfig**

br-lan    Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fda3:994b:abbb::1/60 Scope:Global
          inet6 addr: 2a02:ad8:389a:6100::1/60 Scope:Global
          inet6 addr: fe80::1227:f5ff:fe09:ea51/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1214912 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1865224 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:212301434 (202.4 MiB)  TX bytes:2016691449 (1.8 GiB)

eth0      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          inet6 addr: fe80::1227:f5ff:fe09:ea51/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1504  Metric:1
          RX packets:1587462 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3669997 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:284820353 (271.6 MiB)  TX bytes:3628105889 (3.3 GiB)
          Interrupt:21 

lan1      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:134198 errors:0 dropped:57 overruns:0 frame:0
          TX packets:212094 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:36266776 (34.5 MiB)  TX bytes:138669109 (132.2 MiB)

lan2      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:320669 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2088502 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:20824635 (19.8 MiB)  TX bytes:2124798046 (1.9 GiB)

lan3      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:302613 errors:0 dropped:0 overruns:0 frame:0
          TX packets:275122 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:62958690 (60.0 MiB)  TX bytes:122763718 (117.0 MiB)

lan4      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:71190 errors:0 dropped:0 overruns:0 frame:0
          TX packets:50851 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7014438 (6.6 MiB)  TX bytes:6108749 (5.8 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:725 errors:0 dropped:0 overruns:0 frame:0
          TX packets:725 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:87012 (84.9 KiB)  TX bytes:87012 (84.9 KiB)

phy0-ap0  Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          inet6 addr: fe80::1227:f5ff:fe09:ea51/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:381457 errors:0 dropped:0 overruns:0 frame:0
          TX packets:540101 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:117201906 (111.7 MiB)  TX bytes:577156077 (550.4 MiB)

phy1-ap0  Link encap:Ethernet  HWaddr 10:27:F5:09:EA:50  
          inet6 addr: fe80::1227:f5ff:fe09:ea50/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:745147 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1186128 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:97901685 (93.3 MiB)  TX bytes:1368093272 (1.2 GiB)

pppoe-PPPOE Link encap:Point-to-Point Protocol  
          inet addr:109.165.19.232  P-t-P:178.34.128.50  Mask:255.255.255.255
          inet6 addr: 2a02:ad8:91:c2e3:6062:9c7d:38e7:bb28/64 Scope:Global
          inet6 addr: fe80::6062:9c7d:38e7:bb28/128 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1492  Metric:1
          RX packets:274637 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1644008 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3 
          RX bytes:88598909 (84.4 MiB)  TX bytes:304643533 (290.5 MiB)

wan       Link encap:Ethernet  HWaddr 10:27:F5:09:EA:52  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5525762 errors:10 dropped:22865 overruns:0 frame:0
          TX packets:2848353 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5601357857 (5.2 GiB)  TX bytes:543394231 (518.2 MiB)
          Interrupt:21 

wan-bridge Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          inet6 addr: fe80::1227:f5ff:fe09:ea51/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:898749 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1724478 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:780694437 (744.5 MiB)  TX bytes:343226553 (327.3 MiB)

**ip route**

default via 178.34.128.50 dev pppoe-PPPOE 
178.34.128.50 dev pppoe-PPPOE scope link  src 109.165.19.232 
192.168.1.0/24 dev br-lan scope link  src 192.168.1.1

vadim-brave, 2024-12-26:
> If you can, please attach kernel logs from v24.10.0-rc4 and v23.05.5 (as attachment). Additionally, output of commands (anonymize addresses): ifconfig ip route

24.10.0-rc4

**kernel log**

https://0.0g.gg/?c3bd10bfc003307a#-4EkRE546NLaotMXZmbkBEMMMC4BLmTJNzyQXej8koYe7

**ifconfig**

br-lan    Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::1227:f5ff:fe09:ea51/64 Scope:Link
          inet6 addr: fda3:994b:abbb::1/60 Scope:Global
          inet6 addr: 2a02:ad8:382b:e200::1/60 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:14814 errors:0 dropped:0 overruns:0 frame:0
          TX packets:26283 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1178728 (1.1 MiB)  TX bytes:29381445 (28.0 MiB)

eth0      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          inet6 addr: fe80::1227:f5ff:fe09:ea51/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1504  Metric:1
          RX packets:15262 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22460 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1495381 (1.4 MiB)  TX bytes:16412847 (15.6 MiB)
          Interrupt:19 

lan1      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:182 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:23156 (22.6 KiB)

lan2      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:201 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2111 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:33109 (32.3 KiB)  TX bytes:2087845 (1.9 MiB)

lan3      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8998 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10162 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:915537 (894.0 KiB)  TX bytes:13363761 (12.7 MiB)

lan4      Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6063 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5776 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:485687 (474.3 KiB)  TX bytes:474559 (463.4 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:523 errors:0 dropped:0 overruns:0 frame:0
          TX packets:523 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:50607 (49.4 KiB)  TX bytes:50607 (49.4 KiB)

phy0-ap0  Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:75 errors:0 dropped:0 overruns:0 frame:0
          TX packets:307 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:8926 (8.7 KiB)  TX bytes:57519 (56.1 KiB)

phy1-ap0  Link encap:Ethernet  HWaddr 10:27:F5:09:EA:50  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3030 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11693 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:309148 (301.9 KiB)  TX bytes:15758999 (15.0 MiB)

pppoe-PPPOE Link encap:Point-to-Point Protocol  
          inet addr:100.116.180.32  P-t-P:178.34.128.50  Mask:255.255.255.255
          inet6 addr: 2a02:ad8:91:e8a8:e80f:6171:fac6:9b4e/64 Scope:Global
          inet6 addr: fe80::e80f:6171:fac6:9b4e/128 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1492  Metric:1
          RX packets:1530 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5700 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3 
          RX bytes:1343184 (1.2 MiB)  TX bytes:621500 (606.9 KiB)

wan       Link encap:Ethernet  HWaddr 10:27:F5:09:EA:52  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:16257 errors:0 dropped:46 overruns:0 frame:0
          TX packets:6268 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:19519510 (18.6 MiB)  TX bytes:852760 (832.7 KiB)
          Interrupt:19 

wan-bridge Link encap:Ethernet  HWaddr 10:27:F5:09:EA:51  
          inet6 addr: fe80::1227:f5ff:fe09:ea51/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3543 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5979 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3372989 (3.2 MiB)  TX bytes:756982 (739.2 KiB)

**ip route**

default via 178.34.128.50 dev pppoe-PPPOE 
178.34.128.50 dev pppoe-PPPOE scope link  src 100.116.180.32 
192.168.1.0/24 dev br-lan scope link  src 192.168.1.1

vadim-brave, 2024-12-26:
Only Windows PC dont have connection to Internet - Network is not recognized

Linux PCs, IPTV, devices on Wi-Fi - all is good

vadim-brave, 2024-12-29:
> If you can, please attach kernel logs from v24.10.0-rc4 and v23.05.5 (as attachment). Additionally, output of commands (anonymize addresses): ifconfig ip route

Have you looked at the logs and commands outputs?

namiltd, 2024-12-29:
Links to kernel logs are no longer active

hauke, 2024-12-29:
Do the interface still get the same mac addresses with OpenWrt 24.10 as in 23.05? Please compare the mac addresses of the interfaces with the mac addresses on the label of the router box and with the mac addresses you got with OpenWrt 23.05. Maybe the mac address changed or you get random mac addresses now.

vadim-brave, 2025-01-01:
> Links to kernel logs are no longer active. However, I suspect that the speed detection on lan1 does not work.

Yep

Speed recognition not working

Maybe its the culprit?

vadim-brave, 2025-01-01:
> Do the interface still get the same mac addresses with OpenWrt 24.10 as in 23.05? Please compare the mac addresses of the interfaces with the mac addresses on the label of the router box and with [...]

OK, will check, when return from holidays

vadim-brave, 2025-01-08:
> Links to kernel logs are no longer active. However, I suspect that the speed detection on lan1 does not work.

[image]

It seems like speed recognition on lan1 not working

Im thinking its the cause

Do you know how to fix it or how to send this to developers to fix?

vadim-brave, 2025-01-08:
> Do the interface still get the same mac addresses with OpenWrt 24.10 as in 23.05? Please compare the mac addresses of the interfaces with the mac addresses on the label of the router box and with [...]

Nope

MACs good

Its the same on 23.05 and 24.10

Its the problem with lan1 port speed not detected on 24.10 series

vadim-brave, 2025-01-09:
> Links to kernel logs are no longer active. However, I suspect that the speed detection on lan1 does not work.

Do you know how to fix speed recognition on lan1?

wenqingj, 2025-01-18:
I happen to have the same device (TP-Link Archer AX23) and it runs the same version (24.10.0-rc5) well without issues for weeks now. I set the WAN interface "user broadcast flag" checked (in the "advanced settings" tab), and I disabled the IP6 features. I had the wifi running on a 2.5GHz channel only (not using 5 GHz) using b/g/n/ax (no ac). Other settings are on their defaults. Not sure if this is useful...

vadim-brave, 2025-01-19:
> I happen to have the same device (TP-Link Archer AX23) and it runs the same version (24.10.0-rc5) well without issues for weeks now. I set the WAN interface "user broadcast flag" checked (in the [...]

Thx for answer

Will try using broadcast flag when new rc or maybe next is star out

How youbdiaable ipv6 features?

wenqingj, 2025-01-20:
> How you disable ipv6 features?

in Luci select network -> interfaces -> lan -> DHCP server tab -> IPv6 settings tab, and make checkboxes unchecked, and all drop downs use "disabled" option if that option shows up in list.

Not sure if that's a good way. I disabled it not because IPv6 is giving me any trouble, but because IPv6 is not necessary if you have IPv4 already.

wenqingj, 2025-01-21:
Sorry I didn't read through all the comments. Now that I do, I also noticed that my lan1 port speed is 10Mbps while the connected Ethernet card spec is 1Gbps. I disconnected and then connected it again, and the speed is back to 1000Mbps. Weird.

vadim-brave, 2025-01-21:
> Sorry I didn't read through all the comments. Now that I do, I also noticed that my lan1 port speed is 10Mbps while the connected Ethernet card spec is 1Gbps. I disconnected and then connected it again, and the speed is back to 1000Mbps. Weird.

Yep

I'm also don't know what to do

Will test with broadcast opion, when new openwrt arrive

Very weird, that I'm only not having internet connection on Windows PC (Network is not recognised)

On Linux PCs and on Android through Wi-Fi all working good on 24.10 series

Something is screwed up in 24.10

vadim-brave, 2025-01-24:
> Sorry I didn't read through all the comments. Now that I do, I also noticed that my lan1 port speed is 10Mbps while the connected Ethernet card spec is 1Gbps. I disconnected and then connected it again, and the speed is back to 1000Mbps. Weird.

Im tryed broadcast flag in latest 24.10.0-rc6 and disble IPv6 in lan, but not helping

Windows PC is not having Internet connection

Very weird, coz you had all working good

Im had Linux PCs working good and Wi-Fi all is normal, STB for TV receiving internet too

Only Windows PC not having Internet connection...

Very frustrating...

Now im goung back to 23.0.5.5...

wenqingj, 2025-01-28:
@vadim-brave rc7 is coming out, and the following fix, checked in just in time to be included in rc7, might be related to your problem with lan1:

https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=a5eb5ee1a633f67ae44e43f211396a9f734268be

vadim-brave, 2025-01-30:
> [@vadim-brave](https://github.com/vadim-brave) rc7 is coming out, and the following fix, checked in just in time to be included in rc7, might be related to your problem with lan1:
> 
> https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=a5eb5ee1a633f67ae44e43f211396a9f734268be

Im trued rc7 and had a big hope for fix

But.. Nothing. Stupid Windows 10 PC cant receive IP from OpenWrt and not visible in connection status

No Internet - Network not recognized

On Linux and Android Internet running fast...

wenqingj, 2025-01-30:
@vadim-brave Is it possible that the issue is on your Windows 10 PC instead? Do you have another Windows 10 device to test? Pity I don't have any at my disposal to test it out.

wenqingj, 2025-01-30:
@vadim-brave Sorry I forgot that you mentioned things were good with 23.05.5. Hence it looks you Windows 10 PC is normal...

vadim-brave, 2025-02-05:
> [@vadim-brave](https://github.com/vadim-brave) Is it possible that the issue is on your Windows 10 PC instead? Do you have another Windows 10 device to test? Pity I don't have any at my disposal to test it out.

Can you borrow some Windows machines from friends, family neigbours or colleagues to test?

vadim-brave, 2025-03-17:
@wenqingj Can you borrow some Windows machines from friends, family neighbours or colleagues to test?

wenqingj, 2025-03-17:
@vadim-brave Sorry it seems none of my close friends has a Windows 10 machine now. But I'll keep an eye open for it.