openwrt/openwrt #17372: Unable to get IPv6-PD address after PPPoE restart caused by ISP

d-navikas, 2024-12-25:
### Describe the bug

Sometimes I see there is no IPv6 address. And if I try restart DCHPv6, it says Error: Network device is not present. Works fine after reboot.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Steps to reproduce

1. Setup manual connection via DHCPv6 using PPPoE
2. Do nothing until ISP suddenly restarts PPPoE
3. Discover broken IPv6

### Expected behaviour

IPv6 address obtained after PPPoE connection reestablishment done.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vincejv, 2024-12-25:
See: https://github.com/openwrt/openwrt/issues/16180
Use this config instead

[image]

d-navikas, 2024-12-25:
Thank you. I've changed device from pppoe-wan to "@wan" alias. IPv6 address obtained now without restart. 

@vincejv can you explain please, what the difference? I'm not OpenWrt power user.

Neustradamus, 2024-12-27:
@vincejv: It is not possible to add a fix in main and 24.10/23.05?

vincejv, 2024-12-27:
@Neustradamus the fix is to use the alias instead of the device in your config. it's a config issue.

vincejv, 2024-12-27:
@d-navikas no idea, i'll leave it to the experts on what's the difference... but based on openwrt wiki, use alias for cases wherein a device requires multiple ip addresses, either static or via DHCP.

nczyw, 2025-08-11:
I saw someone analyze the cause as follows: the ISP forces a reconnect, and during the WAN reconnection process, the virtual wan6 has not completely shut down before the WAN is already back online. As a result, wan6 does not re-dial and gets lost. I’ve encountered this issue as well. Version: 24.10.2, x86 system.

An unofficial but rather brute-force solution is to add sleep 3 at the end of /lib/netifd/ppp-down to wait 3 seconds, allowing wan6 to close before bringing it back online. Of course, it would be better if the official version could fix this problem. The issue only occurs when the ISP forces a reconnect.