openwrt/openwrt #17380: Regression in the "option" serial driver with Huawei E3372 modem

patrakov, 2024-12-26:
### Describe the bug

I have a Huawei E3372h modem (reflashed to stick mode).

On 23.05.5 on Linksys E8450, it works as it should. The testcase, inspired by `3ginfo-lite`, is a chain of several AT commands, and it's the number of the AT commands that matters:

```
root@OpenWrt:~# sms_tool -D -d /dev/ttyUSB0 at "AT+CPIN?;+CSQ;+COPS=3,0;I;I;I"
AT+CPIN?;+CSQ;+COPS=3,0;I;I;I
+CPIN: READY

+CSQ: 15,99

Manufacturer: huawei
Model: E3372
Revision: 21.200.07.00.00
IMEI: <censored>
+GCAP: +CGSM,+DS,+ES

Manufacturer: huawei
Model: E3372
Revision: 21.200.07.00.00
IMEI: <censored>
+GCAP: +CGSM,+DS,+ES

Manufacturer: huawei
Model: E3372
Revision: 21.200.07.00.00
IMEI: <censored>
+GCAP: +CGSM,+DS,+ES

OK
root@OpenWrt:~# sms_tool -D -d /dev/ttyUSB0 at "ATI;I;I;I;I;I"
ATI;I;I;I;I;I
Manufacturer: huawei
Model: E3372
Revision: 21.200.07.00.00
IMEI: <censored>
+GCAP: +CGSM,+DS,+ES

<repeated 6 times as requested>

OK
```

On 24.10-rc4, on TP-Link Archer C7 v2, it fails:

```
root@OpenWrt:~# sms_tool -D -d /dev/ttyUSB0 at "AT+CPIN?;+CSQ;+COPS=3,0;I;I;I"
No response from modem.
root@OpenWrt:~# sms_tool -D -d /dev/ttyUSB0 at "ATI;I;I;I;I;I"
failed tcsetattr(/dev/ttyUSB0): I/O error
```

Shorter sequences of AT command work as they should on /dev/ttyUSB0. Longer ones still work on /dev/cdc-wdm0.

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Official downloaded image

### Steps to reproduce

```
sms_tool -D -d /dev/ttyUSB0 at "ATI;I;I;I;I;I"
```

### Actual behaviour

failed tcsetattr(/dev/ttyUSB0): I/O error

### Expected behaviour

Information response repeated 6 times

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mrkiko, 2024-12-26:
Is any message emitted in the dmesg?
Any chance this could be a power-budget related issue? You might try with an active USB hub or something.

patrakov, 2024-12-27:
Observations:

* In an x86_64 VM with OpenWrt 24.10-rc4, there is no failure.
* With a powered hub, no failure.
* With a different cable between the router and the passive hub, also no failure.

Closing as invalid.

closed 2024-12-27 by patrakov