openwrt/openwrt #17384: br-lan device gets a random MAC address after upgrade from 23.05.5 to 24.10.0rc4

skylerbunny, 2024-12-26:
### Describe the bug

When upgrading from 23.05.5 to 24.10.0rc4 and using the same configuration, the MAC address assignment to interfaces appears to have changed in such a way that (among other things) it is not possible to assign a DHCP-received static address for the 'br-lan' interface, because said interface's MAC is generated randomly each boot. The 'wan' device appears to have been assigned the same MAC as 'br-lan'.

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

ipq40xx/generic

### Device

NETGEAR RBR50

### Image kind

Official downloaded image

### Steps to reproduce

Have a working Orbi RBR50 with a sane configuration. Upgrade with a sysupgrade image to 24.10.0-rc4.

(If you revert back to 23.05.5, MAC address assignments revert as described below.)

### Actual behaviour

Devices' MAC addresses on the orbi are assigned like this in 24.10.0-rc4:
br-lan     RANDOM1
wan        RANDOM1
eth0       RANDOM1
lan1       RANDOM1
lan2       RANDOM1
lan3       RANDOM1
phy0-mesh0 HW1
phy1-ap0   HW2
phy2-ap0   HW3

### Expected behaviour

Devices' MAC addresses on the orbi are assigned like this in 24.10.0-rc4:
br-lan     HW2
wan        HW2+1
eth0       HW2
lan1       HW2
lan2       HW2
lan3       HW2
phy0-mesh0 HW1
phy1-ap0   HW2
phy2-ap0   HW3

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-12-27:
Hmm, this is weird since 23.05 already includes the DSA conversion and I do not see any later changes regarding networking.

Can you install dtc on the device, and provide the output of:
`dtc -I fs -O dts -o /tmp/live.dts /proc/device-tree/`

By output, I mean the generated `/tmp/live.dts` file because MAC setting depends on GMAC MAC being set but no ethernet aliases are set (These were previously most likely set in the SoC DTSI) so U-Boot is not setting the MAC at all.

skylerbunny, 2024-12-27:
As requested.

/dts-v1/;

/ {
	compatible = "netgear,rbr50";
	model = "NETGEAR RBR50";
	interrupt-parent = <0x01>;
	#address-cells = <0x01>;
	#size-cells = <0x01>;

	reserved-memory {
		ranges;
		#address-cells = <0x01>;
		#size-cells = <0x01>;

		smem@87e00000 {
			no-map;
			reg = <0x87e00000 0x80000>;
		};

		tz@87e80000 {
			no-map;
			reg = <0x87e80000 0x180000>;
		};
	};

	clocks {

		xo {
			compatible = "fixed-clock";
			#clock-cells = <0x00>;
			phandle = <0x0e>;
			clock-frequency = <0x2dc6c00>;
		};

		sleep_clk {
			compatible = "fixed-clock";
			#clock-cells = <0x00>;
			phandle = <0x0f>;
			clock-frequency = <0x7d00>;
		};
	};

	opp-table {
		compatible = "operating-points-v2";
		phandle = <0x06>;
		opp-shared;

		opp-200000000 {
			opp-hz = <0x00 0xbebc200>;
			clock-latency-ns = <0x3e800>;
		};

		opp-716000000 {
			opp-hz = <0x00 0x2aad4b00>;
			clock-latency-ns = <0x3e800>;
		};

		opp-48000000 {
			opp-hz = <0x00 0x2dc6c00>;
			clock-latency-ns = <0x3e800>;
		};

		opp-500000000 {
			opp-hz = <0x00 0x1dcd6500>;
			clock-latency-ns = <0x3e800>;
		};
	};

[... 853 lines cut ...]
			device_type = "cpu";
			qcom,acc = <0x09>;
			next-level-cache = <0x02>;
			enable-method = "qcom,kpss-acc-v2";
			operating-points-v2 = <0x06>;
			reg = <0x02>;
			clock-frequency = <0x00>;
			clock-latency = <0x3e800>;
		};

		cpu@0 {
			compatible = "arm,cortex-a7";
			clocks = <0x05 0x09>;
			qcom,saw = <0x04>;
			device_type = "cpu";
			qcom,acc = <0x03>;
			next-level-cache = <0x02>;
			enable-method = "qcom,kpss-acc-v2";
			operating-points-v2 = <0x06>;
			reg = <0x00>;
			clock-frequency = <0x00>;
			clock-latency = <0x3e800>;
		};
	};
};

skylerbunny, 2024-12-27:
Two things I should probably mention which may be of help...

1. If I revert back to 23.05.5, the MAC addresses go 'back to normal', so I'm guessing that whatever was set previously remains.
2. In order to use mesh networking (at all), the image is the 'attended sysupgrade' defaults but with wpad-mbedtls installed, so built like this:
  https://firmware-selector.openwrt.org/?version=23.05.5&target=ipq40xx%2Fgeneric&id=netgear_rbr50
  ath10k-board-qca4019 ath10k-firmware-qca4019-ct base-files busybox ca-bundle dnsmasq dropbear firewall4 fstools kmod-ath10k-ct kmod-gpio-button-hotplug kmod-leds-gpio kmod-nft-offload kmod-usb-dwc3 kmod-usb-dwc3-qcom kmod-usb3 libc libgcc libustream-mbedtls logd mtd netifd nftables odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd ath10k-firmware-qca9984-ct e2fsprogs kmod-fs-ext4 losetup luci (( -wpad-basic-mbedtls +wpad-mbedtls ))
  (build)
  (nmrpflash factory image so it properly is put on Orbi partition 1)
  (boot, restore 23.05.5 config and reboot)
  Revert involves using already created image for 23.05.5, using the same procedure as above, and all MACs are restored.

robimarko, 2024-12-27:
Can you also grab the same DTC output on 23.05 as well?

I think the issue is just lack of the ethernet alias, and that will confirm it

skylerbunny, 2024-12-27:
dtc on 23.05:
/dts-v1/;

/ {
	compatible = "netgear,rbr50";
	model = "NETGEAR RBR50";
	interrupt-parent = <0x01>;
	#address-cells = <0x01>;
	#size-cells = <0x01>;

	reserved-memory {
		ranges;
		#address-cells = <0x01>;
		#size-cells = <0x01>;

		smem@87e00000 {
			no-map;
			reg = <0x87e00000 0x80000>;
		};

		tz@87e80000 {
			no-map;
			reg = <0x87e80000 0x180000>;
		};
	};

	clocks {

		xo {
			compatible = "fixed-clock";
			#clock-cells = <0x00>;
			clock-frequency = <0x2dc6c00>;
		};

		sleep_clk {
			compatible = "fixed-clock";
			#clock-cells = <0x00>;
			phandle = <0x15>;
			clock-output-names = "gcc_sleep_clk_src";
			clock-frequency = <0x7d00>;
		};
	};

	opp_table0 {
		compatible = "operating-points-v2";
		phandle = <0x06>;
		opp-shared;

		opp-200000000 {
			opp-hz = <0x00 0xbebc200>;
			clock-latency-ns = <0x3e800>;
		};

		opp-716000000 {
			opp-hz = <0x00 0x2aad4b00>;
			clock-latency-ns = <0x3e800>;
		};

		opp-48000000 {
			opp-hz = <0x00 0x2dc6c00>;
			clock-latency-ns = <0x3e800>;
		};

		opp-500000000 {
			opp-hz = <0x00 0x1dcd6500>;
			clock-latency-ns = <0x3e800>;
		};
	};

	soc {
[... 843 lines cut ...]
			device_type = "cpu";
			qcom,acc = <0x09>;
			next-level-cache = <0x02>;
			enable-method = "qcom,kpss-acc-v2";
			operating-points-v2 = <0x06>;
			reg = <0x02>;
			clock-frequency = <0x00>;
			clock-latency = <0x3e800>;
		};

		cpu@0 {
			compatible = "arm,cortex-a7";
			clocks = <0x05 0x09>;
			qcom,saw = <0x04>;
			device_type = "cpu";
			qcom,acc = <0x03>;
			next-level-cache = <0x02>;
			enable-method = "qcom,kpss-acc-v2";
			operating-points-v2 = <0x06>;
			reg = <0x00>;
			clock-frequency = <0x00>;
			clock-latency = <0x3e800>;
		};
	};
};

robimarko, 2024-12-27:
Can you try adding: `ethernet0 = &gmac;` above
https://github.com/openwrt/openwrt/blob/main/target/linux/ipq40xx/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq4019-orbi.dtsi#L15

skylerbunny, 2024-12-27:
I am guessing this would require me to rebuild one of the packages? I'd probably need a guide on how to do this and what to rebuild. (Or a place in the filesystem where the final result of this change would be added to prove the fix).

professor-jonny, 2024-12-27:
It would require you setting up a build environment and building your own image to test the change as rob suggested.

robimarko, 2024-12-28:
@skylerbunny Can you just try this image for testing?
https://we.tl/t-OL8U4tpY96

grosjo, 2024-12-28:
I have teh same behavior on all my devices (br-lan MAC changes at each reboot)

grosjo, 2024-12-28:
It seems it takes randomly one of the mac of the bridged devices

grosjo, 2024-12-28:
In my case (eth0, eth1 and wireless are all bridged)
and then br-lan takes on of the MAC, but not always the same

grosjo, 2024-12-28:
It was very obvious, as I am limiting access to my devices via MAC-filter, and after upgrade, all the network was blocked

skylerbunny, 2024-12-28:
@robimarko I appreciate it, can I also get the factory image of what you built? I upgraded with the sysupgrade and it booted to the alternate Orbi partition. Once I have it (both) I can confirm for you. Thanks.

robimarko, 2024-12-28:
@skylerbunny Here you go:
https://we.tl/t-eJPmo1Kzr7

skylerbunny, 2024-12-28:
Confirmed your image did it. Behaviour is restored to 23.05.5, even with the same MAC addresses for the eth/lan/br-lan (HW2), and progression of HW2+1 for 'wan'.
OpenWrt 24.10-SNAPSHOT, r28231-76a546a83a

robimarko, 2024-12-28:
If that works then https://github.com/openwrt/openwrt/pull/17414 will fix it

grosjo, 2024-12-29:
What about other devices ? (Comfast, Kuwfi) having the same issue with RC4 ?

skylerbunny, 2024-12-29:
I'd guess there are two ways to deal with this:
1. Create a new issue a lot like the one I opened, explaining the problem for 'your chosen platform' (like how mine is orbi), or...
2. The developers create an umbrella issue referencing https://github.com/openwrt/openwrt/commit/cd9c7211241ea605e8713f5b9f0692ec0f7b765b , where they back-check platforms that _use_ ipq40xx to verify if any (or all) platforms that do, have this MAC issue due to that commit.

If this is widespread, (2) might be the better approach and might only require one larger commit that fixes all (remaining) platforms at once.

grosjo, 2024-12-29:
None of my devices are using ipq40xx.

Not sure what do you suggest in practical terms

skylerbunny, 2024-12-29:
Maybe @robimarko has an opinion?

If the named platforms don't share the same backend chipset(s), then I wonder if the best you can do is create new issues individually for each backend chipset, and let the developers research, because the core reason for this may not be the same across them?

grosjo, 2024-12-29:
On the contrary, it is probably one commit to be reverted

closed 2024-12-29 by openwrt-bot

robimarko, 2024-12-29:
@grosjo If they are not using the same chipset then they have a different root cause, open an issue for them.

grosjo, 2024-12-29:
Then why the problem did not occur before rc3 or 4 ?

robimarko, 2024-12-29:
@grosjo For this device the issue has been present for a long time, its not a newer RC issue.

Again, open an issue for your device as they dont have anything in common

Neustradamus, 2025-01-15:
@grosjo: It has been solved?

grosjo, 2025-01-15:
no

robimarko, 2025-01-15:
This should be fixed in RC5

Neustradamus, 2025-01-16:
@robimarko: What is the PR / commit which solves it?

Linked to:
- https://github.com/openwrt/openwrt/issues/17443

robimarko, 2025-01-16:
Nothing fixes that issue AFAIK, its unrelated to the issue we are commenting in.

commit 9ea174c7 (2024-12-29): ipq40xx: orbi: add ethernet0 alias

Netgear Orbi devices rely on ethernet0 alias to be present to U-Boot will
populate the MAC.

This fixes the random MAC on each boot after the ethernet0 alias was
dropped from the SoC DTSI.

Fixes: cd9c7211241e ("ipq40xx: 6.1: use latest DSA and ethernet patches")
Fixes: #17384
Link: https://github.com/openwrt/openwrt/pull/17414
PR #17414 (2024-12-29): ipq40xx: orbi: add ethernet0 alias
Netgear Orbi devices rely on ethernet0 alias to be present to U-Boot will populate the MAC.

This fixes the random MAC on each boot after the ethernet0 alias was dropped from the SoC DTSI.

Fixes: cd9c7211241e ("ipq40xx: 6.1: use latest DSA and ethernet patches")
Fixes: #17384
commit 1eff7379 (2024-12-29): ipq40xx: orbi: add ethernet0 alias

Netgear Orbi devices rely on ethernet0 alias to be present to U-Boot will
populate the MAC.

This fixes the random MAC on each boot after the ethernet0 alias was
dropped from the SoC DTSI.

Fixes: cd9c7211241e ("ipq40xx: 6.1: use latest DSA and ethernet patches")
Fixes: #17384
Link: https://github.com/openwrt/openwrt/pull/17414

(cherry picked from commit 9ea174c7bf64ec34e96871ce223d7a597ca80d26)