openwrt/openwrt #17385: wget / SSL issues when updating packages

milouk, 2024-12-26:
### Describe the bug

I am on latest SNAPSHOT build on AX3600. When trying to perform `apk update` and `apk upgrade` both via cli or LuCI i am getting SSL issues.

### OpenWrt version

r28406-d1e9c50d06

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Try to run `apk update` or `apk upgrade`. You will get the following respectively:

```
root@OpenWrt:~# apk update
fetch https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/packages.adb
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
WARNING: updating and opening https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/packages.adb: remote server returned error (try 'apk update')
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/packages.adb
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/packages.adb: remote server returned error (try 'apk update')
fetch https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.6.67-1-b03a0b3cb6d43ef49e92e30e0d9e6d5f/packages.adb
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
WARNING: updating and opening https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.6.67-1-b03a0b3cb6d43ef49e92e30e0d9e6d5f/packages.adb: remote server returned error (try 'apk update')
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/packages.adb
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/packages.adb: remote server returned error (try 'apk update')
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/packages.adb
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/packages.adb: remote server returned error (try 'apk update')
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/packages.adb
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/packages.adb: remote server returned error (try 'apk update')
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/packages.adb
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/packages.adb: remote server returned error (try 'apk update')
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/video/packages.adb
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/video/packages.adb: remote server returned error (try 'apk update')
 [https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/packages.adb]
 [https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.6.67-1-b03a0b3cb6d43ef49e92e30e0d9e6d5f/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/video/packages.adb]
8 unavailable, 0 stale; 204 distinct packages available
```

and 

```
root@OpenWrt:~# apk upgrade
fetch https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/packages.adb
[... 231 lines cut ...]
ucode-mod-math-2024.12.06~209f041f-r1 aarch64_cortex-a53 {feeds/base/package/utils/ucode} (ISC) [installed]
ucode-mod-nl80211-2024.12.06~209f041f-r1 aarch64_cortex-a53 {feeds/base/package/utils/ucode} (ISC) [installed]
ucode-mod-rtnl-2024.12.06~209f041f-r1 aarch64_cortex-a53 {feeds/base/package/utils/ucode} (ISC) [installed]
ucode-mod-ubus-2024.12.06~209f041f-r1 aarch64_cortex-a53 {feeds/base/package/utils/ucode} (ISC) [installed]
ucode-mod-uci-2024.12.06~209f041f-r1 aarch64_cortex-a53 {feeds/base/package/utils/ucode} (ISC) [installed]
ucode-mod-uclient-2024.10.22~88ae8f20-r1 aarch64_cortex-a53 {feeds/base/package/libs/uclient} (ISC) [installed]
ucode-mod-uloop-2024.12.06~209f041f-r1 aarch64_cortex-a53 {feeds/base/package/utils/ucode} (ISC) [installed]
uhttpd-2023.06.25~34a8a74d-r4 aarch64_cortex-a53 {feeds/base/package/network/services/uhttpd} (ISC) [installed]
uhttpd-mod-lua-2023.06.25~34a8a74d-r4 aarch64_cortex-a53 {feeds/base/package/network/services/uhttpd} (ISC) [installed]
uhttpd-mod-ubus-2023.06.25~34a8a74d-r4 aarch64_cortex-a53 {feeds/base/package/network/services/uhttpd} (ISC) [installed]
urandom-seed-3 aarch64_cortex-a53 {feeds/base/package/system/urandom-seed} (GPL-2.0-only) [installed]
urngd-2023.11.01~44365eb1-r1 aarch64_cortex-a53 {feeds/base/package/system/urngd} (GPL-2.0 BSD-3-Clause) [installed]
usign-2020.05.23~f1f65026-r1 aarch64_cortex-a53 {feeds/base/package/system/usign} (ISC) [installed]
wifi-scripts-1.0-r1 noarch {feeds/base/package/network/config/wifi-scripts} (GPL-2.0) [installed]
wireless-regdb-2024.10.07-r1 noarch {feeds/base/package/firmware/wireless-regdb} (ISC) [installed]
wpad-basic-mbedtls-2024.09.15~5ace39b0-r2 aarch64_cortex-a53 {feeds/base/package/network/services/hostapd} (BSD-3-Clause) [installed]
zlib-1.3.1-r1 aarch64_cortex-a53 {feeds/base/package/libs/zlib} (Zlib) [installed]
```

- Attended Sysupgrade is working as expected and currently this is the only way i can upgrade my build.
- This happened for the first time like since 3 weeks ago.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-12-27:
How did you get the image?

Was it downloaded from OpenWrt downloads page or via ASU/firmware selector?

milouk, 2024-12-27:
Via attended sysupgrade through the LuCI

robimarko, 2024-12-27:
That is the likely cause of your issue then, because APK requires SSL support and includes the required default packages as dependencies.

milouk, 2024-12-27:
hm i see, is there an easy way to now fix this?

robimarko, 2024-12-27:
Yeah, you can just generate a new sysupgrade image with your desired package list via the firmware selector and it will include APK and its dependencies so after sysupgrade it should just work

Neustradamus, 2024-12-28:
Can you check this PR?
- https://github.com/openwrt/packages/pull/25633

Linked to:
- https://github.com/openwrt/packages/issues/25625
- https://github.com/openwrt/openwrt/issues/17392

robimarko, 2025-01-06:
What is your package list?

pentago, 2025-01-06:
@robimarko this is the list:

```
apk-mbedtls
base-files
ca-bundle
ca-certificates
dnsmasq
dropbear
firewall4
fstools
git
kernel
kmod-crypto-hw-eip93
kmod-gpio-button-hotplug
kmod-leds-gpio
kmod-mt7915-firmware
kmod-nft-offload
kmod-usb-ledtrig-usbport
kmod-usb3
libc
logd
luci
luci-app-attendedsysupgrade
mtd
netifd
odhcp6c
odhcpd-ipv6only
openssh-sftp-server
owut
ppp
ppp-mod-pppoe
procd-ujail
tmux
uboot-envtools
uci
uclient-fetch
urandom-seed
urngd
vim
wpad-basic-mbedtls
```

pentago, 2025-01-10:
@robimarko any thoughts?

robimarko, 2025-01-16:
Is that list that apk returns?

I was asking about the package list you have in the firmware selector package list

pentago, 2025-01-16:
@robimarko that is my list from firmware selector (not apk)..

robimarko, 2025-01-16:
Yeah, well the issue is that you have removed `libustream-mbedtls` from the package list and thus APK cannot fetch and validate the feeds.

Simply include it in the package list as it is part of the default list since APK requires it.

milouk, 2025-01-16:
@robimarko  that seemed to fixed the issue for me. So if thats ok with @pentago as well im closing the issue

pentago, 2025-01-16:
Just tried it.. As for apk over ssh, it works fine but I syill see thes errors in Luci:
[image]

Any thoughts on that perhaps?

robimarko, 2025-01-16:
Those are not actually errors, APK for some weird reason prints those informational prints both on stdout and stderr but they are not errors

pentago, 2025-01-17:
Okay then, since I don't see the same output (APK cli via SSH do not print any errors/warnings) on LUCI and apk CLI, I guess its just apk implementation thats problematic, right?

robimarko, 2025-01-17:
Its just how APK does stuff, just ignore it

closed 2025-01-18 by milouk