openwrt/openwrt #17391: 02_network script not updating mac addresses with baked in config.

professor-jonny, 2024-12-27:
### Describe the bug

The pre boot init script 02_network does not update the mac addresses of the network interfaces if a config is baked into the flash image.

it seems the 02_network creates the board.json using the standard functions with the correct mac addresses but the config update script will only create a config it will not update macs if a network config is present in the device at boot.

This seems like a bug, and I think it should update the mac address if no mac address is set in the network config.

This seeming caused me a bit of a headache as nowhere does it say that is how it works, and it is assumed it updates the mac addresses when it does not, I was trying to create a new target and thinking i have issues with my scripts and timing and as such of my device.

### OpenWrt version

r28147-95c4ba27b0

### OpenWrt release

snapshot

### OpenWrt target/subtarget

qualcommax/generic

### Device

tp-link deco x80-5g

### Image kind

Self-built image

### Steps to reproduce

build a image with a network config built in and omit mac addresses for the interfaces

### Actual behaviour

random macks are used in stead of the macks generated from the 02_network script

### Expected behaviour

If no mac is found in the network config it should be updated at boot time from the the valid label mac or as such as generated by the 02_network script.

### Additional info

removing the built in network configuration network config is created at first boot and macs are populated as expected.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

testuser7, 2025-01-03:
It is designed this way that the config is only created if it does not exist at startup.

If you want to add functionality that adds a MAC address anyway, you have to remember that someone can configure some other MAC address and then it should not be overwritten.

professor-jonny, 2025-01-03:
That was my thinking that if a mac is not configured it would update it as per the labelmac or as such. from the 02_network script.