openwrt/openwrt #17392: /usr/bin/wget is not created when both wget-ssl and wget-nossl are selected

1plus-1, 2024-12-27:
### Describe the bug

Now wget-ssl and wget-nossl have the same alternative priority(300).
https://github.com/openwrt/packages/blob/ca9c3d1baea38f0e5cedae7b74cfe8ef15a3ec83/net/wget/Makefile#L52

When both are selected, function update_alternatives() in function.sh would not create the link due to the best_prio is alwasy equal or larger than pkg_prio.
https://github.com/openwrt/openwrt/blob/d6c5805db37d42d29e190b7f2dc49ed8e4545e00/package/base-files/files/lib/functions.sh#L327

The suggest fix would be changing it to '-le'.
`if [ "$best_prio" -le "$pkg_prio" ]; then`

### OpenWrt version

d1e9c50d06a8cb618cb85ab489cbcccaec220636

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Openwrt One

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

1plus-1, 2024-12-27:
Yes. It's from mainline 3 days ago: https://github.com/openwrt/openwrt/commit/d1e9c50d06a8cb618cb85ab489cbcccaec220636

1plus-1, 2024-12-27:
A ticket to request updating the priority is created: https://github.com/openwrt/packages/issues/25625

Neustradamus, 2024-12-28:
Linked to:
- https://github.com/openwrt/packages/pull/25633
- https://github.com/openwrt/packages/issues/25625
- https://github.com/openwrt/openwrt/issues/17385
- https://github.com/openwrt/openwrt/issues/17392

robimarko, 2025-01-01:
Fixed in: https://github.com/openwrt/packages/issues/25625

closed 2025-01-01 by robimarko