openwrt/openwrt #17398: SNAPSHOT (r28409-de0c143742): Orange Pi R1 Plus LTS: eth0 not working

maximi89, 2024-12-27:
### Describe the bug

No matter how configured eth0 is not passing (neither sending nor receiving) traffic. Link state is shown correctly though.

eth1 is working as it should.

### OpenWrt version

r28409-de0c143742

### OpenWrt release

SNAPSHOT 

### OpenWrt target/subtarget

rockchip/armv8

### Device

Xunlong Orange Pi R1 Plus LTS

### Image kind

Official downloaded image

### Actual behaviour

Interface ETH0 not working. When connect the interface with DHCP, static IP, PPoE, none works, only PCB LED show static green, and Ethernet interface show green LED blinking. No ping, no activity. 

### Expected behaviour

Interface ETH0, should works.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

maximi89, 2025-01-06:
I tried on last  version, same behavior.
OpenWrt SNAPSHOT r28515-70610a5240 / LuCI Master 25.002.60082~f001a35

It recognize when I plug the wire on eth0 (Motorcomm YT8531C), but it can't receive neither send data trough it. Doesn't get any IP as DHCP client, can't ping with static IP.

it show on log, it's using the driver (kernel ID 0x4f51e91b)
rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [YT8531 Gigabit Ethernet] (irq=POLL)

I don't know if that is correct, because it's Motorcomm YT8531C.

1715173329, 2025-01-08:
This issue was fixed on my board by [setting `motorcomm,rx-{clk,data}-drv-microamp` properties](https://github.com/torvalds/linux/commit/fc5a80a432607d05e85bba37971712405f75c546).
Try to change the values and see if it helps.

fyi:
https://github.com/torvalds/linux/blob/09a0fa92e5b45e99cf435b2fbf5ebcf889cf8780/Documentation/devicetree/bindings/net/motorcomm%2Cyt8xxx.yaml#L66
https://github.com/torvalds/linux/blob/09a0fa92e5b45e99cf435b2fbf5ebcf889cf8780/Documentation/devicetree/bindings/net/motorcomm%2Cyt8xxx.yaml#L83

maximi89, 2025-01-09:
> This issue was fixed on my board by [setting `motorcomm,rx-{clk,data}-drv-microamp` properties](https://github.com/torvalds/linux/commit/fc5a80a432607d05e85bba37971712405f75c546). Try to change the [...]

Hi, I tried what you said, I modified the DTB file and now it works, but very unstable. 

	ethernet@ff540000 {
		compatible = "rockchip,rk3328-gmac";
		reg = <0x00 0xff540000 0x00 0x10000>;
		interrupts = <0x00 0x18 0x04>;
		interrupt-names = "macirq";
		clocks = <0x02 0x64 0x02 0x57 0x02 0x58 0x02 0x5a 0x02 0x59 0x02 0x96 0x02 0xdf>;
		clock-names = "stmmaceth\0mac_clk_rx\0mac_clk_tx\0clk_mac_ref\0clk_mac_refout\0aclk_mac\0pclk_mac";
		resets = <0x02 0x63>;
		reset-names = "stmmaceth";
		rockchip,grf = <0x38>;
		tx-fifo-depth = <0x800>;
		rx-fifo-depth = <0x1000>;
		snps,txpbl = <0x04>;
		status = "okay";
		assigned-clocks = <0x02 0x64 0x02 0x66>;
		assigned-clock-parents = <0x4b 0x4b>;
		clock_in_out = "input";
		phy-mode = "rgmii";
		phy-supply = <0x1c>;
		pinctrl-0 = <0x4c>;
		pinctrl-names = "default";
		snps,aal;
		phy-handle = <0x4d>;
		tx_delay = <0x19>;
		rx_delay = <0x05>;

		mdio {
			compatible = "snps,dwmac-mdio";
			#address-cells = <0x01>;
			#size-cells = <0x00>;

			ethernet-phy@0 {
				compatible = "ethernet-phy-ieee802.3-c22";
				reg = <0x00>;
				motorcomm,auto-sleep-disabled;
				motorcomm,clk-out-frequency-hz = <0x7735940>;
				motorcomm,keep-pll-enabled;
				motorcomm,rx-clk-drv-microamp = <0x139c>;
				motorcomm,rx-data-drv-microamp = <0x139c>;
				pinctrl-0 = <0x4e>;
				pinctrl-names = "default";
				reset-assert-us = <0x3a98>;
				reset-deassert-us = <0xc350>;
				reset-gpios = <0x27 0x12 0x01>;
				phandle = <0x4d>;
			};
		};
	};

From 192.168.1.123 icmp_seq=1248 Destination Host Unreachable
From 192.168.1.123 icmp_seq=1249 Destination Host Unreachable
From 192.168.1.123 icmp_seq=1250 Destination Host Unreachable
From 192.168.1.123 icmp_seq=1251 Destination Host Unreachable
From 192.168.1.123 icmp_seq=1252 Destination Host Unreachable
64 bytes from 192.168.1.101: icmp_seq=1264 ttl=64 time=0.779 ms
64 bytes from 192.168.1.101: icmp_seq=1295 ttl=64 time=0.749 ms
64 bytes from 192.168.1.101: icmp_seq=1305 ttl=64 time=0.931 ms
64 bytes from 192.168.1.101: icmp_seq=1308 ttl=64 time=0.957 ms
64 bytes from 192.168.1.101: icmp_seq=1324 ttl=64 time=0.754 ms
64 bytes from 192.168.1.101: icmp_seq=1339 ttl=64 time=0.970 ms
From 192.168.1.123 icmp_seq=1347 Destination Host Unreachable
From 192.168.1.123 icmp_seq=1348 Destination Host Unreachable
From 192.168.1.123 icmp_seq=1349 Destination Host Unreachable
From 192.168.1.123 icmp_seq=1350 Destination Host Unreachable
From 192.168.1.123 icmp_seq=1351 Destination Host Unreachable
From 192.168.1.123 icmp_seq=1352 Destination Host Unreachable
64 bytes from 192.168.1.101: icmp_seq=1354 ttl=64 time=0.786 ms
64 bytes from 192.168.1.101: icmp_seq=1367 ttl=64 time=0.975 ms
64 bytes from 192.168.1.101: icmp_seq=1382 ttl=64 time=0.783 ms

1715173329, 2025-01-10:
Could you please try to set `phy-mode` to `rgmii-id`? This makes the delay to be added by the PHY instead of MAC.

maximi89, 2025-01-11:
do you mean this?

 phy-mode = "rgmii";

`&gmac2io {
 assigned-clocks = <&cru 100>, <&cru 102>;
 assigned-clock-parents = <&gmac_clk>, <&gmac_clk>;
 clock_in_out = "input";
 phy-mode = "rgmii";
 phy-supply = <&vcc_io>;
 pinctrl-0 = <&rgmiim1_pins>;
 pinctrl-names = "default";
 snps,aal;

 mdio {
  compatible = "snps,dwmac-mdio";
  #address-cells = <1>;
  #size-cells = <0>;
 };
};`

1715173329, 2025-01-12:
Yes, change it to rgmii-id.

maximi89, 2025-01-12:
I have to keep the tx-delay.and rx-delay at 1?, or I change to previous values?, 19 and 05?

1715173329, 2025-01-12:
*-delay properties will be ignored in rgmii-id mode.

maximi89, 2025-01-12:
done, it works very well. 
tx_delay = <0x0>;
rx_delay = <0x0>;

phy-mode = "rgmii-id";

```
&gmac2io {
 phy-handle = <&yt8531c>;
 **tx_delay = <0x0>;**
 **rx_delay = <0x0>;**
```

```
&gmac2io {
 assigned-clocks = <&cru 100>, <&cru 102>;
 assigned-clock-parents = <&gmac_clk>, <&gmac_clk>;
 clock_in_out = "input";
 **phy-mode = "rgmii-id";**
```

1715173329, 2025-01-12:
Could you please show me the value of `motorcomm,rx-{clk,data}-drv-microamp` you edited? In your reply it seems unchanged.

maximi89, 2025-01-12:
My mistake, but from first comment I were using the recommended values:
   motorcomm,rx-clk-drv-microamp = <5020>;
   motorcomm,rx-data-drv-microamp = <5020>;

```
&gmac2io {
 phy-handle = <&yt8531c>;
 tx_delay = <0x0>;
 rx_delay = <0x0>;
 status = "okay";

 mdio {

  yt8531c: ethernet-phy@0 {
   compatible = "ethernet-phy-ieee802.3-c22";
   reg = <0>;

   motorcomm,auto-sleep-disabled;
   motorcomm,clk-out-frequency-hz = <125000000>;
   motorcomm,keep-pll-enabled;
   motorcomm,rx-clk-drv-microamp = <5020>;
   motorcomm,rx-data-drv-microamp = <5020>;

   pinctrl-0 = <&eth_phy_reset_pin>;
   pinctrl-names = "default";
   reset-assert-us = <15000>;
   reset-deassert-us = <50000>;
   reset-gpios = <&gpio1 18 1>;
  };
 };
};
```

maximi89, 2025-01-12:
something it's a bit strange for me, in random resets I get more speed on speedtest.net.

Modem to my PC: 
800/700 Mb (down/up)

OrangePi:
350/800 Mb (down/up)
500/600 Mb (down/up)

As I said, the values I get on speedtest are random.

1715173329, 2025-01-12:
> My mistake, but from first comment I were using the recommended values:
> motorcomm,rx-clk-drv-microamp = <5020>;
> motorcomm,rx-data-drv-microamp = <5020>;

hmmm it sounds odd as these values were already defined in kernel 6.6... so I guess `{r,t}x-delay` is the cause.

1715173329, 2025-01-12:
> something it's a bit strange for me, in random resets I get more speed or less, on speedtest.net.

try speed.cloudflare.com or you can test it with iperf3 like this:
[image]

maximi89, 2025-01-12:
What I mean, it's not having same Internet speed at every router start, most times I get 350/800 Mb, but others few, 600/600 Mb, that is random.
do you have any idea why happens this?

1715173329, 2025-01-12:
unfortunately no

maximi89, 2025-01-16:
Due to high temperature, I bought an aluminium heatsink, but it haven't arrive yet. Seem to be a good speed now, more than 500Mb download.

I change Iptables to Nftables and set the NFTables with all my CPU Cores.

/etc/rc.local:
```
echo 7 > /proc/irq/25/smp_affinity
echo f > /sys/class/net/eth0/queues/rx-0/rps_cpus
```

```
  tx_delay = <0x0>;
  rx_delay = <0x5>;
in mdio I set the reg to 1, must be different from the other interface.
reg = <1>;
```

This is my DTS config for my YT8531C:

 ```
gmac2io: ethernet@ff540000 {
  compatible = "rockchip,rk3328-gmac";
  reg = <0x0 0xff540000 0x0 0x10000>;
  interrupts = <0 24 4>;
  interrupt-names = "macirq";
  clocks = <&cru 100>, <&cru 87>,
    <&cru 88>, <&cru 90>,
    <&cru 89>, <&cru 150>,
    <&cru 223>;
  clock-names = "stmmaceth", "mac_clk_rx",
		      "mac_clk_tx", "clk_mac_ref",
		      "clk_mac_refout", "aclk_mac",
		      "pclk_mac";
  assigned-clocks = <&cru 100>, <&cru 102>;
  assigned-clock-parents = <&gmac_clk>, <&gmac_clk>;
  clock_in_out = "input";

  rockchip,grf = <&grf>;
  resets = <&cru 99>;
  reset-names = "stmmaceth";

  tx-fifo-depth = <8192>;
  rx-fifo-depth = <16384>;
  snps,txpbl = <0x4>;

  phy-mode = "rgmii-id";
  phy-supply = <&vcc_io>;
  pinctrl-0 = <&rgmiim1_pins>;
  pinctrl-names = "default";
  snps,aal;

  phy-handle = <&yt8531c>;
  tx_delay = <0x0>;
  rx_delay = <0x5>;
  status = "okay";

  mdio {
   compatible = "snps,dwmac-mdio";
   #address-cells = <1>;
   #size-cells = <0>;

   yt8531c: ethernet-phy@0 {
    compatible = "ethernet-phy-ieee802.3-c22";
    reg = <1>;

    motorcomm,auto-sleep-disabled;
    motorcomm,clk-out-frequency-hz = <125000000>;
    motorcomm,keep-pll-enabled;
    motorcomm,rx-clk-drv-microamp = <5020>;
    motorcomm,rx-data-drv-microamp = <5020>;

    pinctrl-0 = <&eth_phy_reset_pin>;
    pinctrl-names = "default";
    reset-assert-us = <15000>;
    reset-deassert-us = <50000>;
    reset-gpios = <&gpio1 18 1>;
  };
 };
};
```

Also, this is the only clocks working, even when they are set more than this on the DTS file.
```
root@OpenWrtR1PlusLTS:~# cat /sys/kernel/debug/clk//clk_summary |grep -i f54
    clk_mac2io                       2       2        0        125000000   0          0     50000      Y      ff540000.ethernet               stmmaceth                
                                                                                                              ff540000.ethernet               stmmaceth                
       clk_mac2io_tx                 1       1        0        125000000   0          0     50000      Y         ff540000.ethernet               mac_clk_tx               
             aclk_mac2io             1       1        0        163840000   0          0     50000      Y               ff540000.ethernet               aclk_mac                 
                pclk_mac2io          1       1        0        54613334    0          0     50000      Y                  ff540000.ethernet               pclk_mac  
```

1715173329, 2025-01-19:
https://lore.kernel.org/linux-rockchip/20250119091154.1110762-1-cnsztl@gmail.com/

I've sent a patch to change the phy mode to rgmii-id for this board.

maximi89, 2025-01-19:
I'm doing some changes, but I can't test until I get the heatsink and fan, because I get more than 94 ºC and it shutdown. Will be this week arriving to me, so I will post the test then.

/etc/rc.local
echo e > /proc/irq/25/smp_affinity #Not working, it kepts a 8.
echo 7 > /sys/class/net/eth0/queues/rx-0/rps_cpus #not working, it kepts a 1
echo 7 > /sys/class/net/eth0/queues/tx-0/xps_cpus
echo 32768 > /sys/class/net/eth0/queues/rx-0/rps_flow_cnt
echo 33000 > /proc/sys/net/netfilter/nf_conntrack_buckets
echo 512 > /proc/sys/net/netfilter/nf_conntrack_frag6_low_thresh
echo 1024 > /proc/sys/net/netfilter/nf_conntrack_frag6_high_thresh
echo 524288 > /proc/sys/net/ipv4/ipfrag_low_thresh
echo 1048576 > /proc/sys/net/ipv4/ipfrag_high_thresh
echo "rc.local ejecutado" >> /tmp/rc_local.log

actually I'm working with "packet steering" and I get near 450Mb/700Mb (down/up)

maximi89, 2025-01-20:
I see no difference with the lines

> btw if possible could you please try to add the following properties to the `ethernet-phy@0` node and see if it provides more stable connection?
> 
> rx-internal-delay-ps = <1500>;
> tx-internal-delay-ps = <1500>;
> 
> and you can also try to set them to other values and see if it works better, see: https://github.com/torvalds/linux/blob/09a0fa92e5b45e99cf435b2fbf5ebcf889cf8780/Documentation/devicetree/bindings/net/motorcomm%2Cyt8xxx.yaml#L21

[image]

I also added 
```
 gmac2io: ethernet@ff540000 {

  rx-fifo-size = <4096>;
  tx-fifo-size = <4096>;
  rx-fifo-size-gige = <16384>;
  tx-fifo-size-gige = <8192>;
```

Seems to help.

Something strange is, this helps a lot in upload of interface eth0.2 (WAN), now I have:
echo f > /sys/class/net/eth1/queues/rx-0/rps_cpus 

[image]

1715173329, 2025-01-21:
> I also added
> 
> ```
>  gmac2io: ethernet@ff540000 {
> 
>   rx-fifo-size = <4096>;
>   tx-fifo-size = <4096>;
>   rx-fifo-size-gige = <16384>;
>   tx-fifo-size-gige = <8192>;
> ```
> 
> Seems to help.

This really makes me confused, these properties do not even exist in dwmac driver...

maximi89, 2025-01-22:
> This really makes me confused, these properties do not even exist in dwmac driver...

Yes, my mistake. When I were testing, I confuse the better bandwidth values with the additional lines.

[image]  

As far as I remember, I just have this lines:
```
echo e > /proc/irq/25/smp_affinity
echo 3 > /sys/class/net/eth0/queues/rx-0/rps_cpus
echo 5 > /sys/class/net/eth0/queues/tx-0/xps_cpus
echo e > /sys/class/net/eth1/queues/rx-0/rps_cpus 
```

DTS:
```
gmac2io: ethernet@ff540000 {
  compatible = "rockchip,rk3328-gmac";
  reg = <0x0 0xff540000 0x0 0x10000>;
  interrupts = <0 24 4>;
  interrupt-names = "macirq";
  clocks = <&cru 100>, <&cru 87>,
    <&cru 88>, <&cru 90>,
    <&cru 89>, <&cru 150>,
    <&cru 223>;
  clock-names = "stmmaceth", "mac_clk_rx",
		      "mac_clk_tx", "clk_mac_ref",
		      "clk_mac_refout", "aclk_mac",
		      "pclk_mac";
  assigned-clocks = <&cru 100>, <&cru 102>;
  assigned-clock-parents = <&gmac_clk>, <&gmac_clk>;
  clock_in_out = "input";

  rockchip,grf = <&grf>;
  resets = <&cru 99>;
  reset-names = "stmmaceth";

  rx-fifo-depth = <16384>;
  tx-fifo-depth = <8192>;
  #rx-fifo-size = <4096>;
  #tx-fifo-size = <4096>;
  #rx-fifo-size-gige = <16384>;
  #tx-fifo-size-gige = <8192>;
  snps,txpbl = <0x4>;

  rx-internal-delay-ps = <1500>;
  tx-internal-delay-ps = <1500>;

  phy-mode = "rgmii-id";
  phy-supply = <&vcc_io>;
  pinctrl-0 = <&rgmiim1_pins>;
  pinctrl-names = "default";
  snps,aal;

  phy-handle = <&yt8531c>;
  tx_delay = <0x5>;
  rx_delay = <0x0>;
  status = "okay";

  mdio {
   compatible = "snps,dwmac-mdio";
   #address-cells = <1>;
   #size-cells = <0>;

   yt8531c: ethernet-phy@0 {
    compatible = "ethernet-phy-ieee802.3-c22";
    reg = <1>;

    #rx-internal-delay-ps = <1500>;
    #tx-internal-delay-ps = <1500>;

    motorcomm,auto-sleep-disabled;
    motorcomm,clk-out-frequency-hz = <125000000>;
    motorcomm,keep-pll-enabled;
    motorcomm,rx-clk-drv-microamp = <5020>;
    motorcomm,rx-data-drv-microamp = <5020>;

    pinctrl-0 = <&eth_phy_reset_pin>;
    pinctrl-names = "default";
    reset-assert-us = <15000>;
    reset-deassert-us = <50000>;
    reset-gpios = <&gpio1 18 1>;
   };
  };
 };
```